python-microscopy/python-microscopy #928: [BUG] possible bugs? PYME app errors in py27 channel with wxpython-3.x

csoeller, 2021-03-05:
**Describe the bug**

Errors when starting dh5view and visgui with latest PYME and wxpython-3.x

**To Reproduce**
Steps to reproduce the behavior:

1. Open visgui or dh5view with data file argument
2. No window opening
3. Errors are produced as listed further below

This is with py27 and wxpython as immediately below:

```
# Name                    Version                   Build  Channel
wxpython                  3.0.2.0                  py27_5    david_baddeley
```

**Fixed with:**

```
conda install wxpython=4
```

which installs as follows, all good after that:

```
# Name                    Version                   Build  Channel
wxpython                  4.0.4            py27hebc47a7_0  
```

**Expected behavior**
Not quite sure, should it have worked with wxpython-3.x or have dependencies been upgraded and I am behind?

**Error tracebacks**

**dh5view**

```
Traceback (most recent call last):
  File "/Users/csoe002/opt/anaconda2/envs/pyme-default-plain/lib/python2.7/site-packages/wx-3.0-osx_cocoa/wx/_core.py", line 16766, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/DSView/dsviewer.py", line 460, in LoadData
    vframe = DSViewFrame(im, None, im.filename, mode = mode)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/DSView/dsviewer.py", line 200, in __init__
    self.CreateFoldPanel()
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/ui/AUIFrame.py", line 122, in CreateFoldPanel
    genFcn(self.sidePanel)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/DSView/modules/LMAnalysis.py", line 205, in GenAnalysisPanel
    self._populateStdOptionsPanel(pan, vsizer)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/DSView/modules/LMAnalysis.py", line 158, in _populateStdOptionsPanel
    mdhChangedSignal = self.analysisController.onMetaDataChange)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/localization/MetaDataEdit.py", line 528, in createGUI
    clp = afp.collapsingPane(parent, caption=self.name, folded=self.folded)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/ui/manualFoldPanel.py", line 576, in __init__
    self.bmR = BitmapFromBits(r_arrow, 16, 16, ColourFromStyle('BLACK'))
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/ui/manualFoldPanel.py", line 37, in ColourFromStyle
    col = wx.Colour(col)
  File "/Users/csoe002/opt/anaconda2/envs/pyme-default-plain/lib/python2.7/site-packages/wx-3.0-osx_cocoa/wx/_gdi.py", line 130, in __init__
    _gdi_.Colour_swiginit(self,_gdi_.new_Colour(*args, **kwargs))
TypeError: in method 'new_Colour', expected argument 1 of type 'byte'
```

**Visgui**

```
Traceback (most recent call last):
  File "/Users/csoe002/opt/anaconda2/envs/pyme-default-plain/bin/visgui", line 11, in <module>
    load_entry_point('PYME', 'console_scripts', 'VisGUI')()
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/LMVis/VisGUI.py", line 499, in main
    main_(args.file, use_shaders=args.use_shaders, args=args)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/LMVis/VisGUI.py", line 472, in main_
    application = VisGuiApp(filename, use_shaders, args, 0)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/LMVis/VisGUI.py", line 459, in __init__
    wx.App.__init__(self, *args)
  File "/Users/csoe002/opt/anaconda2/envs/pyme-default-plain/lib/python2.7/site-packages/wx-3.0-osx_cocoa/wx/_core.py", line 8628, in __init__
    self._BootstrapApp()
  File "/Users/csoe002/opt/anaconda2/envs/pyme-default-plain/lib/python2.7/site-packages/wx-3.0-osx_cocoa/wx/_core.py", line 8196, in _BootstrapApp
    return _core_.PyApp__BootstrapApp(*args, **kwargs)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/LMVis/VisGUI.py", line 463, in OnInit
    self.main = VisGUIFrame(None, self.filename, use_shaders=self.use_shaders, cmd_args=self.cmd_args)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/LMVis/VisGUI.py", line 180, in __init__
    self.CreateFoldPanel()
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/ui/AUIFrame.py", line 122, in CreateFoldPanel
    genFcn(self.sidePanel)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/LMVis/visCore.py", line 155, in GenPanels
    self.filterPane = CreateFilterPane(sidePanel, self.pipeline.filterKeys, self.pipeline, self)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/LMVis/filterPane.py", line 35, in CreateFilterPane
    pane = FilterPane(panel, mapping, pipeline, visFr)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/LMVis/filterPane.py", line 255, in __init__
    self.pFilter = FilterPanel(self, filterKeys, pipeline.selectedDataSource)
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/LMVis/filterPane.py", line 69, in __init__
    self.populate()
  File "/Users/csoe002/Documents/src/PYME-src/python-microscopy/PYME/LMVis/filterPane.py", line 111, in populate
    ind = self.lFiltKeys.InsertItem(ind+1, key)
  File "/Users/csoe002/opt/anaconda2/envs/pyme-default-plain/lib/python2.7/site-packages/wx-3.0-osx_cocoa/wx/_controls.py", line 4702, in InsertItem
    return _controls_.ListCtrl_InsertItem(*args, **kwargs)
TypeError: ListCtrl_InsertItem() takes at most 2 arguments (3 given)
```

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: macOS 10.15.7
 - Python version: 2.7
 - numpy version: 1.16.6
 - wxpython version [if a UI bug]: 3.0.2.0

**Additional context**
Add any other context about the problem here.

David-Baddeley, 2021-03-05:
We're slowly starting to merge fixes to things which are bugs on wx >=4.1 (needed for python >3.7, and for a proper solution to the BigSur install issues). 

This unfortunately means dropping support for wx3, so you will need to upgrade wx (to wx strictly < 4.1) to use the latest source code (packages on python 2.7 should be stuck at a version where this is not an issue).  For now, we are still trying not to include anything which will break things on python 2.7, but are not investing any time on package management so you'll need to sort out the dependencies on your own. We're trying to gently nudge people towards py3, and it would be good to know what is keeping you on 2.7.

csoeller, 2021-03-05:
Thanks for explanations. Really, nothing keeping me on 2.7 for any (known) reason, I have a fully functioning PYME virtualenv with 3.7. I will switch over my macOS app wrappers to use that virtualenv and will then find out if there are any py3 issues in my own code.

The only other thing is to know the policies, at this stage it would seem you are saying that 2.7 'bugs' are only those occuring with a released conda install.

Is all acquisition now py3 compatible in principle? Changing to that code base on acqusition systems is going to be my summer project.

csoeller, 2021-03-05:
Closing since cannot be regarded as a bug with push for py3.

closed 2021-03-05 by csoeller