python-microscopy/python-microscopy #931: [BUG] event saving via cluster fileresults regression

barentine, 2021-03-06:
**Describe the bug**
The events do not seem to be automatically copied into the h5r files at the moment. 
```
DEBUG:root:pytables file open: /data/PYMEData/Bergamot/2021_3_4/C1/analysis/4_3_series_BO.h5r
Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/IO/h5rFile.py", line 148, in _appendToTable
    table = getattr(self._h5file.root, tablename)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/tables/group.py", line 818, in __getattr__
    return self._f_get_child(name)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/tables/group.py", line 698, in _f_get_child
    self._g_check_has_child(childname)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/tables/group.py", line 395, in _g_check_has_child
    % (self._v_pathname, name))
tables.exceptions.NoSuchNodeError: group ``/`` does not have a child named ``Events``

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/code/python-microscopy/PYME/IO/h5rFile.py", line 246, in _pollQueues
    self._appendToTable(tablename, np.hstack(recs))
  File "/home/ubuntu/code/python-microscopy/PYME/IO/h5rFile.py", line 158, in _appendToTable
    expectedrows=500000)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/tables/file.py", line 1055, in create_table
    chunkshape=chunkshape, byteorder=byteorder)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/tables/table.py", line 808, in __init__
    descr_from_dtype(nparray.dtype)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/tables/description.py", line 786, in descr_from_dtype
    for name in dtype_.names:
TypeError: 'NoneType' object is not iterable
```

I originally thought this was due to potentially having a low pytables version, and we need at least 3.2.x, but I actually have 3.4.2 and 3.6.1 on my cluster

**To Reproduce**
Steps to reproduce the behavior:
1. chain a localization rule push in pymeacquire
2. everything seems fine
3. wonder where your events are in the h5rs, see that they're in fact there in the pcs
4. See a very, very ocassional traceback in the dataservers

**Expected behavior**
copy the events over

**Additional context**
I'm running the cluster at #926, but suspicious that #922 changes are the cause

David-Baddeley, 2021-03-07:
I can't see anything in #922 which could give rise to an error with events. #926 is somewhat plausible (with the series potentially registering as complete too early and the pusher failing to copy events over). The extra directory caching in #926 would also be a potential issue (pusher sees files as complete, but uses an old directory listing and doesn't get the events file). 

The most likely culprit would probably be your`spoollocal` stuff as this directly touches the code related to event filing.

to clarify / narrow down:
- does this effect non-live localisation?
- was localisation started at the beginning or end of the series?
- is this using the spoollocal stuff?

barentine, 2021-03-07:
- will check non-live localization in a minute
- I tried both beginning and end of series localization, but again, will double check here in a minute
- yes this was using the spoollocal stuff

barentine, 2021-03-07:
Correct David that this does not effect non-live localization

barentine, 2021-03-07:
And correct again that this is a spoollocal issue (for both pushing at the start and end)

David-Baddeley, 2021-04-11:
should be fixed with #995 (which has most of #940 in it)

closed 2021-04-11 by David-Baddeley