python-microscopy/python-microscopy #933: [thing] deClump/pyDeClump.findClumps - missing doc, questionable gap parameter?

barentine, 2021-03-07:
**Describe the bug**
see tests PR'd in #932. deClump and pyDeClump have differing behavior, and neither seem to work with nFrame=0, (i.e. no gap, clumping on the same frame only)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to #932
2. pull or copy paste
4. See error

**Expected behavior**
I thought I could use this to group localizations that only appear on the same frame

David-Baddeley, 2021-03-07:
#932 is based on incorrect assumptions, but the solution is probably to write docstring(s)

The docstring should be roughly as follows.
```
findClumps(t, x, y, sig_x, n_frames)

Find repeated localisations of a single fluorophore, connecting localisations which are within 2*sig_x (i.e. a 95% confidence interval) of a localisation in the previous frame. Handles intermittency (rapid blinking etc ...) through the `n_frames` parameter. 

By making sig_x half the maximum movement of a particle from one frame to the next, the code can also be used for sparse particle tracking. 

Parameters
-----------

t : ndarray (int)
    Frame indices
x : ndarray (float)
    x positions
y: ndarray (float)
    y positions
sig_x: ndarray (float)
    localisation precision (std. deviation)
n_frames:  int
    Number of frames an emitter can be off for due to rapid blinking events and still be considered the same emitter. 
    n_frames=0 means that the emitter must have been on in the previous frame. n_frames=1 means that the emitter may be
    dark for one frame etc ....

Notes
------

findClumps **does not** link emitters within a single frame, even if they are within 2*sig_x as the fact that they were detected and fitted as two emitters makes it almost certain that they are 2 separate emitters. When events are so dense that multiple emitters in a single frame are within 2*sig_x, the forward linking will also be a bit dumb, and might link traces incorrectly. In practice this is only likely to occur in a single-particle tracking scenario where sig_x has been inflated to allow for potential emitter movement. In these cases, feature based tracking, which can use other object properties such as event intensity and/or shape to improve linkages is preferred.
```

barentine, 2021-03-08:
@David-Baddeley I think it'd be really helpful to have some more doc in deClump / pyDeClump (or just remove pyDeClump). When we wrote the multiview stuff 4 years ago you added some doc for me to findConnectedN which says nFrames=1 means look in current frame. Which I hope it does for findConnectedN even if it doesn't for findConnected. I can test/look at code/use what's there without knowing the intention unless we doc that module, which is tough for something 11+ years old

barentine, 2021-03-08:
OK, so `deClump.findClumpsN` is not linking the same frame either. This means that the `multiview.find_clumps` is broken and we currently use the biplane information in a very weird way where we link with gap 1 (between planes or not) but not on the same frame between planes

barentine, 2021-03-08:
Meeting with David:

- nFrames, the gap parameter, is not super intuitive and should probably be offset by 2 such that nFrames -1 links on the same frame only, 0 links adjacent frames only, and 1 allows a gap of 1 frame, etc.
- findClumpsN with nFrames 1 as is does work for same frame, but not for the first 2 frames

Action items:
- [ ] look at everywhere findClumps and findClumpsN are used, might be that the recursive version (findClumps) can be removed entirely
- [ ] change nFrames to be more intuitive (normalize it between findClumps and findClumpsN, as well as between pyDeClump and deClump versions which also differ at the moment)
- [ ] document + unit test
- [ ] fix to work on first two frames as well

David-Baddeley, 2021-07-13:
Looks like we might be able to close with #946. Can re-open if still a problem.

closed 2021-07-13 by David-Baddeley