python-microscopy/python-microscopy #935: [BUG] task max_duration getting exceeded following needStartExposure

barentine, 2021-03-07:
**Describe the bug**
Everything looks like a normal series. Only thing weird is the needStartExposure flag and then the task max_duration got exceeded (pausing the microscope)
```
DEBUG:PYME.Acquire.ActionManager:Executing action: (8.161508015700187, CentreROIOn: 53998.800000, 35522.470000 (x, y), 1.7976931348623157e+308, 300.0), CentreROIOn: 53998.800000, 35522.470000 (x, y)

DEBUG:root:Successfully marked rule as complete

DEBUG:root:ending polling loop.

DEBUG:PYME.Acquire.SpoolController:Looking for Bergamot/2021_3_5/EMC0106S2/5_3_series_CEX (.pcs or .h5) on cluster

DEBUG:PYME.Acquire.microscope:restarting framewrangler

DEBUG:PYME.Acquire.microscope:restarted framewrangler

DEBUG:PYME.Acquire.HTTPSpooler:Starting spooling: Bergamot/2021_3_5/EMC0106S2/B5/5_3_series_CEX.pcs

INFO:PYME.Acquire.htsms.rule_ui:protocol name : htsms-staggered

INFO:PYME.cluster.rules:Results file: Bergamot/2021_3_5/EMC0106S2/B5/analysis/5_3_series_CEX.h5r

DEBUG:PYME.cluster.rules:local_queues: []

INFO:PYME.cluster.rules:no local rule server, choosing one at random

DEBUG:root:resultsURI: http://10.150.7.3:15348/__aggregate_h5r/Bergamot/2021_3_5/EMC0106S2/B5/analysis/5_3_series_CEX.h5r

DEBUG:root:fileFormattedResults - URI: http://10.150.7.3:15348/__aggregate_h5r/Bergamot/2021_3_5/EMC0106S2/B5/analysis/5_3_series_CEX.h5r/MetaData

DEBUG:PYME.cluster.rules:Successfully created rule

DEBUG:root:task pusher poll loop started

DEBUG:root:numTotalFrames: 778, _next_release_start: 0

DEBUG:root:we have unpublished frames - push them

DEBUG:root:Successfully released tasks (0:778)

WARNING:PYME.Acquire.frameWrangler:Not keeping up with camera, giving up with 1 frames still in buffer

DEBUG:root:numTotalFrames: 1605, _next_release_start: 778

DEBUG:root:we have unpublished frames - push them

DEBUG:root:Successfully released tasks (778:1605)

DEBUG:root:numTotalFrames: 2435, _next_release_start: 1605

DEBUG:root:we have unpublished frames - push them

DEBUG:root:Successfully released tasks (1605:2435)

DEBUG:root:numTotalFrames: 3259, _next_release_start: 2435

DEBUG:root:we have unpublished frames - push them

DEBUG:root:Successfully released tasks (2435:3259)

DEBUG:root:numTotalFrames: 4089, _next_release_start: 3259

DEBUG:root:we have unpublished frames - push them

DEBUG:root:Successfully released tasks (3259:4089)

DEBUG:root:numTotalFrames: 4916, _next_release_start: 4089

DEBUG:root:we have unpublished frames - push them

DEBUG:root:Successfully released tasks (4089:4916)
[... 241 lines cut ...]
DEBUG:PYME.cluster.rules:Data complete, copying events to output file

WARNING:root:No handler for data type found, using pickle

DEBUG:root:fileFormattedResults - URI: http://10.150.7.3:15348/__aggregate_h5r/Bergamot/2021_3_5/EMC0106S2/B5/analysis/5_3_series_CEX.h5r/Events

DEBUG:PYME.cluster.rules:input data complete

DEBUG:root:numTotalFrames: 35000, _next_release_start: 35000

DEBUG:PYME.cluster.rules:all tasks pushed, marking rule as complete

DEBUG:root:Successfully marked rule as complete

DEBUG:root:ending polling loop.

DEBUG:PYME.Acquire.frameWrangler:Setting needStartExposure flag

ERROR:PYME.Acquire.ActionManager:task exceeded specified max duration

```

**To Reproduce**
Steps to reproduce the behavior:
1. Acquire couple thousand series and maybe you'll get unlucky

barentine, 2021-03-07:
Worth noting that the camera was running fine in continuous mode when I found the scope paused. Unpausing the queue started everything back fine

David-Baddeley, 2021-03-07:
To be able to usefully debug this we probably need to put a bit more info in the `task exceeded max duration` error message.

barentine, 2021-03-07:
I would have thought where we need more info is in why it took >200 s to purge the buffer.

David-Baddeley, 2021-03-07:
I guess what I was getting at was knowing if the task that expired is actually the current series.

barentine, 2021-03-07:
ah, as opposed to the next one? Worth noting I found it this way multiple times yesterday/this morning, and the camera wasn't actually bricked - it was running in continuous mode such that simply unpausing the queue got us going again. Not sure immediately if knowing that helps us or not, but regardless you're right that it'd be nice to print the task when we have to kill it like that (as well as send me an email)

David-Baddeley, 2021-03-07:
I have a hunch as to what might be causing it:

- spool series
- not quite keeping up, so pause at end of series whilst writing finishes (join the writing threads)
- not reading out the camera during this time, so trigger a buffer overflow (despite the fact that we've essentially disconnected everything)
- start a new series between via the action manager between the need start exposure flag getting set and it actually being processed (or a similar low probability race condition)

David-Baddeley, 2021-03-07:
The other thing that might help is adding timestamps to the logging calls

barentine, 2021-03-07:
That seems plausible. I'm not running with `httpspooler-jointhreads: False` config at the moment, but maybe I should switch back to it to make the transitions a bit faster again.

Forgot to come back and add that in the terminal when this happens I also get plenty of hdfspoolframe keyerroring on the cluster status

```
Traceback (most recent call last):
  File "c:\Users\Bergamot\Anaconda2\envs\chaos\lib\site-packages\wx\core.py", line 3259, in <lambda>
    lambda event: event.callable(*event.args, **event.kw) )
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\ui\HDFSpoolFrame.py", line 543, in Tick
    self.UpdateFreeSpace()
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\ui\HDFSpoolFrame.py", line 387, in UpdateFreeSpace
    freeGB = self.spoolController.get_free_space()
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\SpoolController.py", line 286, in get_free_space
    free_storage = sum([n['Disk']['free'] for n in nodes])
  File "c:\users\bergamot\code\python-microscopy\PYME\Acquire\SpoolController.py", line 286, in <listcomp>
    free_storage = sum([n['Disk']['free'] for n in nodes])
KeyError: 'Disk'

```

Another thing to note is that at the moment in spoolcontroller if we are spooling to a subdirectory we check exists in the wrong directory
```
DEBUG:PYME.Acquire.SpoolController:Looking for Bergamot/2021_3_5/EMC0106S2/5_3_series_CEX (.pcs or .h5) on cluster

DEBUG:PYME.Acquire.microscope:restarting framewrangler

DEBUG:PYME.Acquire.microscope:restarted framewrangler

DEBUG:PYME.Acquire.HTTPSpooler:Starting spooling: Bergamot/2021_3_5/EMC0106S2/B5/5_3_series_CEX.pcs
```

barentine, 2021-03-10:
Pushing rules in a separate thread #940 and not joining spooler threads stricly on transition makes this all 'go away'. Tempted to close, but will leave open for now as it'd be nice to track down/fix the underlying race condition despite it no longer being an urgent priority