python-microscopy/python-microscopy #951: [SUPPORT] I am confused by Z-Stack vs. Z Stepped in PYMEAcquire

zacsimile, 2021-03-12:
[image]

What is the difference? When would I use one over the other?

David-Baddeley, 2021-03-12:
### Z-Stack

Conventional, Z-Stack as you'd perform on a normal widefield/confocal microscope. 1 frame per step, synchronous*. 

Use for taking 3D Widefield stacks for deconvolution, bead measurements, etc ...

(*) depending on camera and piezo settings, but that is the aim.

### Z-Stepped Blinking/STORM  series

Step through multiple z planes whilst taking a PALM/STORM/PAINT acquisition.  Stay at each z level for a given dwell time (which is not strictly deterministic, and should normally be on the order of 100 frames or more). Typically cycle multiple times within the PALM/STORM series (cycling faster than bleaching). 

Steps occurs within approximately [-0, +100] ms of when they are nominally supposed to (but this is not strictly bounded). Time at which steps actually taken recorded in *events*, and z position of a given frame reconstructed from events after acquisition.

### More subtle differences

* Z-Stack records to memory, and needs to be manually saved
* Z-Stepped series are spooled to disk
* Only Z-Stepped series can currently be automated with the `ActionManager`

Allowing Z-Stacks to be saved directly to disk / and automated is on the TODO list.

zacsimile, 2021-03-12:
Thanks, @David-Baddeley. 

I can get a bead stack from Z Stepped, though, right? There's a resampling function that lets me use that. It seems like right now, especially since saving is kind of difficult, that the Z-Stack option isn't something I want to use. Is it worth folding that into the Z Stepped protocol and just having a "synchronous" flag for when we want to do that?

David-Baddeley, 2021-03-14:
You can sort of abuse z-stepped, to take stacks but shouldn't. It makes things needlessly complicated and introduces lots of potential opportunities for failure. It also makes taking z-stacks significantly slower (you need long integration times and/or several frames per step), is likely to be less than optimal in terms of bleaching and photodamage (because you have a non-constant number of frames per step, you need to expose for longer on average to ensure that the step with the lowest number of frames is adequately exposed) , potentially messes up the pixel noise / photometric calibration (because each frame in the stack is an average of a different number of raw frames), and is more prone to drift.

There is some resampling logic to get a stack from a z-stepped series, but it's not particularly robust, reliable, or appropriate for taking bead stacks. Think of it more as a way to rescue existing data acquired as a z-stepped series, or to create an average image of a PALM/STORM series rather than something you should be using regularly.

It's super simple to save a z stack via `File->Save` as you would any other image. The only limitations are that the stack must fit in memory and you need to manually start it (which you'd be doing for bead stacks in any case). 

Protocols are inherently asynchronous - you can't just turn on a synchronous flag and call things good. We have occasionally hacked them to do semi-synchronous things (e.g. tiling),  but this is typically a bad idea (or at best an interim solution).

I don't really understand the objection to using the z-stack functionality - it does what you want, and should do so reliably.

zacsimile, 2021-03-14:
I don't have an objection to it, I think it's confusing to have two z-stacking options on top of one another in the GUI and I wondered if there was a way to fold them into a single panel with a "synchronous" flag so it's a bit clearer. Alternatively, we might slap the word "Synchronous" in front of "Z-Stack" for a bit more distinction. It would also be nice to move what you've written above to the docs.

Re: the resampling function, you told me to use it to get a bead stack at one point (that's why I know about it). I didn't know that it's not robust, reliable or appropriate for this task generally, must have been a special case.

David-Baddeley, 2021-03-14:
Would the GUI be clearer if the blinking series panel was hidden to start with and/or had a different name? The design rationale is that you choose the type of acquisition (Z-Stack, PALM/STORM, Tiled, etc ...) you want and the panel that opens has all the relevant controls for doing that type of acquisition and nothing else. At the moment, we have the PALM/STORM panel open at startup, because that is by far the most used acquisition mode, but I can definitely see a clarity argument (particularly for new users) for having it hidden by default and actually presenting the "what do you want to do" question rather than assuming a default.

We could also look at changing it from the current folding panels for each acquisition type to, e.g., having a series of radio buttons at the top.

David-Baddeley, 2021-03-14:
Another option would be to split the acquisition types further ... e.g.

'z-stack', 'blinking series', 'blinking series with z stepping', etc ... Potentially also put the snapshot mode in there ...

David-Baddeley, 2021-03-14:
This is a mock up of what I had in mind ...

[image]

zacsimile, 2021-03-14:
I like the radio buttons (maybe even a dropdown? or something in the menu where you can load/unload modules like in pymeimage?). I don't think we need to split the acquisition types out further.

closed 2022-12-14 by David-Baddeley