python-microscopy/python-microscopy #952: [SUPPORT] plt.show() issue with current wx and py3 in PYME apps (may have nothing to do with py3)

csoeller, 2021-03-13:
**What are you trying to do?**

I am using some of the pyme menu functions/commands that include making a matplotlib plot in a window. These now give me some grief as follows:

When I run such a command and then delete the plot window that was created and run another command again that includes making a plot, `plt.show()` throws an error.

This is a shell command sequence in the PYMEVis shell tab that reliable reproduces the error for me:

```python
PYMEVisualize - note that help, license, etc. below is for Python, not PYME

Python 3.7.6 (default, Jan  8 2020, 13:42:34) 
[Clang 4.0.1 (tags/RELEASE_401/final)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
latest PYME version tag - 21.03.10
latest PYME version is post July 2020
from pylab import *
from PYME.DSView.dsviewer import View3D
plt.figure()
<Figure size 640x480 with 0 Axes>
plt.show()

# now I close the matplotlib figure window that was opened in the previous call
# from then I get errors when using plt.show() as below

plt.figure()
<Figure size 640x480 with 0 Axes>
plt.show()
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Users/csoe002/opt/anaconda2/envs/pyme-py3/lib/python3.7/site-packages/matplotlib/pyplot.py", line 353, in show
    return _backend_mod.show(*args, **kwargs)
  File "/Users/csoe002/opt/anaconda2/envs/pyme-py3/lib/python3.7/site-packages/matplotlib/backend_bases.py", line 3524, in show
    manager.show()  # Emits a warning for non-interactive backend.
  File "/Users/csoe002/opt/anaconda2/envs/pyme-py3/lib/python3.7/site-packages/matplotlib/backends/backend_wx.py", line 1051, in show
    self.frame.Show()
RuntimeError: wrapped C/C++ object of type FigureFrameWxAgg has been deleted
```

### Package Versions
python-microscopy=21.03.10
python=3.7.6, platform=darwin
numpy=1.19.2, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Any pointers appreciated.

csoeller, 2021-03-13:
The below seems an ok workaround on macOS and hopefully ok on win as well but there may be a better fix?

```python
def safepltshow():
    try:
         plt.show()
    except RuntimeError:
         pass
```

David-Baddeley, 2021-03-13:
This smells like an issue with matplotlib (would be good to know the
matplotlib version). Could also be mixing interactive and non-interactive
modes (in general, calling plt.show() in interactive mode, such as from the
console in Visgui, is not well defined/necessary.

Most of the menu functions which call show turn interactive mode off, do
the plotting, call show, and turn it back on. It is possible that switching
back and forth has changed in more recent matplotlib.

zacsimile, 2021-03-14:
This is indeed a `matplotlib` error, and maybe one we should file with them. For now, downgrading to `matplotlib=3.2.2` should fix it.

csoeller, 2021-03-14:
Hi @zacsimile, I can confirm I am also on 3.3.2:

```
# Name                    Version                   Build  Channel
matplotlib                3.3.2                         0  
matplotlib-base           3.3.2            py37h181983e_0  
```

Furthermore, no difference if in interactive mode or not (tried with ion and ioff to manipulate mode). I may try downgrading to 3.2.2 and see how that goes.

@David-Baddeley, if there is indeed a trick how to avoid this issue in PYME core code even with matplotlib 3.3.2 I have not yet found it, couldn't see any manipulation of interactive state in the modules I checked that had `plt.show()` in them.

csoeller, 2021-03-14:
I can confirm all fine after `conda install matplotlib=3.2.2`. I imagine this issue may be macOS specific?

David-Baddeley, 2021-03-15:
Do we need to add a pin in pyme-depends?

zacsimile, 2021-03-16:
Probably the easiest solution for now.

csoeller, 2021-03-27:
Can we close this one or do we need as reminder for adding a pin in pyme-depends?

closed 2021-03-29 by David-Baddeley