python-microscopy/python-microscopy #955: [BUG] Default file types for pymeimage on win10 do not include PNG

zacsimile, 2021-03-14:
See screenshots.

[image]

[image]

PNG/JPG does show up by default on Mac.

zacsimile, 2021-03-15:
Looking at the code, this is absolutely what's supposed to happen. Mac apparently has the incorrect default behavior. Do we want to change this or leave as is? I'd argue that we want to display the option to open other file types by default if `bioformats` is installed. Alternatively, we could rename "Image Data" to "PYME image data" or something like this? What we consider image data by default (`*.h5;*.tif;*.lsm;*.kdf;*.md;*.psf;*.npy;*.dbl`) is a bit hodgepodge.

David-Baddeley, 2021-03-15:
This is a bit of a tricky one ... we could definitely update the list of supported types (e.g. remove .kdf), but I don't think .png or .jpg should be on the default extensions list as they are compressed (generally lossy) and not suitable for biological imaging data. 

Under windows we do have the option of opening anything (through the 'All Files' selector), and it probably does make sense to show the well supported file types under a different selector from those that fall into the *we might be able to open it with bioformats but results will vary* category. 

For context, OSX doesn't really have a concept of different selectors, so concatenates them all - i.e. you see the results of both the 'Image Files' and 'All Files' selectors.

David-Baddeley, 2021-03-15:
I could also entertain the idea of 3 selectors (supported natively, through bioformats, and try your luck [everything else - e.g. .png, .jpg, etc ...]), although this would require us to track bioformats supported extensions.

zacsimile, 2021-03-16:
Yeah I'm not super keen on tracking bioformats extensions. I think the slightly confusing thing is there are a lot of image types we can open that don't fall under what we call "Image Files". If we could change that name to something slightly narrower I think it will no longer be confusing.