python-microscopy/python-microscopy #956: [BUG] temporary file deleted in file request to cluster

adrianhollow, 2021-03-15:
**Describe the bug**
Downloading pcs series from ClusterOfOne as h5 fails as in #746, but with this error message:
```
File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/site-packages/django/core/handlers/exception.py" in inner
  34.             response = get_response(request)

File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/site-packages/django/core/handlers/base.py" in _get_response
  126.                 response = self.process_exception_by_middleware(e, request)

File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/site-packages/django/core/handlers/base.py" in _get_response
  124.                 response = wrapped_callback(request, *callback_args, **callback_kwargs)

File "/home/tphan/thesis/python-microscopy/PYME/cluster/clusterUI/clusterbrowser/views.py" in file
  43.             response['Content-Length'] = os.path.getsize(outf.name)

File "/home/tphan/miniconda3/envs/mypyme/lib/python3.7/genericpath.py" in getsize
  50.     return os.stat(filename).st_size

Exception Type: FileNotFoundError at /files/download/tphan/2021_3_15/15_3_series_A.pcs/
Exception Value: [Errno 2] No such file or directory: '/tmp/tmphwec6l04.h5'
```
**To Reproduce**
Steps to reproduce the behavior:
1. Run `PYMEClusterOfOne --clusterUI`
2. Navigate to pcs series in clusterUI
3. try to download as h5
4. see error

**Expected behavior**
 download the pcs as an h5

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: ubuntu 20.04
 - Python version: 3.7.9
 - numpy version: 1.19.2
 - wxpython version [if a UI bug]: 4.0.4

David-Baddeley, 2021-03-16:
This is super odd - the temporary file should be open as long as the `with ...` context manager is open. This looks like a bug/regression in `NamedTemporaryFile` (or `os.stat`)

David-Baddeley, 2021-03-16:
Do we also get this error on py3.6?

David-Baddeley, 2021-03-16:
What about other platforms?

David-Baddeley, 2021-03-16:
I suspect the solution to this might have to involve writing our own replacement for `NamedTemporaryFile`, using our own filename within a temporary directory as `NamedTemporaryFile` is even more broken on Windows (I though there was an issue for this but can't seem to find it).

PR #1015 (2021-04-18): Don't use context manager on temporary files (issue #956)
We return the file object (or technically a wrapper of it), making closing the temporary file via the context manager a bad idea. Rely on the FileWrapper and HttpResponse objects to close the file for us instead.\n Tested on OSX, py37, but not yet on linux.\n Fixes #956, replaces and closes #957
closed 2021-04-18 by David-Baddeley