python-microscopy/python-microscopy #958: [ENHANCEMENT] PYMEAcquire plugin architecture

zacsimile, 2021-03-15:
**Is your feature request related to a problem? Please describe.**
I want to create a configuration GUI for some liquid crystals that are used in an imaging protocol. There's no reason this should be a part of PYME's core, but I'd like to be able to access it from `PYMEAcquire` to configure at the beginning of each imaging session. I will want to interact with the camera and see a live image during the configuration process.

**Additional context**
I took a brief look at the code I suspect we can almost do this already. Is adding the following to `PYME.Acquire.acquiremainframe.PYMEMainFrame._create_menu()` sufficient?

```
for mn in config.get_plugins('pymeacquire'):
        try:
            m = __import__(mn, fromlist=mn.split('.')[:-1])
            m.Plug(visFr)
        except Exception as e:
            logger.exception('Error loading plugin: %s' % mn)
```

**Are you prepared to work on it?**
Yes.

David-Baddeley, 2021-03-15:
Because PYMEAcquire uses python scripts for its init files, there is not
really any need for a specific plugin architecture (one might argue that it
even preferable for all the config to be in the one place - ie the init
file).

You should be able to simply add an initgui section to your init file that
imports your module and calls mainframe.AddMenuItem().

zacsimile, 2021-03-15:
OK. Where do I stash the module?

David-Baddeley, 2021-03-15:
Short answer - wherever makes sense to you - because you are importing it within the init script, it can be anywhere on the python path. Other than making sure that it's importable, I don't think there's a right answer.

zacsimile, 2021-03-16:
OK. I'm going to close this since it seems the plugin architecture for PYMEAcquire already exists in the format we want to use.

zacsimile, 2021-03-16:
Yeah the documentation needs some work: https://python-microscopy.org/doc/ModifyingPYMEAcquire.html

csoeller, 2021-03-16:
Ok, we probably should start a pull request against that part of the docs. Note that I have an example how to calibrate cameras using this approach, based on suggestions from David and @barentine, in https://github.com/csoeller/PYME-exeter-siteconfig/blob/master/init_scripts/generic/init_Zyla_n.py. This has ended up working really nicely (modulo still having to write analysis code for the generated data sets):

```python
# this tests using the action manager to queue actions
# makes these available as menu entries
@init_gui('ROI Calibration')
def roi_calibration(MainFrame, scope):

    def roi_action_callback(event=None):
        from PYMEcs.Acquire.Actions.custom import queue_roi_series
        queue_roi_series(scope)

    def camera_chip_calibration_callback(event=None):
        from PYMEcs.Acquire.Actions.custom import camera_chip_calibration_series
        camera_chip_calibration_series(scope)

    # ToDo - add help strings
    MainFrame.AddMenuItem('Calibration', 'Camera Maps>Test ROI actions', roi_action_callback)
    MainFrame.AddMenuItem('Calibration', 'Camera Maps>Tile over Chip ROIs', camera_chip_calibration_callback)
```

zacsimile, 2021-03-24:
Reopening this issue to ask a few more things:

1. Where do I place protocols that I want to use that aren't in `PYME.Acquire.Protocols`? Should the PYME plugin template include an `/etc/PYME/protocols` the way @csoeller's PYME-exeter-siteconfig structure does?
2. Can I make PYMEAcquire launch with a specific protocol by default (e.g. I want `init_oidic.py` to always launch with my OIDIC acquisition protocol selected)?
3. Related to 2, can I eliminate other protocol options from the PYMEAcquire GUI in my init script (e.g. can I remove Z-Stack from the GUI entirely when I launch `init_oidic.py`, re: #951)?

David-Baddeley, 2021-03-24:
1) - look at `PYME.config` - we look for protocols in .PYME/protocols, /etc/PYME/protocols etc ... and init scripts in .PYME/init_scripts etc ...
2) not particularly cleanly, although you could call `scope.spoolContoller.SetProtocol(...)` somewhere in your init script.
3) Z-stacks are not protocols, but rather an acquisition mode. Protocols are somewhat specific to PALM/STORM series (or, more generally, for stuff which does not require good sychronisation). Better "skinning" support, to allow more customisation of what is displayed is on the TODO list (see also #516). That said, hiding z-stacks is a bad idea on most systems as it prevents you from obtaining PSF calibrations.

Z-stacks vs protocols aside, there is a lot of clutter in the current default shipped protocols, and this could be rationalised.

As I understand it, OIDIC requires the acquisition of multiple separate polarisation images at each z-level. Ultimately I suspect it needs it's own acquisition mode, rather than somehow being shoe-horned into a protocol. I really need to write some more docs on what a protocol is and how to use them. The short of it is:
- protocols were initially envisaged as a (relatively) simple way to allow a *user* to specify a custom sequence of laser intensity changes during the course of a PALM/STORM acquisition
- by choosing to make them python code, they ended up being way too flexible. They also ended up being a bit too complex for the average user (technically they are supposed to mitigate this by being a very small subset of python - or specifically just a textual list of tasks whose syntax just happens to match the python syntax and can be read by the python parser. If implementing again, I'd likely go for yaml instead)
- the flexibility and power of actually being python code makes it exceptionally tempting to (ab)use them to do other tasks, and I'm probably the chief culprit here.
- there are, however good reasons **not** to use them for more generic tasks. These are as follows:
    - timing is rough - specifying that a task should execute on frame N actually means 'some indeterminate time after frame N'
    - we rely on events to determine what happened when. These are a) not guaranteed to be present for all possible protocol tasks, b) only available after completion of the series, and c) complicated (and hence potentially fragile) to reconstruct and line up with frames.
    - The event reliance means that analysis which depends on protocol tasks can only be performed **after** series acquisition is complete, putting significant constraints on real-time analysis (we get away with it in PALM/STORM as the principle analysis doesn't really depend on what is happening in the protocol.
    - Because they are so flexible, and lack defined structures and parameterisation, it's very easy to write a protocol that does not record all the required metadata for analysis, or to base the analysis on assumptions that might not be valid.
    - Protocols are expected to be user-editable - this means that analysis shouldn't depend on a protocol being executed as written. There is currently no explicit versioning of protocols written to the metadata (adding the protocol text to the acquisition metadata is on the TODO list), which makes this even worse.

You can hack around all the protocol limitations, but it doesn't result in robust, easily maintainable code. For new methodologies, e.g. tiling or OIDIC, writing a custom acquisition mode is going to be much more robust, as this allows you to choose a consistent parameterisation, metadata scheme, and UI for the user to set parameters. An initial protocol based implementation of a new methodology is OK / useful to get a feel for what parameters etc are important and should be presented to the user, but should ideally not persist to the point that you have anyone else using it.

zacsimile, 2021-03-28:
A new `XYZTCAcquisition` class was added in #972, which addresses a number of my concerns. We'll keep beta-testing, but it's now much easier to add a new imaging mode to PYME. We should document the difference between protocols (existing imaging mode, time-based, user-specified) vs. acquisitions (new imaging modes, coders only) somewhere. We should also document `ZTaskListProtocol`. I thought it ran each protocol once per z-step, but it in fact runs a z-stack on top of a protocol that only executes at certain times throughout the acquisition. 

I would like to be able to do things post-init script. Namely, I'd like to collapse `self.pan_spool`, which isn't added to the GUI until post init (`PYME.Acquire.acquiremainfame.PYMEMainFrame.doPostInit`). Is there a way to write post-init hooks into the init script? Can we add a way?

zacsimile, 2021-03-28:
Follow up: much like `PYME.Acquire.ui.seqdialog.seqPanel` controls `stackSettings`, I think there should be an importable panel to start the `XYZTCAcquisiton` with either memory to cluster backends. Probably this means factoring `_spool_pan` out of `PYME.Acquire.ui.HDFSpoolFrame.PanSpool` into its own panel. Seem reasonable, @David-Baddeley?

David-Baddeley, 2021-03-29:
@zacsimile - Re: `PanSpool` - we should probably think about what the backend/model should look like before messing too much with the GUI/view. Ultimately we will likely end up doing some refactoring of `PanSpool`, but it's possible that it will make more sense to start from scratch / do a significant re-write of the UI. 

Re collapsing panels on startup - there is a draft PYMEAcquire skinning spec in the comments on #373. It would be reasonably simple to, e.g. add initial fold and pin states to the entries in the layout dict.  Ideally I'd like the acquisition types panel to behave like the channels panel in PYMEVis, with only one acquisition mode open at once and all the others folded (sort of an equivalent to the webui radio-box). In this case, a `default_acquisition_mode` option in the layout dict might be preferable to specifying each panels initial fold state.