python-microscopy/python-microscopy #969: [ENHANCEMENT] some plugin docs

csoeller, 2021-03-25:
**Is your feature request related to a problem? Please describe.**

We probably need some info/examples on plugin strategies. This was touched on in #958.

It has become clear that we need a bit of info how to do

- a visgui plugin
- a dh5view plugin
- a pymeacquire plugin

**Describe the solution you'd like**

Rather than having long diatribes I would suggest adding stub templates according to current wisdom to [pyme-plugin](https://github.com/python-microscopy/pyme-plugin), including potentially adding an example init file with a plugin example for PYMEAcquire and associated install example.

**Current stub ideas**

Here are a couple of stubs that could do with David's input for improvement where necessary, e.g. should some of these be derived from the plugin base class etc.

dh5view_template_plugin.py:

```python
class MyPlugin:
    def __init__(self, dsviewer):
        self.dsviewer = dsviewer

        dsviewer.AddMenuItem('Experimental>MyMenuSection', "Do something", self.OnDoSomething)

    def OnDoSomething(self, event):
        pass

def Plug(dsviewer):
    """Plugs this module into the gui"""
    MyPlugin(dsviewer)
```

visgui_template_plugin.py:

```python
# import numpy as np

class MyPlugin:
    def __init__(self, visFr):
        self.visFr = visFr

        visFr.AddMenuItem('Experimental>MyMenuSection', "Do something", self.OnDoSomething)

    def OnDoSomething(self, event):
        pass

def Plug(visFr):
    """Plugs this module into the gui"""
    MyPlugin(visFr)
```

**Are you prepared to work on it?**
Sure.

csoeller, 2021-03-27:
I am moving this discussion to [pyme-plugin](https://github.com/python-microscopy/pyme-plugin).

David-Baddeley, 2021-03-28:
Docs live in the main repo, so documentation enhancements probably belong here - moving it to pyme-plugin is a good way to get it forgotten about (I regularly revisit issues here, other modules less so).

csoeller, 2021-03-29:
Ok, will go with that. Note that I would want to have these stubs as actual demo files somewhere rather than in doc only. In doc only they are not testable and breaks go unnoticed when interfaces change etc. Any good idea how this could be done in this dist?

David-Baddeley, 2021-03-29:
I might have misread a bit initially - I think having examples as code snippets is reasonable, but there also needs to be some formal docs. The formal docs obviously need to go here. It's a hard call how much of the contextual stuff be in the formal docs, or as comments in the code snippets. In terms of concrete stuff.

- if you are creating a class, it should derive from the `Plugin` class (the main reason for this is to avoid keeping references to the main frame object in the class, which creates circular references and causes problems on exit - the Plugin class solves this by using weakrefs)
- you should/must not keep a reference to `visFr`/`dsviewer` (or any attributes thereof) as this will create a circular reference
- you only need a plugin class if the plugin needs to record state - for stateless plugins (see, e.g., the new dsviewer `projection` module), it's fine to simply bind menu items to functions in the `Plug()` method.

barentine, 2023-06-21:
One to add to needs for formal docs:
the hasstricttraits stuff for recipe modulebase changes 6 months ago are good to know about! Currently breaking nep-fitting

closed 2023-12-09 by csoeller