python-microscopy/python-microscopy #970: [BUG] Fit Info panel does not work properly?

csoeller, 2021-03-25:
**Describe the bug**
In dh5view I run a `Test this frame` command and immediately get the following error in the console, Not sure if relevant to the problem at hand though.

```
Traceback (most recent call last):

  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/contrib/wxPlotPanel.py", line 69, in _onIdle

    self._SetSize()

  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/contrib/wxPlotPanel.py", line 86, in _SetSize

    float( pixels[1] )/self.figure.get_dpi() )

  File "/Users/csoe002/opt/anaconda2/envs/pyme-py3/lib/python3.7/site-packages/matplotlib/figure.py", line 902, in set_size_inches

    raise ValueError(f'figure size must be positive finite not {size}')

ValueError: figure size must be positive finite not [0.   6.86]

```

**To Reproduce**
Steps to reproduce the behavior:
1. `Test this frame`
2. Click on event that was detected
3. go to `Fit Info` panel
4. ROI of event shows up for a second but then all plot panels are blank.

**Expected behavior**
Would have expected to see event and some plots in `Fit Info` panel.

**Screenshots**

[image]

[image]

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: macos 10.15
 - Python version: py37
 - numpy version: ??
 - wxpython version [if a UI bug]: 4.X

**Additional context**
Add any other context about the problem here.

David-Baddeley, 2021-03-26:
Hmm, which matplotlib version? 

Best guess is that matplotlib has got fussier with what it accepts in `set_size_inches()`, causing it to bail whilst we are creating the plot panel (before the plot panel has been resized). Best guess at a fix / workaround.

- comment out `self._SetSize()` on line 41
- optionally change`self._resizeflag=True` a couple of lines below 

Together these should force size computation to be deferred to `OnIdle()` (I suspect we might not strictly need to set `_resizeflag` as we might get an `EVT_SIZE` when we add the panel to the containing sizer). This has 2 effects - it stops `__init__` from bailing and it means that, even if it does get fail before the window is at full size, it will et run again when the window is resized (EVT_SIZE and EVT_IDLE should already be bound before the failure).

csoeller, 2021-03-27:
The `set_size_inches()` error turned out to be a red herring.

For completeness,

- commenting out `self._SetSize()` on line 41 made no change to the error
- changing `self._resizeflag=True` silenced the error with no observed effect on functionality

Actual issue addressed in PR #973.

David-Baddeley, 2021-03-29:
Cool, thanks!

closed 2021-03-29 by David-Baddeley