python-microscopy/python-microscopy #971: [ENHANCEMENT] figure making streamlining / docs

barentine, 2021-03-25:
**Is your feature request related to a problem? Please describe.**
SMLM dataset -> 3D multicolor image suitable for a figure should be something we do easily/make accesible.

At the moment, with projection from the GUI only working for xy and otherwise just showing a slice (see #811) I feel a little ridiculous with the process:

first running this recipe on my localization files

```
- localisations.DensityMapping:
    colours:
    - chan0
    - chan1
    inputLocalizations: fitResults
    jitterScale: 30.0
    manualXYBounds:
    - -50.0
    - 1100.0
    - 27800.0
    - 27800.0
    outputImage: g2d
    pixelSize: 15.0
    renderingModule: Gaussian
    xyBoundsMode: manual
- output.RGBImageOutput:
    colorblindFriendly: true
    filePattern: '{output_dir}/{file_stub}.png'
    inputName: g2d
    scaling: percentile
- output.ImageOutput:
    filePattern: '{output_dir}/{file_stub}_g2d.h5'
    inputName: g2d
- localisations.DensityMapping:
    colours:
    - chan0
    - chan1
    inputLocalizations: fitResults
    jitterScale: 30.0
    jitterScaleZ: 30.0
    manualXYBounds:
    - -50.0
    - 1100.0
    - 27800.0
    - 27800.0
    outputImage: g3d
    pixelSize: 15.0
    xyBoundsMode: manual
    zBoundsMode: min-max
    zSliceThickness: 15.0
- output.ImageOutput:
    filePattern: '{output_dir}/{file_stub}_g3d.h5'
    inputName: g3d
```

and then
open _g2d.h5 in PYMEImage. In the shell toggle crosshairs and scale bar off
view.crosshairs=False
view.showScaleBar=False

set display to 1:2

chan0 LUT to ‘c’, histogram to [0, 0.13]
chan1 LUT to ‘m’, histogram to [0, 0.13]

save as _2d.png

select a box for a cross section and click File > Export Cropped
open the _g3d.h5 of that same ROI 
switch to the shell tab, and enter

```
from PYME.IO.image import ImageStack
im = ImageStack(data=np.swapaxes(image.data[:, 874:1049, :, :], 1, 2), mdh=image.mdh)
```

where 874: 1049 are the ‘y’ bounds I selected, and the 1, 2 correspond to swapping y and z to get an xz image with y slices.

Now again in the shell 
```
ViewIm3D(im)
```
to open up a viewer for the cropped part of the stack.

Save this as h5 just in case

set display aspect to 1:2
projection -> Standard
chan 0 LUT to ‘c’, histogram to [0, y slice number * 0.0036 / 119]
chan 1 LUT to ‘m’, histogram to [0, y slice number * 0.0036 / 119]

toggle off the crosshairs and scale bar again

save to png with y range cropped in the title

**Describe the solution you'd like**
make this shell-free, or at least document this kind of thing somewhere with a nice walk-through

David-Baddeley, 2021-03-26:
as of #972 we have a new 'projection' dsviewer module, which makes it a lot easier to do x, y projections etc (including cropped projections). Will probably remove the projection functionality from the live view once we have everything across (i.e. colour coded z, which is still missing).

barentine, 2021-04-10:
Phew - thanks @David-Baddeley! Took me coming back here to find it. We should get a SMLM -> 3D multicolor + z figure walkthrough up at some point