python-microscopy/python-microscopy #974: [BUG] issue reading legacy psf and sf files

csoeller, 2021-03-27:
**Describe the bug**
Newer numpy and py2 -> py3 presents issues reading older `.psf` and `.sf` files. 

**To Reproduce**
Steps to reproduce the behavior:
1. Try to open `.psf` or `.sf` file with newer numpy or py3 based pyme
2. hits either `allow_pickle = False` default error or py3 cannot read py2 generated pickle error

**Expected behavior**
Ideally should still be able to read these files.

**Proposed fix**
The  patch below to `PYME.IO.image` seems to remedy things.

**Questions**

1. Ok to use this approach in principle?
2. There are more instances of `numpy.load` in `PYME.IO.image`. Fix those in the same way?

**Suggested fix**

If ok can raise this as PR.

```patch
diff --git a/PYME/IO/image.py b/PYME/IO/image.py
index ff3b172a4..71c645da4 100755
--- a/PYME/IO/image.py
+++ b/PYME/IO/image.py
@@ -54,6 +54,17 @@ import warnings
 #Alias for backwards compatibility
 VS = MetaDataHandler.VoxelSize
 
+def np_load_legacy(f):
+    try:
+        ret = numpy.load(f)
+    except ValueError:
+        try:
+            ret = numpy.load(f, allow_pickle=True)
+        except OSError:
+            # this one based on tips from https://rebeccabilbro.github.io/convert-py2-pickles-to-py3/
+            ret = numpy.load(f, allow_pickle=True, encoding="latin1")
+    return ret
+
 class ImageBounds:
     def __init__(self, x0, y0, x1, y1, z0=0, z1=0):
         self.x0 = x0
@@ -527,7 +538,8 @@ class ImageStack(object):
         """
         from PYME.IO import unifiedIO
         with unifiedIO.local_or_temp_filename(filename) as fn:
-            self.data, vox = numpy.load(fn)
+            self.data, vox = np_load_legacy(fn)
+                
         self.mdh = MetaDataHandler.NestedClassMDHandler(MetaData.ConfocDefault)
 
         self.mdh.setEntry('voxelsize.x', vox.x)
@@ -543,7 +555,8 @@ class ImageStack(object):
     def _loadSF(self, filename):
         self.mdh =MetaDataHandler.NestedClassMDHandler( MetaData.BareBones)
         self.mdh.setEntry('chroma.ShiftFilename', filename)
-        dx, dy = numpy.load(filename)
+        dx, dy = np_load_legacy(filename)
+            
         self.mdh.setEntry('chroma.dx', dx)
         self.mdh.setEntry('chroma.dy', dy)
```         

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: macOS 10.15
 - Python version: 3.7
 - numpy version: 1.19
 - wxpython version [if a UI bug]: NA

David-Baddeley, 2021-03-29:
This seems like a very reasonable approach. As it stands there doesn't seem to be much value in loading with `allow_pickle=False` and then automatically re-trying with `allow_pickle=True`, unless we print log messages to indicate that we have hit a format irregularity.

- The other `numpy.load()` call in image.py is for loading .npy files. The basically exists for interfacing with other software using numpy and I don't think we've used them extensively, if at all. When used, there shouldn't be any pickled data so I don't expect there to be any backwards compatibility issues. I think it's probably better to err on the side of caution here and stick with the default `allow_pickle=False`.

- The latin1 fix is probably also applicable to metadata loading from old files, where we also have an outstanding bug (#713)

csoeller, 2021-03-31:
fixed by #979, still needs looking at implications for #713, but close for now.

closed 2021-03-31 by csoeller