python-microscopy/python-microscopy #975: [BUG] why is six afraid of import

barentine, 2021-03-27:
**Describe the bug**
789

```
 Error whilst running Recipes>Run Current Recipe and Save	Shift+F5
==============================================
python-microscopy=21.03.10
python=3.6.9, platform=linux
numpy=1.16.5, wx=4.0.4 gtk2 (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/home/smeagol/code/python-microscopy/PYME/ui/progress.py", line 121, in func
    return fcn(*args, **kwargs)
  File "/home/smeagol/code/python-microscopy/PYME/DSView/modules/recipes.py", line 88, in RunCurrentRecipeAndSave
    self.RunCurrentRecipe(saveResults=True)
  File "/home/smeagol/code/python-microscopy/PYME/DSView/modules/recipes.py", line 167, in RunCurrentRecipe
    elif isinstance(out_, six.string_types):
UnboundLocalError: local variable 'six' referenced before assignment
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10]
 - Python version:
 - numpy version:
 - wxpython version [if a UI bug]:

**Additional context**
Add any other context about the problem here.

zacsimile, 2021-03-28:
Looks to me like line 153 of `PYME.DSView.modules.recipes.py` is the problem. If the recipe only injects into the namespace, all return values will be `None`,  and you'll never get the `import six` call you need.

barentine, 2021-03-28:
Yep

David-Baddeley, 2021-03-28:
should be fixed with #977

closed 2021-03-28 by David-Baddeley