python-microscopy/python-microscopy #981: [BUG] Error when trying to chain together localizations in time

lukasfue, 2021-03-31:
**Describe the bug**
Error when trying to chain together localizations in time

**To Reproduce**
Steps to reproduce the behavior:
1. Load data
2. Corrections > Chaining > Find consecutive appearances > use default settings >OK
3. See error

**Expected behavior**
Localizations are chained together

 Error whilst running Corrections>Chaining>Find consecutive appearances
==============================================
python-microscopy=21.03.10
python=3.7.9, platform=win32
numpy=1.19.1, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "c:\users\laf62\python-microscopy\PYME\ui\progress.py", line 121, in func
    return fcn(*args, **kwargs)
  File "c:\users\laf62\python-microscopy\PYME\LMVis\Extras\particleTracking.py", line 79, in OnFindClumps
    clumpRadiusScale=dlg.GetClumpRadiusMultiplier()),])
  File "c:\users\laf62\python-microscopy\PYME\recipes\base.py", line 1002, in add_modules_and_execute
    self.execute()
  File "c:\users\laf62\python-microscopy\PYME\recipes\base.py", line 749, in execute
    m.execute(self.namespace)
  File "c:\users\laf62\python-microscopy\PYME\recipes\tracking.py", line 199, in execute
    self.timeWindow, minClumpSize=self.minClumpSize)
  File "c:\users\laf62\python-microscopy\PYME\Analysis\Tracking\trackUtils.py", line 381, in findTracks2
    clumpIndices[I] = deClump.findClumps(t[I], x[I], y[I], delta_x[I], nFrames)
  File "c:\users\laf62\python-microscopy\PYME\Analysis\points\DeClump\__init__.py", line 78, in findClumps
    return deClump.findClumps(t, x, y, delta_x, n_frames, link_within_frame)
TypeError: function takes at most 5 arguments (6 given)

barentine, 2021-03-31:
did you run `python setup.py develop` or `python setup.py install` after updating?

David-Baddeley, 2021-03-31:
Andrew beat me to it - if you have a source / development install and have recently updated from the repository you will need to recompile (`python setup.py develop` or `python setup.py build_ext -i`). I'd recommend against `python setup.py install` as this will mean that you no longer track repository updates (which was presumably the reason for using a development install in the first place). The recompile is needed as the c code in the declump module 
was changed recently and, because it's c-code,  rather than python it needs to be manually recompiled.

lukasfue, 2021-04-01:
Got it. Thanks for the quick response. Zach has definitely told me this several times before I just never think about it. Is there an easy way to know when it is necessary to recompile, or should I just get in the habit of doing this after every git pull just in case?

zacsimile, 2021-04-02:
When you `git pull`, it gives you a list of changed files. If any of those files end in `.pyx` or `.c`, you'll need to recompile.

lukasfue, 2021-04-05:
Okay, thanks for the insight.

David-Baddeley, 2021-05-04:
Guess we can close this. Would be good to think if there is any way of automatically letting you know if a recompile is needed, but if you are running a devel install, things like this go with the territory a bit. 

Short of it is, it's always safe to re-run `python setup.py develop` after pulling from the git repo.

closed 2021-05-04 by David-Baddeley