python-microscopy/python-microscopy #983: [BUG] LIFO behavior for localization tasks

barentine, 2021-04-04:
**Describe the bug**
not really sure if its a bug, but it is not what I expected

**To Reproduce**
Steps to reproduce the behavior:
0. add print statement in taskWorkerHTTP to print tIndex for each fittask
1. clusterUI `Localize Molecules`
2. watch tIndex start at end of series and monotonically step down by one
3. think that's kind of weird since all the queues in task distribution are FIFO, and task pushing is tIndex 0 on up

barentine, 2021-04-04:
OK, so this plays pretty weird, at least with the GPU fit. The following script will pass, but if you swap the order, e.g.
` for ind in range(mdh['Analysis.StartAt'], n_stop)[::-1]:` then it will fail
<details>  <summary> test script </summary>

```
import subprocess
from PYME.IO import clusterIO, MetaDataHandler
from PYME.localization import remFitBuf
import tempfile
import os
import shutil
import signal
import pytest
import time
import sys
import atexit
import logging
import queue
import threading
import numpy as np
import importlib
from PYME.IO import clusterResults

logger = logging.getLogger(__name__)
proc = None
tmp_root = None

def setup_module():
    global proc, tmp_root
    atexit.register(teardown_module)
    # tmp_root = os.path.join('/home/smeagol/Desktop', 'PYMEDataServer_TES1')
    tmp_root = os.path.join(tempfile.gettempdir(), 'PYMEDataServer_TES1')
    
    print('DataServer root: %s' % tmp_root)
    
    if os.path.exists(tmp_root):
        print('Removing existing temp spooler dir')
        shutil.rmtree(tmp_root)
        
    os.makedirs(tmp_root)
    proc = subprocess.Popen([sys.executable, '-m', 'PYME.cluster.HTTPDataServer',  '-r', tmp_root, '-f', 'TES1', '-a', 'local'])
    
    time.sleep(5) #give time for the server to spin up

    logging.info('Advertised services:\n------------------\n%s' % '\n'.join([str(s) for s in clusterIO.get_ns().get_advertised_services()]))
    
def teardown_module():
    global proc, tmp_root
    proc.send_signal(signal.SIGINT)
    time.sleep(5)
    proc.kill()

mdh = MetaDataHandler.DictMDHandler({
    'Analysis.BGRange': (-32, 0),
    'Analysis.DebounceRadius': 4,
    'Analysis.DetectionFilterSize': 4,
    'Analysis.DetectionThreshold': 1.0,
    'Analysis.FiducialROISize': 11,
    'Analysis.FiducialSize': 1000.0,
    'Analysis.FiducialThreshold': 1.8,
    'Analysis.FitModule': 'AstigGaussGPUFitFR',
    'Analysis.GPUPCTBackground': True,
    # 'Analysis.GPUPCTBackground': False,
    'Analysis.PCTBackground': 0.25,
    'Analysis.ROISize': 7.5,
    'Analysis.StartAt': 32,
    'Analysis.TrackFiducials': False,
    'Analysis.subtractBackground': True,
    "Camera.ADOffset": 101.753685,
    "Camera.Binning": [
[... 118 lines cut ...]
    "voxelsize.y": 0.115,
    "voxelsize.z": 0.6
})

series = '/media/smeagol/transfer_ssd/2021_2_17_EMC0105H_A1/17_2_series_BB.h5'
results_uri = 'PYME-CLUSTER://%s/%s' % ('TES1', 'series_bb_0.h5r')
aggregate_endpoint = 'pyme-cluster://TES1/__aggregate_h5r/series_bb_0.h5r/FitResults'
n_stop = 500
setup_module()
results = []
def do_a_frame(ind):
    t = remFitBuf.fitTask(series, ind, mdh, resultsURI=results_uri)
    res = t()
    clusterResults.fileResults(aggregate_endpoint, res.results)

for ind in range(mdh['Analysis.StartAt'], n_stop):
# for ind in range(mdh['Analysis.StartAt'], n_stop)[::-1]:
    mdh['tIndex'] = ind
    t = remFitBuf.fitTask(series, ind, mdh, resultsURI=results_uri)
    res = t()
    t1 = remFitBuf.fitTask(series, ind, mdh, resultsURI=results_uri)
    res1 = t1()
    assert len(res.results) == len(res1.results)
```
</details>

barentine, 2021-04-04:
removing camera maps from the metadata and doing no background estimation/subtraction we still hit throw the assertion error for the backwards frame order

barentine, 2021-04-04:
OK, well, this is drifting away from the original topic name, but:
```
for ind in range(mdh['Analysis.StartAt'], n_stop)[::-1]:
    mdh['tIndex'] = ind
    t = remFitBuf.fitTask(series, ind, mdh, resultsURI=results_uri)
    res = t()
    t1 = remFitBuf.fitTask(series, ind, mdh, resultsURI=results_uri)
    res1 = t1()
    np.testing.assert_array_equal(t.data, t1.data)
```
leads to
```
Exception has occurred: AssertionError

Arrays are not equal

Mismatch: 62%
Max absolute difference: 117.
Max relative difference: 0.7029703
 x: array([[124., 107., 137., ..., 114., 101., 107.],
       [114., 107., 107., ..., 103., 114., 114.],
       [124., 101., 107., ..., 114., 101., 107.],...
 y: array([[103., 103., 107., ..., 103., 107., 107.],
       [114., 107., 103., ..., 114., 114., 114.],
       [114., 107., 114., ..., 114., 101., 101.],...
  File "/home/smeagol/code/pyme-warp-drive/warpdrive/test_full.py", line 213, in <module>
    np.testing.assert_array_equal(t.data, t1.data)
```

while
```
for ind in range(mdh['Analysis.StartAt'], n_stop):
    mdh['tIndex'] = ind
    t = remFitBuf.fitTask(series, ind, mdh, resultsURI=results_uri)
    res = t()
    t1 = remFitBuf.fitTask(series, ind, mdh, resultsURI=results_uri)
    res1 = t1()
    np.testing.assert_array_equal(t.data, t1.data)
```

does not.

barentine, 2021-04-04:
Turns out it isn't the GPU fit. You can do this for any fitter, and not actually call the fit task. Just make the fit task twice in a row and the data doesn't ==. I'll pull up a MWE

barentine, 2021-04-04:
Would appear to be an issue with my pymecompress / environment
<details>

```
(base) smeagol@smeagol:~/code/pymecompress$ python test_pymecompress.py 
(base) smeagol@smeagol:~/code/pymecompress$ pytest
================================================================================================================ test session starts ================================================================================================================
platform linux -- Python 3.6.9, pytest-5.3.5, py-1.9.0, pluggy-0.13.1
rootdir: /home/smeagol/code/pymecompress
collected 3 items                                                                                                                                                                                                                                   

test_pymecompress.py ..*** Error in `/home/smeagol/miniconda3/bin/python': invalid fastbin entry (free): 0x000055821d305630 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777f5)[0x7f433a17c7f5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8038a)[0x7f433a18538a]
/lib/x86_64-linux-gnu/libc.so.6(+0x82da1)[0x7f433a187da1]
/lib/x86_64-linux-gnu/libc.so.6(posix_memalign+0x11d)[0x7f433a18c79d]
/home/smeagol/miniconda3/lib/python3.6/site-packages/numpy/core/_multiarray_umath.cpython-36m-x86_64-linux-gnu.so(+0x2ac45)[0x7f43375bdc45]
/home/smeagol/miniconda3/lib/python3.6/site-packages/numpy/core/_multiarray_umath.cpython-36m-x86_64-linux-gnu.so(+0x2ad07)[0x7f43375bdd07]
/home/smeagol/miniconda3/lib/python3.6/site-packages/numpy/core/_multiarray_umath.cpython-36m-x86_64-linux-gnu.so(+0x70d13)[0x7f4337603d13]
/home/smeagol/miniconda3/lib/python3.6/site-packages/numpy/core/_multiarray_umath.cpython-36m-x86_64-linux-gnu.so(+0x70dc8)[0x7f4337603dc8]
/home/smeagol/miniconda3/lib/python3.6/site-packages/numpy/core/_multiarray_umath.cpython-36m-x86_64-linux-gnu.so(+0x15100d)[0x7f43376e400d]
/home/smeagol/miniconda3/lib/python3.6/site-packages/numpy/core/_multiarray_umath.cpython-36m-x86_64-linux-gnu.so(+0x155dff)[0x7f43376e8dff]
/home/smeagol/miniconda3/lib/python3.6/site-packages/numpy/core/_multiarray_umath.cpython-36m-x86_64-linux-gnu.so(+0x156c31)[0x7f43376e9c31]
/home/smeagol/miniconda3/bin/python(_PyObject_FastCallDict+0x8b)[0x55821bb21e3b]
/home/smeagol/miniconda3/bin/python(PyObject_CallFunctionObjArgs+0xed)[0x55821bb405fd]
/home/smeagol/miniconda3/lib/python3.6/site-packages/numpy/core/_multiarray_umath.cpython-36m-x86_64-linux-gnu.so(+0x135d52)[0x7f43376c8d52]
/home/smeagol/miniconda3/bin/python(PyNumber_Multiply+0xa2)[0x55821bb74ad2]
/home/smeagol/miniconda3/bin/python(_PyEval_EvalFrameDefault+0x13a2)[0x55821bbcd7f2]
/home/smeagol/miniconda3/bin/python(PyEval_EvalCodeEx+0x329)[0x55821bba49b9]
/home/smeagol/miniconda3/bin/python(+0x1958e6)[0x55821bba58e6]
/home/smeagol/miniconda3/bin/python(PyObject_Call+0x3e)[0x55821bb21a5e]
/home/smeagol/miniconda3/bin/python(_PyEval_EvalFrameDefault+0x19e7)[0x55821bbcde37]
/home/smeagol/miniconda3/bin/python(PyEval_EvalCodeEx+0x966)[0x55821bba4ff6]
/home/smeagol/miniconda3/bin/python(+0x1957d4)[0x55821bba57d4]
/home/smeagol/miniconda3/bin/python(PyObject_Call+0x3e)[0x55821bb21a5e]
/home/smeagol/miniconda3/bin/python(_PyEval_EvalFrameDefault+0x19e7)[0x55821bbcde37]
/home/smeagol/miniconda3/bin/python(+0x19329e)[0x55821bba329e]
/home/smeagol/miniconda3/bin/python(+0x193ed6)[0x55821bba3ed6]
/home/smeagol/miniconda3/bin/python(+0x199b95)[0x55821bba9b95]
/home/smeagol/miniconda3/bin/python(_PyEval_EvalFrameDefault+0x10cc)[0x55821bbcd51c]
/home/smeagol/miniconda3/bin/python(+0x192e66)[0x55821bba2e66]
/home/smeagol/miniconda3/bin/python(+0x193ed6)[0x55821bba3ed6]
/home/smeagol/miniconda3/bin/python(+0x199b95)[0x55821bba9b95]
/home/smeagol/miniconda3/bin/python(_PyEval_EvalFrameDefault+0x30a)[0x55821bbcc75a]
/home/smeagol/miniconda3/bin/python(+0x193c5b)[0x55821bba3c5b]
/home/smeagol/miniconda3/bin/python(+0x199b95)[0x55821bba9b95]
/home/smeagol/miniconda3/bin/python(_PyEval_EvalFrameDefault+0x30a)[0x55821bbcc75a]
/home/smeagol/miniconda3/bin/python(+0x192e66)[0x55821bba2e66]
/home/smeagol/miniconda3/bin/python(_PyFunction_FastCallDict+0x3d8)[0x55821bba4598]
/home/smeagol/miniconda3/bin/python(_PyObject_FastCallDict+0x26f)[0x55821bb2201f]
/home/smeagol/miniconda3/bin/python(_PyObject_Call_Prepend+0x63)[0x55821bb26aa3]
/home/smeagol/miniconda3/bin/python(PyObject_Call+0x3e)[0x55821bb21a5e]
/home/smeagol/miniconda3/bin/python(+0x16b371)[0x55821bb7b371]
/home/smeagol/miniconda3/bin/python(_PyObject_FastCallDict+0x8b)[0x55821bb21e3b]
/home/smeagol/miniconda3/bin/python(_PyObject_FastCallKeywords+0xaa)[0x55821bba418a]
/home/smeagol/miniconda3/bin/python(+0x199c0e)[0x55821bba9c0e]
/home/smeagol/miniconda3/bin/python(_PyEval_EvalFrameDefault+0x10cc)[0x55821bbcd51c]
/home/smeagol/miniconda3/bin/python(+0x193c5b)[0x55821bba3c5b]
/home/smeagol/miniconda3/bin/python(+0x199b95)[0x55821bba9b95]
/home/smeagol/miniconda3/bin/python(_PyEval_EvalFrameDefault+0x30a)[0x55821bbcc75a]
/home/smeagol/miniconda3/bin/python(PyEval_EvalCodeEx+0x329)[0x55821bba49b9]
/home/smeagol/miniconda3/bin/python(+0x1957d4)[0x55821bba57d4]
/home/smeagol/miniconda3/bin/python(PyObject_Call+0x3e)[0x55821bb21a5e]
/home/smeagol/miniconda3/bin/python(_PyEval_EvalFrameDefault+0x19e7)[0x55821bbcde37]
/home/smeagol/miniconda3/bin/python(+0x19329e)[0x55821bba329e]
/home/smeagol/miniconda3/bin/python(+0x193ed6)[0x55821bba3ed6]
/home/smeagol/miniconda3/bin/python(+0x199b95)[0x55821bba9b95]
[... cut ...]
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/pluggy/manager.py", line 87 in <lambda>
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/pluggy/manager.py", line 93 in _hookexec
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/pluggy/hooks.py", line 286 in __call__
  File "/home/smeagol/miniconda3/lib/python3.6/site-packages/_pytest/config/__init__.py", line 93 in main
  File "/home/smeagol/miniconda3/bin/pytest", line 11 in <module>
Aborted (core dumped)

```

</details>

barentine, 2021-04-05:
OK, so in summary: my pymecompress was behind and needed to be recompiled. That wasn't actually the issue with the data not matching - I will open a new issue for that in a second.

So, worth shifting this issue back to its original topic: is it a bug that we go backwards?

David-Baddeley, 2021-04-05:
In theory, the ordering (LIFO/FIFO) should not matter - fitting should be completely independent of the order in which frames are fitted. 

It might still be worth working out why you are seeing LIFO though. I initially thought it might have been a `list.pop()` rather than `list.pop(0)`, for which there might be a valid performance argument (`list.pop()` is O(1) whereas `list.pop(0)` is O(N)). Most of the places where this might occur, however, seem to use queues not lists, which should be FIFO.

The one place I've been able to find an order inversion is if the data is rated as non-local (`rulenodeserver.Rater.get_local()` and `rulenodeserver.Rater.get_any()` - the former appends non-local tasks to a list and the later pops them. If this only manifests when the data is non-local (should be detectable from the reported cost) then we might have the explanation. Again, I wouldn't expect this to have any deleterious impact on the fitting (so not strictly a bug), but would be open to changing the `pop()` in   `rulenodeserver.Rater.get_any()` to a `pop(0)` to keep things consistent between local and non-local.

David-Baddeley, 2021-04-05:
Removed the bug label as we make no guarantees as to the ordering of task execution (i.e. this is not a bug, even if it is worth looking at a little more). If this is causing problems it implies an incorrect assumption about ordering somewhere  downstream.

barentine, 2021-04-10:
Yeah, the current bidding stuff is off. I put a log statement in to print np.mean(costs)
```
leserver:bidding, avg cost nan
INFO:ruleserver:bidding, avg cost 1.000000
INFO:ruleserver:bidding, avg cost 1.000000
INFO:ruleserver:bidding, avg cost nan
INFO:ruleserver:bidding, avg cost 1.000000
INFO:ruleserver:bidding, avg cost 1.000000
INFO:ruleserver:bidding, avg cost 1.000000
INFO:ruleserver:bidding, avg cost 1.000000
INFO:ruleserver:bidding, avg cost nan
INFO:ruleserver:bidding, avg cost nan
INFO:ruleserver:bidding, avg cost nan
INFO:ruleserver:bidding, avg cost 1.000000
INFO:ruleserver:bidding, avg cost 1.000000
INFO:ruleserver:bidding, avg cost 1.000000
INFO:ruleserver:bidding, avg cost nan
INFO:ruleserver:bidding, avg cost nan
INFO:ruleserver:bidding, avg cost nan
INFO:ruleserver:bidding, avg cost nan
INFO:ruleserver:bidding, avg cost nan
INFO:ruleserver:bidding, avg cost nan
DEBUG:ruleserver:release_rule_tasks(ruleID = 000001-f0e73b721f8f403ebfa2e59d362223ef, release_start=0, release_end=45000)
INFO:ruleserver:bidding, avg cost nan
INFO:ruleserver:bidding, avg cost 1.000000
INFO:ruleserver:bidding, avg cost 1.000000
INFO:ruleserver:bidding, avg cost 1.000000
INFO:ruleserver:bidding, avg cost nan

```

barentine, 2021-04-13:
Seems like things generally work OK for shorter series. I'm going to take a peak at the cases when the rulenodeserver returns empty lists, as this would currently kick us into the bidding and I'm not sure we want that
```

In [8]: np.array([], 'f4').mean()
/home/smeagol/miniconda3/bin/ipython:1: RuntimeWarning: Mean of empty slice.
  #!/home/smeagol/miniconda3/bin/python
/home/smeagol/miniconda3/lib/python3.6/site-packages/numpy/core/_methods.py:85: RuntimeWarning: invalid value encountered in true_divide
  ret = ret.dtype.type(ret / rcount)
Out[8]: nan

In [9]: np.array([], 'f4').mean() < 0.2
/home/smeagol/miniconda3/bin/ipython:1: RuntimeWarning: Mean of empty slice.
  #!/home/smeagol/miniconda3/bin/python
/home/smeagol/miniconda3/lib/python3.6/site-packages/numpy/core/_methods.py:85: RuntimeWarning: invalid value encountered in true_divide
  ret = ret.dtype.type(ret / rcount)
Out[9]: False

```

barentine, 2021-04-13:
OK, so wasn't that, though I'll PR it shortly anyway. @David-Baddeley the issue with not getting local tasks has to do with how many rules the ruleserver has. If I launch loads they all get cost 1.0, if I launch a couple, they're ~all local

barentine, 2021-04-13:
Note that the real issue was that in my notebook, I clusterIO.cglob'd, but then 'pyme-cluster:///
 + s for s in cglob` rather than writing out my local serverfilter. clusterIO.is_local is strict, and inserts the local serverfilter by default... I think that's fairly inconvenient, but now that I know I'll just be better with my notebooks, but might be something to consider tweaking @David-Baddeley

David-Baddeley, 2021-04-13:
I'm pretty sure there was a reason for making the `is_local()` test strict. Can't recall with 100% certainty, but potentially either to allow you to run stuff with multiple servers on a single machine, or for when you have multiple clusters / cluster of ones on the same network segment.

barentine, 2021-04-27:
If we want to keep that `is_local` test strict, it would be nice to make sure we are substituting correctly in rules. At the moment, if I batch hundreds of recipe rules from clusterUI, I can fail hundreds of them on IO errors, even if the nodeserver chunk size is 1, with 1 worker. In other words we aren't going local there either, which you can sort of see from the rule description, where the output_dir has the serverfilter substituted, but it is blank on the `input` key, which again will fail the is_local
```
===========================================================================
Error in rule: 000001-d293eee6ee9b451dbd4370da20daceb8 running task: 69 on stromberg:12496

taskDescr:
----------
{'id': '000001-d293eee6ee9b451dbd4370da20daceb8~69', 'type': 'recipe', 'inputs': {'input': 'pyme-cluster:///Bergamot/2021_2_16/EMC0105H/mapped/17_2_series_W_mapped.hdf'}, 'output_dir': 'pyme-cluster://villains/Bergamot/2021_2_16/EMC0105H/shell-kde-gmm-ripleys/', 'taskdef': {'recipe': "- localisations.DensityMapping:\n    colours:\n    - chan0\n    inputLocalizations: mapped\n    jitterScale: 30.0\n    outputImage: jt2d\n    pixelSize: 40.0\n    renderingModule: Gaussian\n- processing.Threshold:\n    inputName: smoothed\n    method: otsu\n    outputName: thresholded\n    processFramesIndividually: false\n- processing.Label:\n    inputName: thresholded\n    minRegionPixels: 1\n    outputName: labeled\n- processing.SelectLargestLabel:\n    inputName: labeled\n    outputName: nucleus\n- processing.BinaryDilation:\n    inputName: nucleus\n    iterations: 10\n    outputName: mask\n    processFramesIndividually: false\n- localisations.LabelsFromImage:\n    inputImage: mask\n    inputName: mapped\n    outputName: labeled_points\n- tablefilters.FilterTable:\n    filters:\n      objectID:\n      - 0.5\n      - 1.5\n    inputName: labeled_points\n    outputName: inside_nucleus_mask\n- localisations.ExtractTableChannel:\n    channel: chan0\n    inputName: inside_nucleus_mask\n    outputName: lamin\n- surface_fitting.SphericalHarmonicShell:\n    input_name: lamin\n    output_name: harmonic_shell\n    output_name_mapped: lamin_shell_mapped\n- surface_fitting.AddSphericalHarmonicShellMappedCoords:\n    input_localizations: inside_nucleus_mask\n    input_shell: harmonic_shell\n    output_mapped: shell_mapped\n- output.HDFOutput:\n    filePattern: '{output_dir}/{file_stub}_spharm-gmm-ripleys.hdf'\n    inputVariables:\n      gmm_labeled: gmm_labeled\n      harmonic_shell: harmonic_shell\n      measured_gmm: measured_gmm\n      r_uniform_kde: r_uniform_kde\n      ripleys: ripleys\n      shell_mapped: shell_mapped\n    scheme: pyme-cluster://\n- filters.MaxFilter:\n    inputName: jt2d\n    outputName: smoothed\n    processFramesIndividually: false\n    sizeX: 15\n    sizeY: 15\n    sizeZ: 1\n- localisations.ExtractTableChannel:\n    channel: chan1\n    inputName: shell_mapped\n    outputName: mapped_coilin\n- pointcloud.GaussianMixtureModel:\n    input_points: mapped_coilin\n    mode: bic\n    n: 50\n    output_labeled: gmm_labeled\n- tablefilters.FilterTable:\n    filters:\n      gmm_label_log_prob:\n      - -26.0\n      - 0.0\n    inputName: gmm_labeled\n    outputName: filtered_clusters\n- localisations.MeasureClusters3D:\n    inputName: filtered_clusters\n    labelKey: gmm_label\n    outputName: measured_gmm\n- surface_fitting.ImageMaskFromSphericalHarmonicShell:\n    input_image_bound_source: mapped_coilin\n    input_shell: harmonic_shell\n    output: shell_image_mask\n- pointcloud.Ripleys:\n    binSize: 30.0\n    inputMask: shell_image_mask\n    inputPositions: mapped_coilin\n    nbins: 150\n    normalization: L\n    outputName: ripleys\n    three_d: true\n- output.ImageOutput:\n    filePattern: '{output_dir}/{file_stub}_nucleus_mask.h5'\n    inputName: mask\n    scheme: pyme-cluster://\n- surface_fitting.SHShellRadiusDensityEstimate:\n    input_shell: harmonic_shell\n    output: r_uniform_kde\n    r_bin_spacing: 0.02\n"}}

Traceback:
----------
Traceback (most recent call last):
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/urllib3/connectionpool.py", line 426, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/urllib3/connectionpool.py", line 421, in _make_request
    httplib_response = conn.getresponse()
  File "/home/ubuntu/miniconda3/lib/python3.6/http/client.py", line 1346, in getresponse
    response.begin()
  File "/home/ubuntu/miniconda3/lib/python3.6/http/client.py", line 307, in begin
    version, status, reason = self._read_status()
  File "/home/ubuntu/miniconda3/lib/python3.6/http/client.py", line 268, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/home/ubuntu/miniconda3/lib/python3.6/socket.py", line 586, in readinto
    return self._sock.recv_into(b)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/urllib3/connectionpool.py", line 727, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/urllib3/util/retry.py", line 410, in increment
    raise six.reraise(type(error), error, _stacktrace)
[... cut ...]
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/requests/sessions.py", line 530, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/requests/sessions.py", line 643, in send
    r = adapter.send(request, **kwargs)
  File "/home/ubuntu/miniconda3/lib/python3.6/site-packages/requests/adapters.py", line 529, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: HTTPConnectionPool(host='10.150.7.9', port=15348): Read timed out. (read timeout=10)

```

barentine, 2021-04-27:
Have a fix locally, will PR shortly

David-Baddeley, 2021-11-11:
Closing as old / mostly fixed. Can be re-opened if needed

closed 2021-11-11 by David-Baddeley