python-microscopy/python-microscopy #984: [BUG] dictmdhandler attribute access not functional

barentine, 2021-04-04:
**Describe the bug**

in remFitBuf:
```
    @classmethod
    def calcSigma(cls, md, data):
        """
        Estimate a per-pixel noise standard deviation.
        Parameters
        ----------
        md: PYME.IO.MetaDataHandler.MDHandlerBase
            metadata
        data: ndarray
            data, having been dark-map and flatfield corrected. [ADU]

        Returns
        -------
        sigma: ndarray
            estimated per-pixel noise as a standard deviation. [ADU]

        Notes
        -----
        TrueEMGain: float
            conversion factor between electrons post-electron-multiplication and raw photoelectrons. [e-/pe-]
        NoiseFactor: float
            typically 1.4 for EMCCD when gain is > 10, 1 for CCD and CMOS. [pe-^(-1/2)] see doi: 10.1109/TED.2003.813462
        var: ndarray
            (per-pixel) variance, readout noise [e-^2]
        """
        var = np.atleast_3d(cameraMaps.getVarianceMap(md)) # this must be float type!! Should we enforce with an 'astype' call?
        return np.sqrt(var + (float(md.Camera.NoiseFactor)**2)*(float(md.Camera.ElectronsPerCount)*float(md.Camera.TrueEMGain)*np.maximum(data, 1.0) + float(md.Camera.TrueEMGain)*float(md.Camera.TrueEMGain)))/float(md.Camera.ElectronsPerCount)
```

```
Exception has occurred: AttributeError
'DictMDHandler' object has no attribute 'Camera'
  File "/home/smeagol/code/python-microscopy/PYME/localization/remFitBuf.py", line 597, in calcSigma
    return np.sqrt(var + (float(md.Camera.NoiseFactor)**2)*(float(md.Camera.ElectronsPerCount)*float(md.Camera.TrueEMGain)*np.maximum(data, 1.0) + float(md.Camera.TrueEMGain)*float(md.Camera.TrueEMGain)))/float(md.Camera.ElectronsPerCount)
  File "/home/smeagol/code/python-microscopy/PYME/localization/remFitBuf.py", line 454, in __call__
    self.sigma = self.calcSigma(md, self.data)
  File "/home/smeagol/code/pyme-warp-drive/warpdrive/test_full.py", line 210, in <module>
    res = t()
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10]
 - Python version:
 - numpy version:
 - wxpython version [if a UI bug]:

**Additional context**
Add any other context about the problem here.

csoeller, 2021-04-04:
This could be similar to #976, where, according to @David-Baddeley, newer MD handlers do not expose an attribute interface and code should be changed to use a dict-like interface. But @David-Baddeley also said:

_Hitting all the attribute style usages at once is likely to be very hard, so a reasonable short-term approach is likely to be:_

1. _implement __getattr__ on the MDHandlerBase to permit attribute based access for all metadata handlers_
2. _make this raise/flag a DeprecationWarning_

I haven't tried this yet but rather changed for the files I was dealing with to dict-like access.

barentine, 2021-04-05:
Thanks @csoeller! Switched the accesses I bumped into to __getitem__ in #982

David-Baddeley, 2021-04-05:
I think #502 got merged prematurely to test some of the CI stuff - I should have clearly marked `DictMDHandler` as work in progress. Actually using it anywhere where it gets to, e.g.,  `remFitBuf` should probably be discouraged until it's a bit more finished. At this point it probably makes sense to just add the necessary `__getattr__` bindings and finish it off.

PR #993 (2021-04-08): Metadata attribute access
fixes #984 and fixes #976. Patch over differences in metadata handler implementation
closed 2021-04-08 by David-Baddeley