python-microscopy/python-microscopy #985: [BUG] PYME.IO.buffers.dataBuffer returns views, can easily botch percentile buffer stuff + more

barentine, 2021-04-05:
**Describe the bug**

PYME.IO.buffers.dataBuffer returns a view into a larger array, which it will then casually change later. The small size (12), combined with the fact that we use the same dataBuffer instance for background buffering means if you are estimating background using more than 12 frames and are using a fitfactory with `GPU_PREFIT` then the `fitTask.data` as passed to the fit module is not the droid you're looking for. If you are doing a standard CPU fit, you are probably OK because `fitTask.data` gets reassigned with a camera-map-corrected slice (which should no longer be referenced in memory to dataBuffer) before the background is calculated

```
from PYME.IO.DataSources.RandomDataSource import DataSource
from PYME.IO.buffers import dataBuffer
import numpy as np

IMSIZE_R = 960
IMSIZE_C = 240

ds = DataSource(IMSIZE_R, IMSIZE_C, 100)
DBUFF = dataBuffer(ds)

d0_cp = np.copy(DBUFF.getSlice(0))
d0 = DBUFF.getSlice(0)

for ind in range(50):
    DBUFF.getSlice(ind)

np.testing.assert_array_equal(d0, d0_cp)

```
yields
```
AssertionError: 
Arrays are not equal

Mismatched elements: 230392 / 230400 (100%)
Max absolute difference: 32711
Max relative difference: 31201.
 x: array([[27612, 26198, 13864, ...,  7566, 12828, 14079],
       [ 6413,  1712, 23920, ...,  2898, 11571,  3135],
       [  551, 27093,  9900, ..., 22442, 30678, 16577],...
 y: array([[31297, 27226, 32548, ..., 14141,  8417, 18052],
       [15002, 17696, 19151, ...,  8631,  6207, 28774],
       [ 7977, 26806, 12551, ...,  4782,  5989, 20026],...
```

**Expected behavior**
`fitTask.data` not to change during background estimation

barentine, 2021-04-05:
everything before https://github.com/python-microscopy/python-microscopy/pull/120/files should befine, saved by the camera map correction

David-Baddeley, 2021-04-05:
Returning a view rather than a copy is desirable from a performance (avoid an un-necessary alloc and copy) and memory usage perspective.  There would also seem to be a size error on the buffer (off by one) although there is a chance that this might be down to a different interpretation of BGRange in CPU and GPU fits (this is just a hunch - needs checking). I suspect that increasing the data buffer allocation would mask / eliminate this error, whilst still allowing view based access with the desirable memory properties.

That said, the current behaviour (i.e. behaves properly as long as the buffer is large enough, but fails messily when this is not the case) is definitely fragile, and the copying fix in #982 would be a lot more robust. An ideal solution would likely re-factor the buffer so that you took a lease on a given frame (released when it goes out of scope), but it's hard to know if it's worth the effort. A pragmatic solution might be to merge #982, but add some FIXME/TODO comments to indicate that a non-copying solution would be preferable.

On second thoughts, a larger buffer would probably mask this in a large fraction, but not all cases - circular buffer overwrites depend on the the age of a frame in the buffer, not strictly order, so out of order analysis could still cause issues.

David-Baddeley, 2021-04-11:
should be fixed with #982

closed 2021-04-11 by David-Baddeley