python-microscopy/python-microscopy #987: [SUPPORT] Is there an easy way to grab the mean value of the most recent image off the camera?

zacsimile, 2021-04-06:
Is there an easy way to grab the mean intensity value of the most recent image off of the camera?

zacsimile, 2021-04-06:
Hang on, I think this requires a discussion of a specific case.

closed 2021-04-06 by zacsimile

David-Baddeley, 2021-04-06:
Might not specifically need a special case discussion. Most recent frame is available at `scope.frameWrangler.currentFrame`. Calling e.g. `scope.frameWrangler.currentFrame.mean()` should work for one-off accesses, although I wouldn't do it too frequently. 

If you want to poll the frames as they come in, it would be better to connect to either the `frameWrangler.onFrame` or `frameWrangler.onFrameGroup` signals which get called for every frame, and at a slower frequency suitable for GUI updates (typically 10-20 Hz) respectively.

zacsimile, 2021-04-06:
Ah, this is perfect! I only need it for like 8 frames during a calibration step.