python-microscopy/python-microscopy #988: [SUPPORT] is action manager and associated code in a usable state?

csoeller, 2021-04-06:
Note: Support is also available by emailing support@python-microscopy.org, choose this option if you would like the question and answer to be publicly searchable.

**What are you trying to do?**
I am trying to use the action manager / spoolcontroller code. I found, to my slight dismay, that the interface code that I used and extensively tested in December (implemented in https://github.com/csoeller/PYME-extra/blob/master/PYMEcs/Acquire/Actions/custom.py) has stopped working.

Similarly, the `Queue Actions` action manager UI currently has a button to issue a `spoolController.StartSpooling` command which gives an error since all syntax seems to have changed to the `start_spooling` etc syntax. I also have trouble to figure out how to use the `max_frames` setting to `start_spooling` which gives me

```
TypeError: start_spooling() got an unexpected keyword argument 'max_frames'
```

My questions are:

- is the action manager interface in a sane state?
- did I miss that these syntax changes etc were instigated?
- do we have docs that explain things properly?

**Where are you stuck?**
Explanations see above.

David-Baddeley, 2021-04-07:
The action manager is in a state of ongoing, active, development - it is usable and should be fairly robust, but falls under the category of "internals" rather than "API". In other words whilst there is an ongoing commitment that to maintain the functionality encapsulated in the `ActionManager`, there is no guarantee that the **programatic** interface to that functionality will be static.  This will likely change as the implementation details shake out a bit more (I think the current iteration has a fair chance of sticking around a bit, but there are still a couple of non-finalised design choices which might result in small changes). 

The "internals" status notwithstanding, the latest changes are pretty small, and it should be really easy to adapt to the new interface (basically you pass a dictionary of settings rather than a constantly growing/changing set of keyword arguments - the `max_frames` argument is now one of the keys in the settings dictionary). This is documented in the `start_spooling()` docstring. The relevant changes came in https://github.com/python-microscopy/python-microscopy/pull/799, which was potentially merged a bit earlier than might otherwise have been the case as, in addition to refactoring etc ... it happened to fix a few bugs/feature requests.

David-Baddeley, 2021-04-07:
In general it's a bit hard to know what other people are using/relying on. PYMEAcquire could use a **lot** of refactoring / general maintenance and doing it all in a backwards compatible way would likely be prohibitive. In general I regard the following as "API", or places where user code might be expected to interact with PYMEAcquire and as a result should maintain backwards compatibility and/or have clearly signalled deprecation paths.

- init scripts, and functions which are typically called from them (e.g. the `@init_hardware()` and `@init_gui()` decorators)
- the protocol spec
- the piezo/positioning interface
- the laser interface
- the camera interface (although this is really in need of refactoring)
- `scope.state` / `microscope.StateHandler`
- `scope.register_camera()`, `scope.register_piezo()` ( `scope.SetPos()`, `scope.GetPos()`, `scope.EnableJoystick()`, `scope.centre_roi_on()`, `scope.PanCamera()`, `scope.turnAllLasersOff()` might change, but are "semi-API" and will be provided with a deprecation path if they do change).
- `scope.frameWrangler.start()`, `scope.frameWrangler.stop()`, `scope.frameWrangler.start()`, `scope.frameWrangler.onStart`, `scope.frameWrangler.onStop`, `scope.frameWrangler.onFrame`, `scope.frameWrangler.onFrameGroup`, `scope.frameWrangler.currentFrame`
- `PYME.IO.MetadataHandler.provideStartMetadata` and `PYME.IO.MetadataHandler.provideStopMetadata` - although these are in serious need of refactoring (the deprecation route here might be a bit more abrupt, and break code with a pointer to where the replacement is rather than supporting both new and old methods for a transition period)
- `PYME.Acquire.eventLog.WantEventNotification` and `PYME.Acquire.eventLog.logEvent` - also needs serious refactoring, and might have a more abrupt break than some of the others.
- `MainFrame.AddMenuItem()`, `MainFrame.AddPage()`, `MainFrame.AddTool()`, `MainFrame.AddAqTool()`, `MainFrame.AddCamTool()` (although the later options might end up deprecated in favour of a more flexible layout config)

This is open to discussion - if you feel I have missed something which should be programatic API, let me know (I'd be open to potentially adding the action stuff to this list). All this stuff obviously needs to be documented a lot better. 

Another good way of doing this, short of making everything "API" would be to find a way of making sure that you got notified when a PR touched a bit of code that you were using (which would have picked up the changes in #799). I feel that this should be possible, but it's a bit hard to know how in practice - something like `CODEOWNERS` (https://docs.github.com/en/github/creating-cloning-and-archiving-repositories/about-code-owners) but for users of a particular part of the code.

csoeller, 2021-04-07:
Thanks for the explanations, this helps a lot. I have in a first pass adapted my bindings, still using the string type interface and will try the more neat action classes in a next step.

Part of my confusion was from `StartSpooling` mentions and calls (in `actionUI`) still hanging around. I have submitted PR #989 to try to fix that.

I'll look through the API/function list a bit more and see how that affects what we are trying to do. It is a good reference to have here in any case.

David-Baddeley, 2021-04-07:
I could have sworn I'd hit all usages, but maybe not - or potentially they got committed but not merged.

David-Baddeley, 2021-04-07:
Ahh - the usages in #989 are mostly in the docs, it does look like there's still a problem in the ROIs bit though.

csoeller, 2021-04-08:
Thanks, this is now very good IMHO with the PR and associated ROI fix. As I am tweaking my camera map ROI tiling code which uses the actions to good effect I have a question how to achieve something.

I would like to spool all series for a given map to the same subdirectory, possibly even forcing `FILE` mode. The subdirectory should have some form like `mapA` and increase to `mapB`, `mapC` if several maps are recorded on a given day. I see that I can use settings `subdirectory` and `method` with `start_spooling`. Is that the way to go?

Also, how would the code, being a init file 'plugin',  best check if subdirs `mapA`, `mapB` etc already exist and increase the subdir name accordingly?

David-Baddeley, 2021-04-08:
Using `subdirectory` and `method` would be a sensible approach. As to subdirectory naming, it's a bit trickier - if spooling to file, you should be able to handle this fairly easily with a `os.path.exists(spoolController.get_dirname(subdirectory))` check. For cluster/cluster of one spooling it might be a bit trickier.

As I understand it, the map tiling you are doing is to handle high frame rates which need a ROI smaller than the whole chip. There are some caveats with using file mode here - namely that it probably won't be fast enough  - even if saving to a SSD, as the builtin compression in pytables/HDF5 is really slow and represents a bottleneck at around 50 FPS. In this case, frames will pile up in internal buffers waiting to get to disk. These buffers might not all be size limited, and there is a chance that you will end up using up all your memory and swap and causing things to crash/grind to a halt.

csoeller, 2021-04-09:
Many thanks, all good points. On our systems even "slow" acquisition times started blocking/freezing at full chip size but things become really easy with smaller ROIs. I have tried down to 10 ms as yet and that is all good, cannot remember now, I might have tried 5 ms as well.

Plan is to do the simple `FILE` based implementation first and get that proof of concept going smoothly so that is is robust and REALLY EASY. Currently we don't have maps because it is a P** in the B** to do it, so actually nobody does it.

If indeed we run into issues with faster AQ times and this is actually needed (not so much by us as yet) then we can use this as a starting point and tweak, for example, to work with a cluster as well. Will open up to you guys anyway to ask for feedback once we have something that runs well in our production systems.

David-Baddeley, 2021-07-13:
Is this sufficiently well resolved that we can close?

csoeller, 2021-07-13:
can close for now;  the thread above contains really useful info that is not captured elsewhere; I suppose it is searchable while closed but quite easy to loose track of such discussions - would be nice if there were a better way to preserve.

closed 2021-07-13 by csoeller