python-microscopy/python-microscopy #990: [SUPPORT] possible issues launching PYMEClusterOfOne on macOS - how to overcome?

csoeller, 2021-04-07:
Note: Support is also available by emailing support@python-microscopy.org, choose this option if you would like the question and answer to be publicly searchable.

**What are you trying to do?**
I am trying to experiment with `PYMEClusterOfOne` on macOS 10.15 (Catalina). I overcame an initial issue `socket.gaierror: [Errno 8] nodename nor servname provided, or not known` using tips at https://stackoverflow.com/questions/39970606/gaierror-errno-8-nodename-nor-servname-provided-or-not-known-with-macos-sie.

But now am hitting a further issue I have not been able to suss out as yet.

**Where are you stuck?**
I am now hitting an error `AttributeError: 'ServerThread' object has no attribute 'nodeserver'`, details as below. Is this critical? Any obvious fix?

```
$PYMEClusterOfOne 
Setting icon
INFO:PYME.cluster.cluster_of_one:Launching data server: root=/Users/csoe002/PYMEData
INFO:PYME.cluster.cluster_of_one:Launching rule server
INFO:ruleserver:Starting ruleserver on 127.0.0.1:0
INFO:numexpr.utils:NumExpr defaulting to 8 threads.
INFO:root:========================================
PYMEDataServer, running on python 3.7.6 (default, Jan  8 2020, 13:42:34) 
[Clang 4.0.1 (tags/RELEASE_401/final)]

INFO:root:Serving from directory: /Users/csoe002/PYMEData
INFO:root:Serving HTTP on 127.0.0.1 port 64422 ...
INFO:PYME.cluster.cluster_of_one:Launching node server
dict_values(['http://127.0.0.1:64421/'])
DEBUG:nodeserver:Starting to poll for tasks
Traceback (most recent call last):
  File "/Users/csoe002/opt/anaconda2/envs/pyme-py3/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/Users/csoe002/opt/anaconda2/envs/pyme-py3/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/cluster/PYMERuleNodeServer.py", line 167, in <module>
    main()
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/cluster/PYMERuleNodeServer.py", line 105, in main
    sa = proc.nodeserver.socket.getsockname()
AttributeError: 'ServerThread' object has no attribute 'nodeserver'
127.0.0.1 - - [07/Apr/2021 20:07:54] code 404, message No handler for /favicon.ico
nodeserver:INFO:Starting nodeserver on 127.0.0.1:0
INFO:nodeserver:Starting nodeserver on 127.0.0.1:0
```

barentine, 2021-04-07:
Hi @csoeller , would you mind sharing your conda list? I'm OK running `PYMEClusterOfOne` on my macOS 10.15 (Catalina).

barentine, 2021-04-07:
To your point about it being critical, if you don't see a log about 'launching worker processes' then there aren't any workers and the launch has failed. Launching via PYMEClusterOfOne doesn't necessarily print as much to stdout/stderr - might be helpful to first `PYMERuleServer` in one terminal, followed by `PYMERuleNodeServer` in another as a debug step.

csoeller, 2021-04-07:
Hi @barentine, many thanks for looking into it, here is my conda listing. Will try the separate calls next.

```
$ conda list
# packages in environment at /Users/csoe002/opt/anaconda2/envs/pyme-py3:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                   py37_0  
appnope                   0.1.2           py37hecd8cb5_1001  
argon2-cffi               20.1.0           py37h9ed2024_1  
async_generator           1.10             py37h28b3542_0  
attrs                     20.3.0             pyhd3eb1b0_0  
babel                     2.8.1              pyhd3eb1b0_0  
backcall                  0.2.0                      py_0  
backports                 1.0                pyhd3eb1b0_2  
backports.lzma            0.0.14           py37hd621484_0  
blas                      1.0                         mkl  
bleach                    3.3.0              pyhd3eb1b0_0  
blosc                     1.19.0               hab81aa3_0  
brotlipy                  0.7.0           py37h9ed2024_1003  
bzip2                     1.0.8                h1de35cc_0  
ca-certificates           2021.1.19            hecd8cb5_1  
cairo                     1.14.12              hc4e6be7_4  
certifi                   2020.12.5        py37hecd8cb5_0  
cffi                      1.14.3           py37h2125817_2  
chardet                   3.0.4           py37hecd8cb5_1003  
cheroot                   8.4.7            py37hecd8cb5_1  
cherrypy                  18.6.0                   py37_0  
cloudpickle               1.6.0                      py_0  
cryptography              3.2.1            py37hbcfaee0_1  
cycler                    0.10.0                   py37_0  
cython                    0.29.21          py37h23ab428_0  
cytoolz                   0.11.0           py37haf1e3a3_0  
dask-core                 2.30.0                     py_0  
dbus                      1.13.18              h18a8e69_0  
decorator                 4.4.2                      py_0  
defusedxml                0.7.1              pyhd3eb1b0_0  
docutils                  0.16                     py37_1  
entrypoints               0.3                      py37_0  
expat                     2.2.10               hb1e8313_2  
fftw                      3.3.4                         0    david_baddeley
fontconfig                2.13.0               h5d5b041_1  
freetype                  2.10.4               ha233b18_0  
fsc                       0.1                       dev_0    <develop>
future                    0.18.2                   py37_1  
gettext                   0.19.8.1             hb0f4f8b_2  
glib                      2.66.1               h9bbe63b_0  
hdf5                      1.10.4               hfa1e0ec_0  
icu                       58.2                 h0a44026_3  
idna                      2.10                       py_0  
ifaddr                    0.1.6                      py_0    david_baddeley
imageio                   2.9.0                      py_0  
imagesize                 1.2.0                      py_0  
importlib-metadata        3.7.3            py37hecd8cb5_1  
importlib_metadata        3.7.3                hd3eb1b0_1  
intel-openmp              2019.4                      233  
ipykernel                 5.3.4            py37h5ca1d4c_0  
ipython                   7.19.0           py37h01d92e1_0  
ipython_genutils          0.2.0              pyhd3eb1b0_1  
ipywidgets                7.6.3              pyhd3eb1b0_1  
jaraco.classes            3.1.0                      py_0  
jaraco.collections        3.0.0                      py_0  
jaraco.functools          3.0.1                      py_0  
jaraco.text               3.2.0                    py37_1  
jedi                      0.17.2           py37hecd8cb5_1  
jinja2                    2.11.2                     py_0  
joblib                    0.17.0                     py_0  
jpeg                      9b                   he5867d9_2  
jsonschema                3.2.0                      py_2  
jupyter                   1.0.0                    py37_7  
[... 114 lines cut ...]
tifffile                  2020.10.1        py37h0cf3a3e_2  
tk                        8.6.10               hb0a8c7a_0  
toolz                     0.11.1                     py_0  
toposort                  1.5                      py37_3    david_baddeley
tornado                   6.0.4            py37h1de35cc_1  
traitlets                 5.0.5                      py_0  
traits                    6.1.1            py37haf1e3a3_0  
traitsui                  7.1.0                      py_0  
typing_extensions         3.7.4.3            pyha847dfd_0  
ujson                     4.0.1            py37hb1e8313_0  
urllib3                   1.25.11                    py_0  
wcwidth                   0.2.5                      py_0  
webencodings              0.5.1                    py37_1  
wheel                     0.35.1             pyhd3eb1b0_0  
widgetsnbextension        3.5.1                    py37_0  
wxpython                  4.0.4            py37hebc47a7_0  
xz                        5.2.5                h1de35cc_0  
yaml                      0.2.5                haf1e3a3_0  
zc.lockfile               2.0                        py_0  
zeroconf                  0.28.0                   py37_3    david_baddeley
zeromq                    4.3.3                hb1e8313_3  
zipp                      3.4.1              pyhd3eb1b0_0  
zlib                      1.2.11               h1de35cc_3  
zstd                      1.4.5                h41d2c2f_0  
```

barentine, 2021-04-07:
first guess would be zeroconf version is too high. I know 0.28's in David's channel, but I seem to recall (see #442) 0.26.3 being the latest we can use currently

csoeller, 2021-04-07:
Hm, just downgraded to zeroconf=0.26.3 and no change. Fired up `PYMERuleServer` in a terminal and got a very similar looking:

```
$ PYMERuleServer
Traceback (most recent call last):
  File "/Users/csoe002/opt/anaconda2/envs/pyme-py3/bin/PYMERuleServer", line 33, in <module>
    sys.exit(load_entry_point('python-microscopy', 'console_scripts', 'PYMERuleServer')())
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/cluster/PYMERuleServer.py", line 85, in main
    sa = proc.distributor.socket.getsockname()
AttributeError: 'ServerThread' object has no attribute 'distributor'
INFO:ruleserver:Starting ruleserver on 127.0.0.1:15346
```

David-Baddeley, 2021-04-07:
`PYMEClusterOfOne` defaults to using the non-zeroconf, sqlite based discovery, so zeroconf would not be high on my list of suspects.

Another potentially useful place to look for error output would be `~/PYMEData/LOGS/<computer name>/nodeserver.log`.

What is your computer name (I'm wondering if this might be a problem with spaces or special chars in the computer name)

csoeller, 2021-04-07:
Computer name has potential nastiness in it:

```
$ scutil --get ComputerName
Christian’s MacBook Pro
```

Logs are in `~/PYMEData/LOGS/Christians-MacBook-Pro/` with nothing obvious in there.

David-Baddeley, 2021-04-07:
Other potential debugging suggestions as follows:

Your traceback occurs in the following snippet:

```python
proc = rulenodeserver.ServerThread(distributors[0], serverPort, externalAddr=externalAddr, profile=False)
proc.start()
  
# TODO - do we need this advertisement
#get the actual adress (port) we bound to
time.sleep(0.5)
sa = proc.nodeserver.socket.getsockname()
```

`proc.start()` calls `rulenodeserver.ServerThread.run()` (copied below) in a new thread:

```python
def run(self):
    self.nodeserver = WFNodeServer('http://' + self.distributor + '/', port=self.port, ip_address=self.externalAddr)

    try:
        logger.info('Starting nodeserver on %s:%d' % (self.externalAddr, self.port))
        self.nodeserver.serve_forever()
    except:
        logger.exception('Error running nodeserver')
    finally:
        self.nodeserver._do_poll = False
        logger.info('Shutting down ...')
        self.nodeserver.shutdown()
        self.nodeserver.server_close()
```

`proc.nodeserver` is only assigned within this thread. Usually the 0.5s `sleep()` would be enough to make sure it's there by the time you query it, but if (for whatever reason) it's slow to launch, then it might not be present in time. For debugging purposes, increasing this sleep to several seconds could be useful. If this does prove to the the problem, then we might want to look at a more robust solution by, e.g. waiting on a `threading.Condition` rather than simply doing a `time.sleep()`.

barentine, 2021-04-07:
Computer name should be fine
```
(base) Andrews-MacBook-Pro-5:~ Andrew$ scutil --get ComputerName
Andrew’s MacBook Pro

```

csoeller, 2021-04-07:
For a quick try I bumped the sleep to 10.5, and it seems to have succeeded:

```
$ PYMEClusterOfOne 
Setting icon
INFO:PYME.cluster.cluster_of_one:Launching data server: root=/Users/csoe002/PYMEData
INFO:PYME.cluster.cluster_of_one:Launching rule server
INFO:ruleserver:Starting ruleserver on 127.0.0.1:0
INFO:numexpr.utils:NumExpr defaulting to 8 threads.
INFO:root:========================================
PYMEDataServer, running on python 3.7.6 (default, Jan  8 2020, 13:42:34) 
[Clang 4.0.1 (tags/RELEASE_401/final)]

INFO:root:Serving from directory: /Users/csoe002/PYMEData
INFO:root:Serving HTTP on 127.0.0.1 port 65299 ...
INFO:PYME.cluster.cluster_of_one:Launching node server
dict_values(['http://127.0.0.1:65298/'])
DEBUG:nodeserver:Starting to poll for tasks
127.0.0.1 - - [07/Apr/2021 22:05:40] code 404, message No handler for /favicon.ico
nodeserver:INFO:Starting nodeserver on 127.0.0.1:0
INFO:nodeserver:Starting nodeserver on 127.0.0.1:0
DEBUG:nodeserver:Launching worker processors
```

csoeller, 2021-04-08:
A few more quick tests indicated 4.5 s still too short, 5.5 s appears to work reliably. It may have something to do with a slow home network? Will try again at work ASAP but the implementation may want a more robust check as David indicates above.

PR #1102 (2021-08-16): Cluster localization fixes (issue 1098)
Fixes #1098, fixes #990, closes #786 

3 for the price of one!

We were silently ignoring `startAt` when doing cluster (and cluster of one) based analyses. This fixes that, and at the same time  shifts everything to the rule classes in `cluster.rules` so we don't need to maintain two lots of code for pushing (meaning that we can finally close #786).

Whilst testing (newer OSX version with slower network IO :( ) , I also ran into #990, so this includes a fix for that as well (which turned out to be surprisingly simple - just moving the socket opening into [... cut]
closed 2021-08-16 by David-Baddeley