python-microscopy/python-microscopy #994: [BUG] nameUtils.numToAlpha wraps one-indexed instead of zero-indexed

barentine, 2021-04-08:
**Describe the bug**
nameUtils.numToAlpha(0) ->  'A'
nameUtils.numToAlpha(26) ->  'BA', should be 'AA'
nameUtils.numToAlpha(675) -> 'ZZ'
nameUtils.numToAlpha(676) -> 'BAA'

Say one (me!) tries to get back to numbers because they need to match something up, they might write
```
def alpha_to_num(alpha):
    res = [ord(x.lower()) - 96 for x in alpha]
    mag = (26 ** np.arange(len(res))[::-1])
    return np.sum(res * mag) - 1
```
which would work, if we zero-indexed numToAlpha, rather than whatever you want to call 1-indexing the leading digit unless it's the only digit and zero-indexing the rest

David-Baddeley, 2021-04-09:
This is deliberate behaviour to ensure a sane search order (if `numToAlpha(26)` evaluated to `'AA'` then series 26 would sort ahead of series_B et c... . It's equivalent to omitting the leading 0 on decimal numbers. ie, writing 0, 1, 2 rather than 00, 01, 02. The alternative would be to effectively zero-pad with `'A'`s, but that requires a-priori knowledge of how may series you are going to acquire in a given day, and any reasonably conservative choice gives gross, hard to read filenames.

closed 2021-04-09 by David-Baddeley

David-Baddeley, 2021-04-09:
ie your function should read

```python
def alpha_to_num(alpha):
    out = 0
    for i, x in enumerate(alpha.lower()[::-1]):
        out += (ord(x - 97))*(26**i)
   
    return out
```