python-microscopy/python-microscopy #996: [BUG] event io changes introduced bug, 'unknown name sp' introduced via merged PR  #995

csoeller, 2021-04-09:
**Describe the bug**
Run PYMEAcquire and click start, hitting error as below, apparently introduced in merged PR #995.
```
Traceback (most recent call last):
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/Acquire/ui/HDFSpoolFrame.py", line 475, in OnBStartSpoolButton
    self.spoolController.start_spooling(fn)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/Acquire/SpoolController.py", line 524, in start_spooling
    self.scope.frameWrangler.start()
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/Acquire/frameWrangler.py", line 455, in start
    self.cam.StartExposure()
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/Acquire/Hardware/Simulator/fakeCam.py", line 502, in StartExposure
    eventLog.logEvent('StartAq', '')
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/Acquire/eventLog.py", line 73, in logEvent
    evl.logEvent(eventName, eventDescr, timestamp = timestamp)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/IO/events.py", line 101, in logEvent
    ev['Time'] = sp.timeFcn()
NameError: name 'sp' is not defined
```

**To Reproduce**
Steps to reproduce the behavior as above.

**Expected behavior**
No bug.

Appears to be fixable by including the following statement in PYME.IO.events:

```python
import PYME.Acquire.Spooler as sp
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
macOS, latest PYME from 4/9/2021

**Additional context**
Add any other context about the problem here.

David-Baddeley, 2021-04-11:
Oops - managed to do that for the cluster backend but not for the direct HDF one. Should be a simple fix.

David-Baddeley, 2021-04-11:
Have a candidate fix - the error message goes away, but we seem to get a corrupt events table in the .h5 when streaming to file. Not sure that this is specific to any of the recent changes, so will check against an older version.

David-Baddeley, 2021-04-11:
event table corruption was also present in older code, so have merged #998. Will try and chase the corruption issue down, but suspect it might be specific to a) the simulator and b) my specific version of pytables.

PR #998 (2021-04-11): issue996
Fixes #996, also includes some tidying.
closed 2021-04-11 by David-Baddeley