python-microscopy/python-microscopy #1014: [ENHANCEMENT] PYMEVis animations: default rotation views, care about layer visibility changes

barentine, 2021-04-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when I try to make an animation. Looks like changes in layer visibility do not count as a part of a view, which is a bummer. Would also be nice to have some default rotations one could load (ones with enough steps not to mess up z).

zacsimile, 2021-04-16:
I agree these would be nice. Not sure what you mean by not messing up z, but you can get more steps by increasing the duration. Then you can just adjust fps in ImageJ when you make the animation.

barentine, 2021-04-16:
At least in the preview, if you do a 180 flip of an extended structure like a nucleus it rotates something which looks weirdly flat along the original view direction. FPS is also easy enough to set with ffmpeg, which I did after following your more step advice (the frame duration I can't adjust, because unfolding settings doesn't reveal anything for me).

zacsimile, 2021-04-16:
Hmm, this sounds like it might be two different errors. You need to pull the vertical bar way out after unfolding settings to see anything (weird wx bug), but the duration is actually in the table above settings--double-click on it to get a dialog.

[image]

David-Baddeley, 2021-04-18:
I suspect the rotation issues might be because we linearly interpolate between view positions / key frames. I think this is done in R3 on the view vectors, not in polar co-ordinates. This will give a reasonable approx of smooth rotation if your key frames are not too far apart in angular space, but could do something weird if you have key frames which are, e.g., 180 degrees apart. 

Adding layer visibility to the animation model would be nice, but surprisingly complex and hard to do cleanly. It also makes it much harder to e.g. save and reload set animation sequences.