python-microscopy/python-microscopy #1020: [SUPPORT] MeasureClusters3D output

lukasfue, 2021-05-03:
**What are you trying to do?**
Measure 3D clusters and output those measurements.

**Where are you stuck?**
I use the Pipeline Recipe Module "MeasureClusters3D", but I see no output once its finished. How can I find the output and save it as a .csv file?

David-Baddeley, 2021-05-03:
The recipes need a whole lot more documentation ... Is this in PYMEVis,  PYMEImage/DSView, or the standalone recipe runner, Bakeshop. Getting the output differs slightly in each case.

#### PYMEVis or PYMEImage

- go to the 'Recipe' Tab
- [**PYMEImage** only]: make sure the recipe has run by pressing F5 or choosing `Recipes->Run Recipe` from the menu
- Click on the desired output in the recipe display (indicated by the red oval in the image below - click on the text)
[image]
- This should launch a tabular viewer, with a save button in the upper left
[image]

#### Bakeshop (batch processing)

- add an `output.CSVOutput` module to the recipe and run

There are a few things in the pipeline / on the TODO list which should make this a little easier, such as being able to click on outputs/data in the collapsed pipeline view and open/save them. In terms of discoverability, It would probably also make sense to add options to save each intermediate dataset to the  `File` menu.

David-Baddeley, 2021-05-03:
PS - you can see if a recipe module has run because it's outline changes from black to green

lukasfue, 2021-05-03:
Okay, this is how I expected it to work. It appears it is not running, though I'm not sure why that is. Perhaps I am not using the correct parameters. I'm determining clusters by using 3D triangularization, thresholding and labeling, and "Get segmented IDs from image". That all seems to work okay and it seems the IDs of the clusters are under the labelKey "objectID" so that's what I used for that. Should I be using a different labelKey maybe?

Also, when I click Recipes, the only options I see are "Reconstruct from open image" and "Reconstruct from image file".

David-Baddeley, 2021-05-03:
I should have been clearer -  the run option on the recipes menu is only in PYMEImage (as PYMEVis recipes run automatically).

Could you give me a screenshot of the recipe tab, and a copy of the recipe text (if it's too long to see in the screenshot). I'm specifically interested if any of the recipe boxes have a red border  (which would indicate that that module had failed). Another option might be that one of the settings is such that there are no clusters detected. I'm not sure what happens with the viewer if there are 0 entries.

I think you'd get an error message and a red box if you had the label key wrong (you should definitely get a red box - although there is a chance that the error message gets hidden when in PYMEVis - a few improvements here are also on the TODO).

David-Baddeley, 2021-05-03:
This is what it would look like if you had the wrong labelKey:

[image]

- The MeasureClusters3D box has a red outline
- The error message is printed below (albeit truncated - this is where we need to put a bit of work into making it more visible)
- The module is shaded red in the left hand pipeline view

lukasfue, 2021-05-03:
My first attempt definitely failed but that's because I gave it the wrong input. The second attempt seems to have neither run nor failed so I'm not sure how to interpret it. I don't think its that there are no clusters detected as I can definitely visualize them. 

[image]

David-Baddeley, 2021-05-03:
Ahh - the recipes stop on the first failure - you need to delete the failing recipe module and then it will hopefully run fine. 

At this point, it's simplest to delete recipe modules in the recipe text on the right (then hit 'Apply Text Changes')

lukasfue, 2021-05-03:
Yep, that was it! It runs fine now. A part of me thought maybe that could be it as I posted the screenshot. Thanks for your help and patience, is appreciated!

closed 2021-05-03 by lukasfue

David-Baddeley, 2021-05-03:
No worries - it's actually really useful to see where we need to make things more intuitive - e.g. we really need a GUI way of deleting a recipe module (e.g. right clicking on it and having delete show up in a menu).

As a general strategy, it's best to modify the parameters of a failing recipe module so that it succeeds, rather than adding it again with new parameters, as a failing recipe module is always likely to generate a failing recipe. We should probably look at visually cueing the fact that a single module failure is essentially (from a user perspective) failure of the entire recipe.

lukasfue, 2021-05-04:
Yes, I agree with that. I had no idea once could delete things by deleting the text on the right.

Okay, lesson learned! I thought it was possible to right click a module to edit it, but that wasn't working in this case. Is the text-editing the best way to edit modules?

David-Baddeley, 2021-05-04:
*left* clicking a recipe module should bring up a dialog which lets you open it. You can also expand the module in the condensed data pipeline on the left