python-microscopy/python-microscopy #1027: [BUG] f

csoeller, 2021-05-07:
**Describe the bug**

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10]
 - Python version:
 - numpy version:
 - wxpython version [if a UI bug]:

**Additional context**
Add any other context about the problem here.

David-Baddeley, 2021-05-07:
As a temporary workaround, click in the centre of the image and then try scaling. The issue is that the scaling is now calculated based on a ROI around the current selected position (which defaults to (0,0) when opening an image.

csoeller, 2021-05-07:
Ok, sort of works but depending where I click (e.g. region with little staining) still blows up. Something more robust would eventually be nice (or make ROI big by default) - not sure the average user would know what is going on. See examples below:

[image]

[image]

David-Baddeley, 2021-05-08:
Note, there will still be times when the subsampling makes the display scaling a little off for sparse data.

PR #1028 (2021-05-08): revert display scaling changes for non chunked datasets
fixes #1027
closed 2021-05-08 by David-Baddeley