python-microscopy/python-microscopy #1032: [BUG] error fitting Gaussian to line profile in dh5view

csoeller, 2021-05-12:
**Describe the bug**
Trying to run a Gaussian fit on a profile extracted from an image in dh5view I get an error message as below. I suspect this may be a side effect of the xyztc data model changes but not totally sure.

**To Reproduce**
Steps to reproduce the behavior:
1. extract intensity profile in dh5view
2. Select `Fitting>Gaussian`
4. See error

**Error message**

```
 Error whilst running Fitting>Gaussian
==============================================
python-microscopy=21.03.10
python=3.7.6, platform=darwin
numpy=1.19.2, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/ui/progress.py", line 121, in func
    return fcn(*args, **kwargs)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/DSView/modules/profileFitting.py", line 67, in OnGaussianFit
    plt.plot(xv, I, cols[chan] + 'x', label=self.image.names[chan])
  File "/Users/csoe002/opt/anaconda2/envs/pyme-py3/lib/python3.7/site-packages/matplotlib/pyplot.py", line 2763, in plot
    is not None else {}), **kwargs)
  File "/Users/csoe002/opt/anaconda2/envs/pyme-py3/lib/python3.7/site-packages/matplotlib/axes/_axes.py", line 1647, in plot
    lines = [*self._get_lines(*args, data=data, **kwargs)]
  File "/Users/csoe002/opt/anaconda2/envs/pyme-py3/lib/python3.7/site-packages/matplotlib/axes/_base.py", line 216, in __call__
    yield from self._plot_args(this, kwargs)
  File "/Users/csoe002/opt/anaconda2/envs/pyme-py3/lib/python3.7/site-packages/matplotlib/axes/_base.py", line 342, in _plot_args
    raise ValueError(f"x and y must have same first dimension, but "
ValueError: x and y must have same first dimension, but have shapes (112,) and (1,)

```

**Expected behavior**
Plot fitted Gaussian, no error.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
- python-microscopy=21.03.10
- python=3.7.6, platform=darwin
- numpy=1.19.2, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

**Additional context**
Add any other context about the problem here.

David-Baddeley, 2021-05-14:
Thanks! should now be fixed

closed 2021-05-14 by David-Baddeley

csoeller, 2021-05-14:
Great, thanks for prompt fixing! I confirm all working now.

PR #1034 (2021-05-14): Profile fitting fix
fixes #1032