python-microscopy/python-microscopy #1035: [BUG] Downloading Distance transform colocalization raw histogram data into .txt file.

Carmen-Rodilla, 2021-05-14:
I am trying to save the raw histogram data from the Distance transform colocalization of the coloc module as indicated in the page (https://www.python-microscopy.org/doc/howtos/distancecolocalization.html) for further quantification. After download, the document appears to be empty. Is there any other way -apart from the described in the page- to get the data? 
Thank you!

zacsimile, 2021-05-14:
I've looked into this a bit. Following steps in https://www.python-microscopy.org/doc/howtos/distancecolocalization.html. When I try to use the signal threshold, I get

```
Traceback (most recent call last):
  File "c:\users\zrc4\code\python-microscopy\PYME\DSView\DisplayOptionsPanel.py", line 296, in OnSignalFracThresh
    t = thresholding.signalFraction(self.do.ds[:,:,:,i], frac)
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\DataSources\BaseDataSource.py", line 291, in __getitem__
    raise RuntimeError('Must provide a 5-dimensional slice')
RuntimeError: Must provide a 5-dimensional slice
```

When I hit `File > Save` and try to save the histogram txt, I get the following traceback.

```
Traceback (most recent call last):
  File "c:\users\zrc4\code\python-microscopy\PYME\DSView\dsviewer.py", line 324, in OnSave
    self.image.Save()
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\image.py", line 1252, in Save
    self.filename = dataExporter.ExportData(self.data_xyztc, self.mdh, self.events, defaultExt=self.defaultExt, filename=filename, progressCallback=progressCallback)
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\dataExporter.py", line 647, in ExportData
    exp.Export(ds, filename, slice(0, ds.shape[0], 1), slice(0, ds.shape[1], 1), slice(0, ds.shape[2], 1),mdh, events, origName, progressCallback=progressCallback)
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\dataExporter.py", line 429, in Export
    dat = [data[xslice, yslice, zslice, chan].squeeze() for chan in range(data.shape[3])]
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\dataExporter.py", line 429, in <listcomp>
    dat = [data[xslice, yslice, zslice, chan].squeeze() for chan in range(data.shape[3])]
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\DataSources\BaseDataSource.py", line 291, in __getitem__
    raise RuntimeError('Must provide a 5-dimensional slice')
RuntimeError: Must provide a 5-dimensional slice
```

If I fix these two, the histogram saves, but I still get the following traceback. I'm not sure how to fix this one as I think maybe `XMLMDHandler` isn't used much anymore?

```
Traceback (most recent call last):
  File "c:\users\zrc4\code\python-microscopy\PYME\DSView\dsviewer.py", line 324, in OnSave
    self.image.Save()
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\image.py", line 1252, in Save
    self.filename = dataExporter.ExportData(self.data_xyztc, self.mdh, self.events, defaultExt=self.defaultExt, filename=filename, progressCallback=progressCallback)
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\dataExporter.py", line 647, in ExportData
    exp.Export(ds, filename, slice(0, ds.shape[0], 1), slice(0, ds.shape[1], 1), slice(0, ds.shape[2], 1),mdh, events, origName, progressCallback=progressCallback)
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\dataExporter.py", line 454, in Export
    xmd = MetaDataHandler.XMLMDHandler(mdToCopy=metadata)
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\MetaDataHandler.py", line 766, in __init__
    self.copyEntriesFrom(mdToCopy)
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\MetaDataHandler.py", line 398, in copyEntriesFrom
    self.setEntry(en, mdToCopy.get(en))
  File "c:\users\zrc4\code\python-microscopy\PYME\IO\MetaDataHandler.py", line 817, in setEntry
    node.setAttribute('value', base64.b64encode((pickle.dumps(value))).decode('ascii'))
_pickle.PicklingError: Can't pickle <class 'multiprocessing.sharedctypes.c_double_Array_11088000'>: attribute lookup c_double_Array_11088000 on multiprocessing.sharedctypes failed
```

zacsimile, 2021-05-14:
The last traceback is a bit moot, since it's not necessary to fix it to fix the error here, but I'm not quite sure what to do. This error stems from an interaction between the metadata handler and `PYME.util.shmarray.shmarray.shmarray`. Since the metadata handler has no problem with `np.array`, I kind of want to just cast the `shmarray` to an `np.array` in the metadata assignment on lines 319/320 of `coloc.py`, but is it better/safer in the long run to change how `shmarray` returns itself?

barentine, 2021-05-14:
Sorry to be chiming in briefly without having really read through the context @zacsimile , but I think you probably just want to cast the array to list.

zacsimile, 2021-05-14:
Thanks, @barentine! I suspect either will be fine, since `im.mdh['Profile.XValues'] = im.xvals` on line 313 of `coloc.py` is an `np.array`.

zacsimile, 2021-05-14:
Update: it does not like `list`. `np.array` casting does work, though.

David-Baddeley, 2021-05-17:
It's super weird for the `shmarray` stuff to come up now (potentially because I've reduced the number of times we pass by value rather than reference). In any case, a cast to a numpy array should handle it, although I'm wondering if we'll also hit this anywhere else.  In either case, putting the xvals in the metadata is a dirty hack, and something that might want to be revisited at some point.

David-Baddeley, 2021-05-17:
Can we verify that this is fixed with #1036?

zacsimile, 2021-05-18:
It works for me now. Does it work for you, @Carmen-Rodilla?

csoeller, 2021-05-18:
@zacsimile, would it be possible to make the data available that you used for the testing in https://github.com/python-microscopy/python-microscopy/pull/1036#issuecomment-841521033. I would like to test as well and use something that can be used for future checks of expected behaviour?

zacsimile, 2021-05-18:
Yep, I'll email it to you.

Carmen-Rodilla, 2021-05-19:
Hi! 
It looks like it is still not working, should I do some update in the programme or something like that?

zacsimile, 2021-05-19:
Hi @Carmen-Rodilla! Sorry to hear that. Right now, this will only work in a development install (https://python-microscopy.org/doc/Installation/Installation.html#development-installs). Assuming you have one, you can cd to the development install folder and run `git pull`, which will update the code and let you try out this fix.

csoeller, 2021-05-19:
@zacsimile and @David-Baddeley, it would be good to give some guidance what the options are to pick up changes when you went down the one-click installer route. Does one need to wait for an updated installer and install over the top for non-specialists? I am asking since I have been recommending the one-click installer to a few people now and anticipate getting to the stage where I need to tell them how to get a fix for a problem they encountered.

zacsimile, 2021-05-19:
I'm still pushing for a GUI-based updater (#483). Right now, you have to periodically check our website for new builds. Once one is available, you can go into the command line in the installed environment and run `conda update python-microscopy`. Current instructions for this are here: https://python-microscopy.org/doc/Installation/Installation.html#updating.

csoeller, 2021-05-19:
Cool, thanks @zacsimile. How do you find out if a new build is available? Is that when a new release was tagged in the github repo or similar? Can one figure out up to which commits are included?

zacsimile, 2021-05-19:
You can check https://github.com/python-microscopy/python-microscopy/releases. You can hit compare to see which commits are included. 

We could start notifying people of new releases on image.sc (I often see announcements for new releases) and maybe even make a PYME twitter and post it there.

zacsimile, 2021-05-19:
One other note: you can't "install over the top" the way our packaged installers work, you have to go the `conda update` route. Each version of PYME has a different name, so rather than replacing a `python-microscopy` folder you will simply have `python-microscopy 21.03.10` and `python-microscopy 21.03.03` installed, and you'll have desktop shortcuts for both. You can do this, you just have to remember to use the latest one. I would personally like to change this so it's always just called `python-microscopy` and the version number is stored separately from the name.

barentine, 2021-05-19:
Ideally our reply to 'should I update, and how?' would be a link to a page with clear instructions

csoeller, 2021-05-19:
Thanks @zacsimile and @barentine for feedback. I won't hijack this issue for too much longer but rather probably raise a separate issue on this. The sign of a maturing package is that increasing amounts of effort need to be spent on polishing the support infrastructure rather than more exciting developments. Let's get @David-Baddeley 's input on the direction this could take, a good doc page with instructions might be a starting point.

David-Baddeley, 2021-05-19:
Short answer, is once you have an executable version installed, you should be able to update with `conda update python-microscopy`. You will need updated conda packages for that .Our packages are a bit behind the bleeding edge at the moment (partly on purpose so that some of the recent changes like the one which introduced this bug can shake out a little before we push them onto everyone), but it feels like it's getting time for another release.

@Carmen-Rodilla - what version are you running. The traceback and error mentioned by Zach is specific to the development version. **If** you are getting the same traceback as Zach, you will be on the development version, not a packaged version, which means that the standard development update procedure of:

```bash
git pull origin master
python setup.py build_ext -i
```
should be followed.

If you didn't do a development install, this could be a different error, and it would be great to know how you installed and what version you have (Help->About).

zacsimile, 2021-05-20:
@David-Baddeley I also tested this in the packaged version and I'm pretty sure the error is a bytes/string error (a py3 one we missed), which I also fixed here.

zacsimile, 2021-05-20:
@Carmen-Rodilla You can work with your version of miniconda, but I suspect it will take a bit longer for this fix to make it into a compiled package. In the meantime, I recommend creating a separate environment in your miniconda for PYME's development install so you can get this fix quickly. Open a miniconda prompt and execute the following.

```
conda create -n pyme_dev pyme-depends python=3.6
conda activate pyme_dev

git clone https://github.com/python-microscopy/python-microscopy.git
cd python-microscopy
python setup.py develop
```

After this you should be able to run the version of `pymevis` with this fix available.

David-Baddeley, 2021-08-03:
I think we can close this now ... feel free to re-open if needed.

closed 2021-08-03 by David-Baddeley