python-microscopy/python-microscopy #1041: [ENHANCEMENT] get some clear instructions up somewhere how to update the one-click installer based installs

csoeller, 2021-05-19:
**Is your feature request related to a problem? Please describe.**
I am increasingly recommending that people use the one-click installer for PYME. The issue is that I am not sure where to point them on
1. how to update that install when fixes are made available
2. how to find out that fixes have been made available

**Describe the solution you'd like**
Initially a doc page to link to on these questions would be a good starting point. More longer term a simpler way friendly to non-experts would be desirable, possibly along the line of @zacsimile's #483 or something different altogether.

**Describe alternatives you've considered**
If we say in the PYMEVisualise paper we would like to become the ImageJ of SMLM data processing then we will eventually have to bite this bullet.

**Are you prepared to work on it?**
Certainly, once a strategy has been agreed.

David-Baddeley, 2021-05-19:
updating is covered briefly at 

https://python-microscopy.org/doc/Installation/Installation.html#updating

but could potentially use a little expanding (mostly to explain what to do when `conda update python-microscopy` fails).

Update **notification** has been on the TODO list for ages, and the proposed implementation is sketched out in some detail in #483.

In short:

- when we make a release, update a file on python-microscopy.org which gives the current release number 
- request this file on  program startup, and check against the currently running pyme version
- show a dialog that says an update is available, and gives instructions on how to update. This will probably initially involve running `conda update python-microscopy`, or `conda update -S python-microscopy` in the console to avoid issues over-writing loaded, compiled, modules, but it would be nice to make this more seamless.

Basically we just need to write it, and decide how annoying we should be if someone skips an update (i.e. how long do we wait before telling them again that an update is available).

duplicates #483

David-Baddeley, 2023-09-18:
I think we can close this with update notification now in

closed 2023-09-18 by David-Baddeley