python-microscopy/python-microscopy #1044: [BUG] Can't display PYME.Acquire.xyztc.MemoryBackend.image in ViewIm3D in Py38/wx4.1.1

zacsimile, 2021-05-22:
**Describe the bug**
If I call `ViewIm3D` on this memory backend image after acquiring, the display is all white and won't render, and I get the following traceback.

```
Traceback (most recent call last):
  File "c:\users\scope_baby\code\python-microscopy\PYME\DSView\scrolledImagePanel.py", line 58, in OnPaint
    self.renderer(MemDC)
  File "c:\users\scope_baby\code\python-microscopy\PYME\DSView\arrayViewPanel.py", line 513, in DoPaint
    self.DrawScaleBar(self, dc)
  File "c:\users\scope_baby\code\python-microscopy\PYME\DSView\arrayViewPanel.py", line 347, in DrawScaleBar
    dc.DrawLine(x0, y1, x1, y1)
TypeError: DC.DrawLine(): arguments did not match any overloaded call:
  overload 1: argument 1 has unexpected type 'numpy.float64'
  overload 2: argument 1 has unexpected type 'numpy.float64'
```

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: win10
 - Python version: 3.8.8
 - numpy version: 1.20.3
 - wxpython version [if a UI bug]: 4.1.1

David-Baddeley, 2021-05-23:
This looks like it'd be fixable with some casts. Might indicate a problem with metadata, rather than display though (so might be happening a long way upstream).

David-Baddeley, 2021-05-23:
Can't easily see how the parameters end up being numpy floats - they should be python floats (although it's possible that wx4.1.1 wants int here).

zacsimile, 2021-05-23:
I don't get this error with any other image I pass to dh5view, just the memory backend one.

zacsimile, 2021-05-23:
`view.voxelsize` is somehow coming through as `VoxelSize(x=0.0, y=0.0, z=200.0)`.  Looked into it and do I need to set the voxel size for this camera in settingsdb somewhere?

David-Baddeley, 2021-05-23:
`Camera->Set Pixel Size` on the menu to set the pixel size.

Not sure why it's defaulting to 0.,0.,200. though ...

David-Baddeley, 2021-05-23:
On the simulator, I also get `numpy.float64` for the metadata type, with the voxelsize set

David-Baddeley, 2021-05-23:
#1046 should fix the display issue (although I'm still not sure how you are getting a 0.,0.,200. voxelsize).

zacsimile, 2021-05-23:
Thanks! Does it make sense to store the `VoxelSize` of the camera with the `noiseProps`? That's serial number-specific. What do you think of making a `cameraProps` variable and shifting `noiseProps` and the voxel size entirely to `settingsDB` or entirely to the code?

David-Baddeley, 2021-05-23:
`VoxelSize` should not be stored with `noiseProps` - it's a function of the entire microscope system (camera + tube lens + objective), **not** the camera. Some microscopes have switchable objectives, some microscopes have switchable tube lenses, and you might want to move a camera from one microscope (or one side port) to another (both of which are likely to result in magnification changes). If you notice, the current settingsDB config lets you have multiple pixel sizes per camera to cope with different light paths.

Moving noiseprops out of the camera classes is subject of a long-standing TODO - I'm pretty sure there's an issue for it somewhere, along with an implementation sketch.  We could think about adding the camera plane (as opposed to sample plane) pixel size to that so that the sample pixel size could be computed, but I'm not sure it's worthwhile. In general it's not possible to know the magnification of a particular tube-lens objective pair without measuring it, and it's simplest to just measure an effective pixel size for entire system and record that.

David-Baddeley, 2021-05-23:
PS - I don't think noiseprops should go into settingsDB. It should be in it's own database/yaml, and preferably human readable. In a lab usage scenario, you will often have multiple cameras, which might get swapped between instruments. Ideally you'd have a central noiseprops "database" (or, e.g., version-controlled .yaml) with the properties of all your cameras in it.

David-Baddeley, 2021-05-23:
There is a bunch of discussion on how `noise_properties` should be refactored in the discussion around #144. It's just a matter of fleshing out the details and doing it.

csoeller, 2021-05-24:
Hi both, I'll have a stab at a `noise_properties` refactoring along the lines of what I have made as a mock interface in https://github.com/csoeller/PYME-camprops-mock in the next couple of months, unless someone wants to beat me to it. Not quite sure how this relates to the #144 discussion.

zacsimile, 2021-05-24:
Thanks, @csoeller! Good point, @David-Baddeley, I was thinking of storing camera pixel size in settingsDB and then putting objective magnification, etc. in a separate repo, but this would be come quite annoying quite quickly for scanning microscopes and/or scopes with objectives plus additional magnifications. 

What would be nice is to be prompted for `VoxelSize` if you're using a new camera for the first time. It's been so long since I set up a camera I'd totally forgotten about this aspect.

zacsimile, 2021-05-24:
The `VoxelSize` thing is my fault for caching the binning and passing a default value of `0` instead of `1`. But hey, never would have found this issue if I hadn't done something dumb. You're welcome.

PR #1046 (2021-05-25): Cast to int in dc.DrawLine (scalebar)
fixes #1044\n doesn't address why voxelsize is a np.float64 rather than a straight float, but this shouldn't matter. More robust with cast in any case
closed 2021-05-25 by David-Baddeley