python-microscopy/python-microscopy #1057: [BUG] Recipe modules don't read incident data source information on initial add

zacsimile, 2021-06-07:
I press add module and I'll add, e.g., a `FilterTable`, and then select my input data source. If then right-click and Add a Filter, no keys are available.

[image]

If I press "OK" and close the `FilterTable` and then double-click it again, now the keys populate.

[image]

I'd like them to populate when I select the input data source the first time.

David-Baddeley, 2021-06-07:
This is because the data source for the filter editor is set at creation time (see the `dataSource` parameter in custom_traits_editors.py L 84).  The FilterPanel can also take a callable which returns the datasource, which means that a solution like this might work:

`dataSource = lambda : self.factory.datasource`

David-Baddeley, 2021-06-07:
key is to delay the datasource access to when the `filterPanel` calls `dataSource()`

David-Baddeley, 2021-06-08:
unfortunately the above doesn't work. Neither does `dataSource = lambda : getattr(self.factory, 'datasource')`

David-Baddeley, 2021-07-23:
Getting warmer, but not with an obvious solution - The core problem is that the editor updates the input datasource when it closes - hence the input is not defined until things have closed.

commit a7aa3b92 (2021-11-24): finally fix #1057
closed 2021-11-24 by David-Baddeley