python-microscopy/python-microscopy #1064: [BUG] Getting different results from FilterTable on different machines

xzlao, 2021-06-14:
I'm getting different results when applying `FilterTable` on different Windows 10 machines. Both run the latest development version of PYME.

[image]

is the result for the following configuration 1.

<details>
<summary>Configuration 1</summary>
<pre>
# packages in environment at C:\Users\xl592\Miniconda3\envs\pyme:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                   py36_0
babel                     2.9.0              pyhd3eb1b0_0
backports                 1.0                pyhd3eb1b0_2
backports.lzma            0.0.14           py36hdc8397c_0
blas                      1.0                         mkl
blosc                     1.19.0               h7bd577a_0
brotlipy                  0.7.0           py36h2bbff1b_1003
bzip2                     1.0.8                he774522_0
ca-certificates           2020.12.8            haa95532_1
cairo                     1.14.12              hf171d8a_3
certifi                   2020.12.5        py36haa95532_0
cffi                      1.14.4           py36hcd4344a_0
chardet                   3.0.4           py36haa95532_1003
cheroot                   8.5.0            py36haa95532_1
cherrypy                  18.6.0           py36haa95532_0
cloudpickle               1.6.0                      py_0
conda                     4.9.2            py36haa95532_0
conda-package-handling    1.7.2            py36h76e460a_0
cryptography              3.3.1            py36hcd4344a_1
cycler                    0.10.0           py36haa95532_0
cython                    0.29.21          py36hd77b12b_0
cytoolz                   0.11.0           py36he774522_0
dask-core                 2.30.0                     py_0
decorator                 4.4.2              pyhd3eb1b0_0
docutils                  0.16                     py36_1
fftw                      3.3.8                he774522_3
freetype                  2.10.4               hd328e21_0
future                    0.18.2                   py36_1
hdf5                      1.10.4               h7ebc959_0
icc_rt                    2019.0.0             h0cc432a_1
icu                       58.2             vc14hc45fdbb_0  [vc14]  anaconda
idna                      2.10               pyhd3eb1b0_0
ifaddr                    0.1.6                      py_0    david_baddeley
imageio                   2.9.0              pyhd3eb1b0_0
imagesize                 1.2.0              pyhd3eb1b0_0
importlib-metadata        2.0.0                      py_1
importlib_metadata        2.0.0                         1
importlib_resources       3.3.0            py36haa95532_0
intel-openmp              2019.4                      245
jaraco.classes            3.1.0                      py_0
jaraco.collections        3.0.0                      py_0
jaraco.functools          3.0.1                      py_0
jaraco.text               3.2.0                    py36_1
jinja2                    2.11.2             pyhd3eb1b0_0
joblib                    0.17.0                     py_0
jpeg                      9b               vc14h4d7706e_1  [vc14]  anaconda
kiwisolver                1.3.0            py36hd77b12b_0
libpng                    1.6.37               h2a8f88b_0
libtiff                   4.1.0                h56a325e_1
lz4-c                     1.9.2                hf4a77e7_3
lzo                       2.10             vc14h0a64fa6_1  [vc14]  anaconda
markupsafe                1.1.1            py36he774522_0
matplotlib                3.2.2                         0
matplotlib-base           3.2.2            py36h64f37c6_0
menuinst                  1.4.16           py36he774522_1
mkl                       2020.2                      256
mkl-service               2.3.0            py36h196d8e1_0
mkl_fft                   1.2.0            py36h45dec08_0
[... 219 lines cut ...]
threadpoolctl             2.1.0              pyh5ca1d4c_0
tk                        8.6.10               he774522_0
toolz                     0.11.1             pyhd3eb1b0_0
toposort                  1.5                      py36_3    david_baddeley
tornado                   6.1              py36h2bbff1b_0
tqdm                      4.54.1             pyhd3eb1b0_0
traits                    6.1.1            py36he774522_0
traitsui                  7.1.0                      py_0
ujson                     4.0.1            py36ha925a31_0
urllib3                   1.25.11                    py_0
vc                        14.2                 h21ff451_1
vs2015_runtime            14.27.29016          h5e58377_2
wheel                     0.36.1             pyhd3eb1b0_0
win_inet_pton             1.1.0            py36haa95532_0
wincertstore              0.2              py36h7fe50ca_0
wxpython                  4.0.4            py36ha925a31_0
xz                        5.2.5                h62dcd97_0
yaml                      0.2.5                he774522_0
zc.lockfile               2.0                        py_0
zeroconf                  0.24.0                   py36_3    david_baddeley
zipp                      3.4.0              pyhd3eb1b0_0
zlib                      1.2.11           vc14h1cdd9ab_1  [vc14]  anaconda
zstd                      1.4.5                h04227a9_0
</pre>
</details>

xzlao, 2021-06-14:
Simply removing `pyme-depends` while keeping `pyme-depends-strict` fixes this issue. However, all the other package versions are the same...

David-Baddeley, 2021-06-14:
This is super, super odd. There is no explanation of why removing pyme-depends whilst keeping pyme-depends-strict should make any difference, especially if all the package versions remain the same. The only thing I can think of is that there is something stochastic in the import or processing pipeline, and that the observed behaviour was a stochastic fluke rather than some real difference in the installs. I'm grasping at straws, but it might be useful to know the following:

- What is the input file format (I'm guessing .mat or .csv)?
- Are the results repeatable over multiple iterations on one machine?
- `MapVutaraErrorX` is presumably from a plugin - I imagine this is just a simple mapping so would be surprised if this introduced stochasticity, but would be equally surprised if any of the other pipeline steps were stochastic.
- Did you re-run `python setup.py build_ext -i` or `python setup.py develop` after removing pyme-depends? We have made some recent (~3 months ago) bugfixes in to the c code used by FindClumps, and running an older compiled version might explain the differences (even if the actual code has been updated).

zacsimile, 2021-06-14:
Input file is `.csv`. Will try on another machine shortly. `MapVutaraErrorX` is a plugin and is non-stochastic, but I will share the code with you. I did rerun `python setup.py develop` after re-installing, but I believe the problem was on a fresh install of PYME. @xzlao, did you install PYME on that machine before today?

xzlao, 2021-06-14:
No, I installed PYME on that machine for the first time on 3pm (EST time),
June 9. Then I found out the result is different from that from another
machine, on which I installed PYME on June 4.

zacsimile, 2021-06-14:
OK, let me see if I can replicate this on one of my Windows machines and we'll go from there.

csoeller, 2021-06-15:
One strategy to debug further would be to start taking out variables you filter for from the FilterTable to see which one triggers the behaviour, e.g. does it seem to be `ClumpSize` etc. I admit it is a little bit weird but could be strange floating point precision etc including upstream change in a numpy module (although not sure if every package was the same version in both installs...).

zacsimile, 2021-06-15:
I am able to replicate this exact problem on my Windows machine if I execute the following commands.

```
conda config --append channels anaconda
conda config --add channels david_baddeley
conda create -n pyme python=3.6 pyme-depends-strict python-microscopy

git clone https://github.com/python-microscopy/python-microscopy.git
cd python-microscopy
python setup.py develop
```

I believe, in this case, I was able to fix the problem by removing and then re-compiling `python-microscopy` from the source code. If I instead execute

```
conda config --append channels anaconda
conda config --add channels david_baddeley
conda create -n pyme python=3.6 pyme-depends-strict

git clone https://github.com/python-microscopy/python-microscopy.git
cd python-microscopy
python setup.py develop
```

I do not get this problem. As such, I do not believe it is at all related to `pyme-depends` vs. `pyme-depends-strict`.

@xzlao, in the future, when you install pyme, follow the development install instructions only and disregard the "Installing using conda" section. Simply install miniconda, open a miniconda command prompt, and execute only the following.

```
conda config --add channels david_baddeley
conda create -n pyme pyme-depends python=3.6
conda activate pyme

git clone https://github.com/python-microscopy/python-microscopy.git
cd python-microscopy
python setup.py develop
```

This should prevent you from running into this problem again.

xzlao, 2021-06-15:
Sounds good! Thanks.

csoeller, 2021-06-15:
I guess this shows how easily things can go wrong with little misunderstandings. While the suggestion to avoid doing things in this way is very rerasonable it would be a good idea to come up with a check that detects this being attempted, i.e. development install on top of full conda install and warns appropriately...

David-Baddeley, 2021-06-15:
In addition to being quite a lot of work and a maintenance nightmare,  I think doing explicit checks on the installation environment for development installs is far too much handholding for the people we *should* be targeting for development installs. My expectation is that development installs are performed by people with a reasonable working knowledge of python and python packaging and who want to hack. I realise that we currently also use development installs to push bugfixes to people in between releases, but I really want to move away from dev installs as the default.

Ultimately this issue probably a bug in either conda or setuptools. A package installed using `python setup.py develop` or `python setup.py install` **should** override a package which is already installed using a package manager. This doesn't seem to be happening in this case. If the much better resourced teams at conda and setuptools can't get it right, I don't think we have much chance of patching around it and/or checking in a robust and maintainable way. Our setup instructions do warn that all previous installs should be removed, and this should be enough for the target dev install audience.

On a more constructive note, there is a major improvement that we could make that would help when debugging issues such as this, namely to report install type in `PYME.version` and in the about dialogs. As it stands, a development install is indistinguishable from the last official release when you look at `PYME.version`. A simple solution would be to use the install type detection we've written for update checking and append `dev<commithash>` to the version number if we are running a development install.

David-Baddeley, 2021-06-15:
commit hash logic is in `update_version.py`, to make scheme at least semi compatible with PEP440 (https://www.python.org/dev/peps/pep-0440/) we might need to do, e.g. `.post0.dev<hash>`, or omit the hash completely and make it available as a separate member of the version module.

David-Baddeley, 2021-08-03:
I think we can call the initial issue resolved, leaving open and changing to an enhancement to display install type (and potentially commit hash) in about dialog / error messages to make debugging easier in the future.

PR #1154 (2021-11-26): version logic updates
Improved version reporting for development installs, closes #1064
closed 2021-11-26 by David-Baddeley