python-microscopy/python-microscopy #1068: [BUG] point set generation - image.openImages not in name space, missing import

csoeller, 2021-06-17:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure point generation with image source
2. Generate fluorophore positions and events
4. See error

**Expected behavior**
Just do as told.

**Screenshots**
N/A

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
- python-microscopy=21.03.10
- python=3.7.6, platform=darwin
- numpy=1.19.2, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

**Error message**

```
 Error whilst running Extras>Synthetic Data>Generate fluorophore positions and events
==============================================
python-microscopy=21.03.10
python=3.7.6, platform=darwin
numpy=1.19.2, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/ui/progress.py", line 121, in func
    return fcn(*args, **kwargs)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/LMVis/Extras/pointSetGeneration.py", line 138, in OnGenPoints
    self.OnGenEvents(None)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/LMVis/Extras/pointSetGeneration.py", line 170, in OnGenEvents
    pipeline.imageBounds = image.openImages[self.source.image].imgBounds
NameError: name 'image' is not defined

```

David-Baddeley, 2021-06-21:
Can we call this fixed?

csoeller, 2021-06-22:
When you put back the `ed._values_changed()` check back in you check for a `ValueError` while the actual error raised is a `TypeError`. Therefore the current committed and merged code still does not work. Can you change the code as shown below? I believe after that we can close the issue.

In addition, I run into OpenGL errors when points are generated in an empty `visGUI`. This is not an issue if you generate the points in a `visGUI` instance that already shows events prior to generating the synthetic data. I probably better raise a separate issue for that?

[image]

David-Baddeley, 2021-06-22:
Sorry, I thought I'd done a 1:1 revert, but have been a bit distracted with a Marsden submission.

csoeller, 2021-06-22:
No worries, fully understandable. I suggest I close this as soon as we have the fix in and I had a chance to test briefly.

PR #1072 (2021-06-23): proper TypeError fix
fixes #1068 (properly this time)
closed 2021-06-23 by David-Baddeley