python-microscopy/python-microscopy #1082: [BUG] data handling in measure.py needs updating to new 5D data model

csoeller, 2021-07-09:
**Describe the bug**
Trying to run the `Measure2D` module on image data currently raises a bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Open mask image with `dh5view`
2. Add `Label` and `Measure2D` modules
3. Execute recipe
4. See bug

**Expected behavior**
Not raise bug.

**Error message**

```
Traceback
=========
Traceback (most recent call last):
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/ui/progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/DSView/modules/recipes.py", line 98, in RunCurrentRecipe
    self.outp = self.activeRecipe.execute(input=self.image)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/recipes/recipe.py", line 203, in execute
    m.execute(self.namespace)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/recipes/measurement.py", line 741, in execute
    m.addFrameMeasures(i, *self._measureFrame(i, labels, intensity))
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/recipes/measurement.py", line 750, in _measureFrame
    li = labels.data[:,:,frameNo].squeeze().astype('i')
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/IO/DataSources/BaseDataSource.py", line 171, in __getitem__
    col_indices =  range(*keys[3].indices(self.shape[3]))
IndexError: list index out of range
```

**Additional remarks**

The following patch makes things work but I am not confident it is the right solution:

```patch
diff --git a/PYME/recipes/measurement.py b/PYME/recipes/measurement.py
index 0c2e3eddd..c3cab0f3c 100644
--- a/PYME/recipes/measurement.py
+++ b/PYME/recipes/measurement.py
@@ -747,10 +747,10 @@ class Measure2D(ModuleBase):
     def _measureFrame(self, frameNo, labels, intensity):
         import skimage.measure
         
-        li = labels.data[:,:,frameNo].squeeze().astype('i')
+        li = labels.data_xyztc[:,:,:,frameNo].squeeze().astype('i')
 
         if intensity:
-            it = intensity.data[:,:,frameNo].squeeze()
+            it = intensity.data_xytc[:,:,frameNo].squeeze()
         else:
             it = None
```

In relation to the patch above (regardless if it is the right way to patch or not) I made the following observations which make me think I really do not understand how the new datamodel is supposed to be used:

1. Certain ways to index a slice break when using the `data_xyzt` and `data_xyzct` access which I thought should work.
2. Which ones break was different for different images in a way that was not obvious to me.

Indexing the input image (mask image):

```python
>>> ﻿﻿image.data_xytc[:,:,0]
array([[0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0],
       ...,
       [0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0],
       [0, 0, 0, ..., 0, 0, 0]], dtype=uint8)
>>> image.data_xyztc[:,:,:,0] # not clear why this one fails - trailing dims can be omitted in numpy slices which just assumes :
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/IO/DataSources/BaseDataSource.py", line 291, in __getitem__
    raise RuntimeError('Must provide a 5-dimensional slice')
RuntimeError: Must provide a 5-dimensional slice
>>> image.data_xyztc.shape
(1404, 1558, 1, 1, 1)
```

Indexing the labels image (generated by the `Label` module):

```python
>>> image.data_xytc.shape
(1404, 1558, 1, 1)

>>> image.data_xytc[:,:,0] # not clear why this one fails
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/IO/DataSources/BaseDataSource.py", line 171, in __getitem__
    col_indices =  range(*keys[3].indices(self.shape[3]))
IndexError: list index out of range

>>> image.data_xyztc[:,:,:,0] # whereas this one does not fail
array([[[0],
        [0],
.
.
.
        [0],
        [0]]], dtype=int32)

```

**Screenshots**

[image]

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
- python-microscopy=21.07.07
- python=3.7.6, platform=darwin
- numpy=1.19.2, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

**Additional context**
Add any other context about the problem here.

David-Baddeley, 2021-07-09:
I'm pretty sure this occurs when you a) load 2D data in a format (.tiff, .h5) for which the raw datasource is now an XYZTC datasource, and then use the nominally  backwards compatible "demotion" to XYTC (`image.data`, `image.data_xytc`) - i.e. it's a bug in this demotion process. 

As to the specific failures:
- Accessing `data_xyztc` with only 4 slices is always an error - it's not a numpy array, and doesn't assume anything about missing slices. You must always provide 5 slices/indices when using `data_xyztc`. This is both because of how it's implemented, and because it's safer that way - at least how we use things. It forces you to think about how you want to handle, e.g. colour when you write the code and avoids ambiguous promotion behaviour which is probably going to be wrong.
- Will have to look into why the lables `_xyztc` access with 4 slices does not fail.

David-Baddeley, 2021-07-09:
OK, I'm getting a better understanding of why you are getting error messages where you are.

- demotion to xyztc always results in a 4D demoted data source, even if the result would have previously been 3D. As the demoted data should be backwards compatible, the solution is probably to make the demoted datasource ignore missed dimensions (like the numpy behaviour). Anywhere you get this error, however,  would have been likely to break on different dimensioned data, even in the old code. The medium term solution is to replace everything with `data_xyztc` based access so you get consistent results. 
- accessing the labeled `_xyztc` data with only 4 indices doesn't enforce 5D indexing as, although it's 5D, it's a `ListWrap` rather than an `XYZTCDataSource` instance. This is down to lines 76-79 in dataWrap.py which defaults to taking the 0th item if you miss the index.

David-Baddeley, 2021-07-09:
Proposed solution:

- make the backwards compatible `_xytc` datasource ignore missed dimensions, but make it warn when doing so.
- change `ListWrapper` to add a `strict_dims` option which disables the missing dimension logic, and instead causes it to generate an error when not indexing with all dimensions. Use this when using the list wrapper in the 5D model

**NOTE:** the previous missing dimension behaviour was different to numpy's we took the 0th item along that dimension rather than a : slice.

csoeller, 2021-07-09:
The proposed solution sounds good and will make the behaviour of the datasources consistent.

WRT to the `Measure2D` implementation, I guess switching over to using these datasources with the suggested indexing is the way to go? There is at least one other place where a change needs to be made in `measure.py` which implements the loop over frames and currently merely generates a warning:

```
for i in xrange(labels.data.shape[2]):
            m.addFrameMeasures(i, *self._measureFrame(i, labels, intensity))
```

Possibly the whole concept of the loop over frame numbers in `Measure2D` needs an overhaul with the newer data model?

David-Baddeley, 2021-07-09:
`Measure2D` needs lots of attention ... see also #524. Should also be made voxel-size aware so measurements are in sensible (physical) units. Needs a little thought as to how to do this whilst maintaining backwards compatibility - might end up being better writing a replacement module and leaving the current one pretty much as is for backwards compatibility.

In the short-term, changing it to loop over z and t instead of frame numbers and adding `z` and `t` columns to the output might be the right approach (we probably still want to provide frame numbers as, e.g., `t*size_z + z` for backwards compatibility).

csoeller, 2021-07-09:
Cool, we are trying to get some analysis done for a paper/presentation and it would be good to get a workaround in place that allows using the current `Measure2D` relatively swiftly. However, we could always go with a local custom patch for a little while if that is too tricky.

A replacement module as the proper medium term fix seems the right way to go.

David-Baddeley, 2021-07-09:
Patch in #1082 should fix things, along with a bit more documentation. `Measure2D` still projects both z and t to a frames dimension for now (i.e. uses the backwards compatible `data_xytc`). I've added a single-colour assertion to `Measure2D` which could potentially break some old recipes, but I'd argue these were broken anyway / relying on glitches to operate.

PR #1084 (2021-07-09): Fix for #1082
Fixes #1082, also adds a couple of sanity checks and doc fixes to Measure2D
closed 2021-07-09 by David-Baddeley