python-microscopy/python-microscopy #1094: [BUG] No module named PYMEnf

zacsimile, 2021-07-29:
```
Traceback (most recent call last):
  File "/Users/zachcm/Code/python-microscopy/PYME/Acquire/acquiremainframe.py", line 238, in doPostInit
    cm.run(self, self.scope)
  File "/Users/zachcm/Code/python-microscopy/PYME/Acquire/ExecTools.py", line 205, in run
    self.codeObj(parent, scope)
  File "/Users/zachcm/Code/python-microscopy/PYME/Acquire/Scripts/init.py", line 120, in fake_dmd
    from PYMEnf.Hardware import FakeDMD, DMDGui
ModuleNotFoundError: No module named 'PYMEnf'
```

This seems to result in an inability to display/use Simulation Settings on both Mac and Windows.

[image]

zacsimile, 2021-07-29:
Ugh, just kidding. These are two separate errors. The `PYMEnf` error isn't actually a problem. We could consider commenting out `PYMEnf` calls by default in the main repo so as to not have to consider irrelevant errors.

The display doesn't work until you resize the window, then it works fine.

[image]

This is a problem on `wx 4.0.4` and `wx 4.1.1`.

zacsimile, 2021-07-29:
It seems like the easiest thing to do is bind `PYME.ui.manualFoldingPanel.foldPanel.fold1` to `SetSelection()`. Not sure why this isn't bound/exactly how to bind it.

David-Baddeley, 2021-07-29:
Or maybe bind to a resize, rather than `SetSelection()` specifically. 

I swear this is new - maybe a regression from the work on the dh5view colour channels which involved subtle tweaks to the folding panes etc ... to make them do sensible things when resized.

David-Baddeley, 2021-07-29:
I'm guessing the underlying problem is something along the following lines:

- window gets created hidden with 0 size
- because tab is inactive, all layout is done with 0 size
- this doesn't refresh when changing to that tab (until the whole window is resized)

PR #1095 (2021-08-03): issue1094
fixes #1094 by binding a resize/layout to EVT_SHOW
closed 2021-08-03 by David-Baddeley