python-microscopy/python-microscopy #1096: [SUPPORT] OpenGL / MSVC issue(?) on new dev install

barentine, 2021-08-09:
**What are you trying to do?**
Trying to install from source on two Win 10 computers. Installation went smoothly on one (call this machine A), where I had started by running `conda config --append channels anaconda` as is in the installation from conda instructions. Everything seems to work great. Installation on the other (call this B), I did not do that step (it is not listed under dev install instructions) and am running into OpenGL errors in PYMEAcquire and PYMEVis both (which print continuously when you drag the window). 

Both were installed today, from main. When we realized we didn't have MSVC 14.0+ installed on machine B (which we found out during python setup.py develop), we realized we didn't need the full dev install and did a `conda install python-microscopy`. Trying to run, we ran into these OpenGL errors, did a `conda uninstall python-microscopy` followed by installing MSVC 14.0+ and running a successful `setup.py develop`. Still have the opengl errors. Haven't really looked into it yet, but wanted to post and see if anyone else had bumped into something similar. The difference I note of course is the `[vc14]  anaconda` in the conda list

<details> <summary> Traceback </summary>

```
OpenGL - Version: b'1.1.0'
Traceback (most recent call last):
  File "c:\userfiles\ashwin\code\python-microscopy\PYME\LMVis\gl_render3D_shaders.py", line 211, in OnPaint
    self.initialize()
  File "c:\userfiles\ashwin\code\python-microscopy\PYME\LMVis\gl_render3D_shaders.py", line 254, in initialize
    self.InitGL()
  File "c:\userfiles\ashwin\code\python-microscopy\PYME\LMVis\gl_render3D_shaders.py", line 610, in InitGL
    print('Shader - Version: {}'.format(glGetString(GL_SHADING_LANGUAGE_VERSION)))
  File "C:\Users\abalaji8\Miniconda3\envs\pyme\lib\site-packages\OpenGL\error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
        err = 1280,
        description = b'invalid enumerant',
        baseOperation = glGetString,
        cArguments = (GL_SHADING_LANGUAGE_VERSION,)
)
```

</details>

<details> <summary> not-functional environment </summary>

```
# Name                    Version                   Build  Channel
alabaster                 0.7.12                   py37_0
babel                     2.9.1              pyhd3eb1b0_0
blas                      1.0                         mkl
blosc                     1.19.0               h7bd577a_0
bottleneck                1.3.2            py37h2a96729_1
brotlipy                  0.7.0           py37h2bbff1b_1003
bzip2                     1.0.8                he774522_0
ca-certificates           2021.7.5             haa95532_1
cairo                     1.14.12              hf171d8a_3
certifi                   2021.5.30        py37haa95532_0
cffi                      1.14.6           py37h2bbff1b_0
chardet                   4.0.0           py37haa95532_1003
cheroot                   8.5.2            py37haa95532_1
cherrypy                  18.6.0           py37haa95532_0
cloudpickle               1.6.0                      py_0
cryptography              3.4.7            py37h71e12ea_0
cycler                    0.10.0                   py37_0
cython                    0.29.24          py37hd77b12b_0
cytoolz                   0.11.0           py37he774522_0
dask-core                 2021.7.2           pyhd3eb1b0_0
docutils                  0.17.1           py37haa95532_1
fftw                      3.3.4                         0    david_baddeley
freetype                  2.10.4               hd328e21_0
fsspec                    2021.7.0           pyhd3eb1b0_0
future                    0.18.2                   py37_1
hdf5                      1.10.4               h7ebc959_0
icc_rt                    2019.0.0             h0cc432a_1
icu                       58.2                 ha925a31_3
idna                      2.10               pyhd3eb1b0_0
ifaddr                    0.1.6                      py_0    david_baddeley
imageio                   2.9.0              pyhd3eb1b0_0
imagesize                 1.2.0              pyhd3eb1b0_0
intel-openmp              2021.3.0          haa95532_3372
jaraco.classes            3.2.1              pyhd3eb1b0_0
jaraco.collections        3.3.0              pyhd3eb1b0_0
jaraco.functools          3.3.0              pyhd3eb1b0_0
jaraco.text               3.5.1            py37haa95532_2
jinja2                    3.0.1              pyhd3eb1b0_0
[... 216 lines cut ...]
toolz                     0.11.1             pyhd3eb1b0_0
toposort                  1.5                      py37_3    david_baddeley
tornado                   6.1              py37h2bbff1b_0
traits                    6.2.0            py37h2bbff1b_0
traitsui                  7.1.0                      py_0
ujson                     4.0.2            py37hd77b12b_0
urllib3                   1.26.6             pyhd3eb1b0_1
vc                        14.2                 h21ff451_1
vs2015_runtime            14.27.29016          h5e58377_2
wheel                     0.36.2             pyhd3eb1b0_0
win_inet_pton             1.1.0            py37haa95532_0
wincertstore              0.2                      py37_0
wxpython                  4.0.4            py37ha925a31_0
xz                        5.2.5                h62dcd97_0
yaml                      0.1.7            vc14h4cb57cf_1  [vc14]  anaconda
zc.lockfile               2.0                        py_0
zeroconf                  0.24.0                   py37_3    david_baddeley
zlib                      1.2.11           vc14h1cdd9ab_1  [vc14]  anaconda
zstd                      1.4.9                h19a0ad4_0
```

</details>

**Where are you stuck?**
Haven't tried much yet - had to stop for the day and wanted to pop in quick to see if we'd seen this before

David-Baddeley, 2021-08-09:
I wonder if this has something to do with the graphics card and/or graphics card driver, rather than anything to do with the PYME installation process? The OpenGL version reported in the traceback (or strictly speaking, directly before the traceback) is very old, and the traceback would seem to suggest that shaders are not supported by that version of OpenGL (Not surprising as shaders were introduced in OpenGL version 2.0 - i.e. 2004). 

For context, the OpenGL version on my ( very old) mac with intel (onboard graphics) is `'2.1 INTEL-16.4.5'`, with GLSL version 1.2. The version on my Win10 desktop with an RTX2070 is `4.6.0 NVIDIA 460.89` with GLSL `4.60 NVIDIA`. Modern graphics cards all tend to be at version 4 ish or above. The fact that you have 1.1.0 to me suggests that you are using software OpenGL emulation - i.e. Mesa  or similar, and that the driver for your graphics card hasn't been installed properly.

David-Baddeley, 2021-08-09:
A video card with hardware OpenGL is one of our very few hardware requirements ...

David-Baddeley, 2021-08-09:
Maybe we should do an explicit check here and give an informative error message. This does feel rather like an edge case though - the chances of not having hardware OpenGL these days is vanishingly slim

David-Baddeley, 2021-08-09:
I've just had another idea - is the windows installation in question running in a VM by any chance? That's the only scenario these days where I'd expect to see software OpenGL.

barentine, 2021-08-10:
Thanks @David-Baddeley ! You're right that the opengl version is exceptionally old - I'd only registered the pyopengl version. Will keep you posted - hopefully just a driver update away.

barentine, 2021-08-11:
GPU in question appears to have opengl 1.3 support (but no windows 10 drivers). Going to close the issue. Thank you, @David-Baddeley !

closed 2021-08-11 by barentine

David-Baddeley, 2021-08-11:
Would be interesting to know what type of machine that was ... Might still be a reason to make the error message informative - i.e. check version and if <2 report that we need OpenGL > 2.