python-microscopy/python-microscopy #1098: [BUG] Localization analysis doesn't run at single-frame image

Yujin-Bao, 2021-08-10:
**Describe the bug**
I use `LatGaussFitFR` to analyse localization data from a one-frame image of 1um beads. An error shows in `pymeclusterofone` terminal.

**To Reproduce**
1. For `point finding`, I set the threshold to 5 and run `test this frame` to make sure for each bead, there is only one candidate region to be fitted, labeled by the small green square.
2. This is only a one-frame image, so set `start at` as 0, and cancel `subtract background in fit`, then click `go`
[image]
3. See error in `pymeclusterofone` terminal: `tables.exceptions.NoSuchNodeError: group / does not have a child named Events`.
[image]

**Expected behavior**
I expect to see 2D Gaussian fittings in each small green square as well as their acceptance and rejection.

David-Baddeley, 2021-08-10:
Hmm, there looks to be multiple things going on here, and the tracebacks are not particularly useful ...

- What format is the input data? The tracebacks would suggest that it's .h5 formatted, but lacks an events table. (This might not actually be a critical error - it might get handled fine in normal operation, but we might be seeing the error because something else fails when we try and handle it)
- The second error (NoneType is not iterable) is probably more important. The only thing that I can think of is that this might be caused by an empty array (i.e. no molecules detected and fit).

In line with this error, it's worth pointing out that to disable background subtraction, you need to both uncheck the `subtract background in fit` box **and** set the `Background:` range to `0:0`. With the settings as shown, you will be attempting to use background subtraction for object detection, but not for fitting. Revamping the GUI here to show separate "subtract background for detection" and "subtract background in fit" options (which would be much more understandable) has been on the  TODO list for a while now, but as it stands, having anything other than `0:0` in the background range implies subtract background for detection. There might also be subtle differences in how this works when testing and when analyising.

David-Baddeley, 2021-08-10:
PS - if you could share the data you're trying to fit that might help further debugging.

Yujin-Bao, 2021-08-11:
Thank you for the reply, David! 
- The format of the input data is `.tif` file, but I did create a `.md` metadata file as it suggested in the documetation. However, I only set the minimum set of metadata parameters.
- I set the `Background` range to `0:0` and ran it again, but still got the same problem. 
-  I'll send you the original data through email for further reference.

David-Baddeley, 2021-08-13:
Thanks for the data! Have got it ~90% fixed (at least enough that the analysis will run and generate correct output data).

closed 2021-08-16 by David-Baddeley

Yujin-Bao, 2021-08-19:
Hi David! Thanks for solving the problem! However, when I tried it again, the traceback in `PymeImage` reported another error, saying `Names must only include alphanumeric characters and underscore`. But I think I have the proper name for this. Could you tell me how I could solve this?
[image]

zacsimile, 2021-08-20:
@lukasfue has the same problem. I tried this on my mac and it worked fine, so I suspect this is windows-specific.

zacsimile, 2021-08-20:
Backslashes again, as well as Windows's file-naming scheme including a `:` in every path. Fixed in #1104.

PR #1102 (2021-08-16): Cluster localization fixes (issue 1098)
Fixes #1098, fixes #990, closes #786 

3 for the price of one!

We were silently ignoring `startAt` when doing cluster (and cluster of one) based analyses. This fixes that, and at the same time  shifts everything to the rule classes in `cluster.rules` so we don't need to maintain two lots of code for pushing (meaning that we can finally close #786).

Whilst testing (newer OSX version with slower network IO :( ) , I also ran into #990, so this includes a fix for that as well (which turned out to be surprisingly simple - just moving the socket opening into [... cut]