python-microscopy/python-microscopy #1099: [BUG] pyfftw compat / dask /numpy version issue in 21.03.03 installer

barentine, 2021-08-11:
**What are you trying to do?**
Trying to set some folks up on PYME using the installers. Generally great, very helpful route. With the 'current' version, 21.03.03, there is a bug: PYMEAcquire (simulator) can't run. Not sure if that's a known issue or an unknown issue but will file here. Dask tries to import a numpy function that was introduced in numpy 1.15.0, but pyme-depends-strict is built with numpy 1.14

<details> <summary> Traceback </summary>

```
ERROR:PYME.Acquire.ExecTools:Error running background init task Fake Camera

Traceback (most recent call last):

  File "C:\01Y-Ddisk\python-microscopy\lib\site-packages\PYME\misc\fftw_compat.py", line 6, in <module>

    from fftw3f import Plan, create_aligned_array

ModuleNotFoundError: No module named 'fftw3f'

 

During handling of the above exception, another exception occurred:

 

Traceback (most recent call last):

  File "C:\01Y-Ddisk\python-microscopy\lib\site-packages\dask\array\chunk.py", line 15, in <module>

    from numpy import take_along_axis

ImportError: cannot import name 'take_along_axis'

 

During handling of the above exception, another exception occurred:

 

Traceback (most recent call last):

  File "C:\01Y-Ddisk\python-microscopy\lib\site-packages\PYME\Acquire\ExecTools.py", line 140, in _bginit

    self.codeObj(defLocals['scope'])

  File "C:\01Y-Ddisk\python-microscopy\lib\site-packages\PYME\Acquire\Scripts\init.py", line 55, in cm

    from PYME.Acquire.Hardware.Simulator import fakeCam

  File "C:\01Y-Ddisk\python-microscopy\lib\site-packages\PYME\Acquire\Hardware\Simulator\fakeCam.py", line 24, in <module>

    from . import rend_im

  File "C:\01Y-Ddisk\python-microscopy\lib\site-packages\PYME\Acquire\Hardware\Simulator\rend_im.py", line 42, in <module>

    from PYME.Deconv.wiener import resizePSF

  File "C:\01Y-Ddisk\python-microscopy\lib\site-packages\PYME\Deconv\wiener.py", line 29, in <module>

    import PYME.misc.fftw_compat as fftw3f

  File "C:\01Y-Ddisk\python-microscopy\lib\site-packages\PYME\misc\fftw_compat.py", line 8, in <module>

    import pyfftw

  File "C:\01Y-Ddisk\python-microscopy\lib\site-packages\pyFFTW-0.11.1+26.g101c792-py3.6-win-amd64.egg\pyfftw\__init__.py", line 43, in <module>

    from . import interfaces

  File "C:\01Y-Ddisk\python-microscopy\lib\site-packages\pyFFTW-0.11.1+26.g101c792-py3.6-win-amd64.egg\pyfftw\interfaces\__init__.py", line 279, in <module>

    from dask.array.fft import fft_wrap

[... 182 lines cut ...]
 

DEBUG:PYME.Acquire.acquiremainframe:Backround initialization done

DEBUG:PYME.Acquire.acquiremainframe:Starting post-init

Traceback (most recent call last):

  File "C:\01Y-Ddisk\python-microscopy\lib\site-packages\PYME\ui\mytimer.py", line 42, in Notify

    a()

  File "C:\01Y-Ddisk\python-microscopy\lib\site-packages\PYME\Acquire\acquiremainframe.py", line 170, in _check_init_done

    self.doPostInit()

  File "C:\01Y-Ddisk\python-microscopy\lib\site-packages\PYME\Acquire\acquiremainframe.py", line 232, in doPostInit

    if self.scope.cam.CamReady():# and ('chaninfo' in self.scope.__dict__)):

AttributeError: 'NoneType' object has no attribute 'CamReady'
```
</details>

Tested on multiple Windows 10 systems.

David-Baddeley, 2021-08-11:
Is that still the case for the latest automated build (21.07.07) (https://python-microscopy.org/downloads/automated_builds/)? That's dropped the numpy pin. I should probably just make that the new official build, it just hasn't had the full degree of manual testing.

barentine, 2021-09-15:
This issue could be closed, but would be nice to update default download first. More recent builds do not have this problem.

closed 2021-09-26 by barentine