python-microscopy/python-microscopy #1100: [SUPPORT]

zhaohh567, 2021-08-12:
**What are you trying to do?**
I’m trying to use the focus lock function of PYME Acquire. I ran the code in init_htsms.py and init_htsms_focus_lock.py, created the client and server with port numbers 9797 and 9798 respectively, and successfully opened two PYME Acquire interfaces. 
But when I click the lock button or Set Focus button in the FocusLockPanel of the main program, it‘s giving me a connection error. No connection and communication between the client and the server
[image]

[image]

**Where are you stuck?**
I confirm that I have opened these two ports 
At the same time, through the created OffsetPiezoServer, I can manually control piezo_e709 to realize the z-axis movement of the stage. 
The problem may be caused by class RLPIDFocusLockClient and class RLPIDFocusLockClient. I'm not sure.
Os： The type of focus lock I use is based on laser reflection. Tested on multiple Windows 10 systems

David-Baddeley, 2021-08-12:
My first guess would be that the windows firewall (or a 3rd party firewall) might be to blame. The first time you run a PYME program that uses networking functionality (as, e.g. the focus lock does) it'll prompt you as to whether or not you want to allow it. Subsequently you need to go into the firewall settings in the control panel and delete / modify the "python" rule to change what you set the first time you were prompted.

Another possibility is that there is a race condition in getting the server side on the focus lock set up. The whole reflection lock stuff is pretty alpha-quality and not particularly portable to systems other than the HTSMS system. It's structure with 2 separate servers which talk to each other makes it particularly fragile and prone to things like race conditions on initialisation. Ultimately we should probably refactor both focus locks to bring the "lock" logic back into the main program, just leaving camera control and e.g. offset calculation in the child program.

barentine, 2021-08-23:
@zhaohh567 Is there actually no communication between client/server? Is there any traceback coming from the focus lock instance of PYMEAcquire? As @David-Baddeley mentioned that code is fairly special-cased to the system I was working on, and if you try and use it as-is out of the box I would not be surprised if e.g. your camera is rotated different than mine relative to the reflection strip (I used a cylindrical lens) and the fit fails to converge. If you don't see a traceback, have a try at manually walking through the fitting code on a single frame through the shell of the focus lock instance. David's point about windows firewall is probably the place to start though - did that help?