python-microscopy/python-microscopy #1111: [BUG] ValueError When Opening PYMEImage in the Command Line on Win10

Yujin-Bao, 2021-09-06:
**Describe the bug**
I use python 3.6.13 in my `PYME` environment both on my lab computer and laptop. Both of them are on Window systems. The one on my lab computer works well, but these days when I activate `PYME` environment and type `pymeimage` in the command line (Anaconda Prompt), it reports `ValueError: source code string cannot contain null bytes`. I have also tried `pymeacquire` and `pymevis` on my laptop, but errors were there, too.

**To Reproduce**
Steps to reproduce the behavior:
1. Use development installation following the documentations
2. I use the python version 3.6.13, Win10 system (Win10 Pro on the lab computer, and Win10 Family Chinese Version on my laptop)
3. Activate `PYME` environment, and type `pymeimage`, see error
4. At first, both the lab computer and my laptop work well. But these days it suddenly doesn't work on my labtop. I have tried` git pull` and remove `PYME` environment then reinstall and `git pull` again on my laptop, but it still reports the same problem. 
5. It also reports error when `pymeacquire` and `pymevis`

**Screenshots**

```
(pyme) C:\Users\MI>pymeimage
INFO:numexpr.utils:NumExpr defaulting to 8 threads.
D:\Installation\Miniconda\envs\pyme\lib\_collections_abc.py:702: MatplotlibDeprecationWarning: The global colormaps dictionary is no longer considered public API.
  return len(self._mapping)
D:\Installation\Miniconda\envs\pyme\lib\_collections_abc.py:720: MatplotlibDeprecationWarning: The global colormaps dictionary is no longer considered public API.
  yield from self._mapping
Traceback (most recent call last):
  File "D:\Installation\Miniconda\envs\pyme\Scripts\PYMEImage-script.py", line 33, in <module>
    sys.exit(load_entry_point('python-microscopy', 'console_scripts', 'PYMEImage')())
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\pkg_resources\__init__.py", line 474, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\pkg_resources\__init__.py", line 2846, in load_entry_point
    return ep.load()
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\pkg_resources\__init__.py", line 2450, in load
    return self.resolve()
  File "D:\Installation\Miniconda\envs\pyme\lib\site-packages\pkg_resources\__init__.py", line 2456, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "c:\users\mi\python-microscopy\PYME\DSView\__init__.py", line 31, in <module>
    from .dsviewer import View3D, ViewIm3D
  File "c:\users\mi\python-microscopy\PYME\DSView\dsviewer.py", line 64, in <module>
    from PYME.ui.AUIFrame import AUIFrame
  File "c:\users\mi\python-microscopy\PYME\ui\AUIFrame.py", line 6, in <module>
    from PYME.misc import check_for_updates
  File "c:\users\mi\python-microscopy\PYME\misc\check_for_updates.py", line 136, in <module>
    check_for_updates(False)
  File "c:\users\mi\python-microscopy\PYME\misc\check_for_updates.py", line 71, in check_for_updates
    with shelve.open(update_info_fn) as s:
  File "D:\Installation\Miniconda\envs\pyme\lib\shelve.py", line 243, in open
    return DbfilenameShelf(filename, flag, protocol, writeback)
  File "D:\Installation\Miniconda\envs\pyme\lib\shelve.py", line 227, in __init__
    Shelf.__init__(self, dbm.open(filename, flag), protocol, writeback)
  File "D:\Installation\Miniconda\envs\pyme\lib\dbm\__init__.py", line 94, in open
    return mod.open(file, flag, mode)
  File "D:\Installation\Miniconda\envs\pyme\lib\dbm\dumb.py", line 324, in open
    return _Database(file, mode, flag=flag)
  File "D:\Installation\Miniconda\envs\pyme\lib\dbm\dumb.py", line 71, in __init__
    self._update()
  File "D:\Installation\Miniconda\envs\pyme\lib\dbm\dumb.py", line 106, in _update
    key, pos_and_siz_pair = _ast.literal_eval(line)
  File "D:\Installation\Miniconda\envs\pyme\lib\ast.py", line 48, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')
  File "D:\Installation\Miniconda\envs\pyme\lib\ast.py", line 35, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
ValueError: source code string cannot contain null bytes
```

David-Baddeley, 2021-09-06:
Hi Yujin,

this looks odd - my best guess at the moment is that the database we use to check for updates somehow got corrupted. First attempt at resolving the issue would be to try and delete the database file. You can find the database file at:

`<your home directory>\.PYME\update.shv.db`

Let me know if this works.

If we see this problem again (i.e. if it wasn't a fluke), we can look at why the database is getting corrupted.

David

Yujin-Bao, 2021-09-06:
Hi David,

Thanks for the reply! However I don't have that `.db` file, instead I have `update.shv.bak`, `update.shv.dat`, and `update.shv.dir` files. Do I need to delet them all to try?

Best,
Yujin

David-Baddeley, 2021-09-06:
You should be safe to delete them all.

Cheers, David

Yujin-Bao, 2021-09-06:
Cool! It worked! Thank you so much!

Cheers,
Yujin

David-Baddeley, 2021-11-11:
Think we can close this now

closed 2021-11-11 by David-Baddeley