python-microscopy/python-microscopy #1112: [BUG] Visualization only displaying in 1/4 of window size on macOS, python 3.8

zacsimile, 2021-09-10:
[image]

Problem persists on restart.

<details>
<summary>Terminal</summary>
<pre>
(pyme) MWC02KC2D4FFRR:~ zrc4$ pymevis
/opt/miniconda3/envs/pyme/lib/python3.8/_collections_abc.py:702: MatplotlibDeprecationWarning: The global colormaps dictionary is no longer considered public API.
  return len(self._mapping)
/opt/miniconda3/envs/pyme/lib/python3.8/_collections_abc.py:720: MatplotlibDeprecationWarning: The global colormaps dictionary is no longer considered public API.
  yield from self._mapping
ERROR:PYME.recipes.modules:Could not import skimage
Traceback (most recent call last):
  File "/Users/zrc4/Code/python-microscopy/PYME/recipes/modules.py", line 26, in <module>
    from . import skfilters
  File "/Users/zrc4/Code/python-microscopy/PYME/recipes/skfilters.py", line 90, in <module>
    a = args[-j]
IndexError: list index out of range
Trying to load 3rd party recipe module ch_shrinkwrap.recipe_modules.surface_fitting
Loaded 3rd party recipe module ch_shrinkwrap.recipe_modules.surface_fitting
quaternion module not found, disabling custom clip plane orientations
quaternion module not found, disabling custom clip plane orientations
New Canvas
DEBUG:PYME.LMVis.visCore:Creating VisGUI menu bar
DEBUG:PYME.LMVis.Extras:Initializing QPobjectSegment plugin
DEBUG:PYME.LMVis.Extras:Initializing animation plugin
DEBUG:PYME.LMVis.Extras:Initializing clusterAnalysis plugin
DEBUG:PYME.LMVis.Extras:Initializing dockedPanel plugin
DEBUG:PYME.LMVis.Extras:Initializing extra_layers plugin
DEBUG:PYME.LMVis.Extras:Initializing fiducial plugin
DEBUG:PYME.LMVis.Extras:Initializing flagZMotionArtifacts plugin
DEBUG:root:Adding menu items for event filters
DEBUG:PYME.LMVis.Extras:Initializing fusionUtils plugin
DEBUG:PYME.LMVis.Extras:Initializing multiColourAnalysis plugin
DEBUG:PYME.LMVis.Extras:Initializing multiviewMapping plugin
DEBUG:root:Adding menu items for multi-view manipulation
DEBUG:PYME.LMVis.Extras:Initializing objectMeasurements plugin
DEBUG:PYME.LMVis.Extras:Initializing particleTracking plugin
DEBUG:PYME.LMVis.Extras:Initializing perFrameVariable plugin
DEBUG:PYME.LMVis.Extras:Initializing photophysics plugin
DEBUG:PYME.LMVis.Extras:Initializing pointSetGeneration plugin
DEBUG:PYME.LMVis.Extras:Initializing pointwiseColoc plugin
DEBUG:PYME.LMVis.Extras:Initializing saved_views plugin
Saved Views are deprecated -  use the ones with animation instead and remove me!
DEBUG:PYME.LMVis.Extras:Initializing scaleBox plugin
DEBUG:PYME.LMVis.Extras:Initializing sequentialSpecies plugin
DEBUG:PYME.LMVis.Extras:Initializing setBGImage plugin
DEBUG:PYME.LMVis.Extras:Initializing shiftmapGenerator plugin
DEBUG:PYME.LMVis.Extras:Initializing showShortcuts plugin
DEBUG:PYME.LMVis.Extras:Initializing snapshot plugin
DEBUG:PYME.LMVis.Extras:Initializing spherical_harmonics plugin
DEBUG:root:Adding menu items for spherical harmonic (shell) fitting
DEBUG:PYME.LMVis.Extras:Initializing subsets plugin
DEBUG:PYME.LMVis.Extras:Initializing surface_fitting plugin
DEBUG:PYME.LMVis.Extras:Initializing temperaturePlot plugin
DEBUG:PYME.LMVis.Extras:Initializing vibration plugin
Creating fold panel
GenPanels
Creating datasource panel
RecipeView._layout
Input datasources:  []
Image list size: 1
Creating layers GUI
height:  0
NB best size: wx.Size(200, 0)
RecipeView._layout
Input datasources:  []
recipe.execute()
RecipeView._layout
[... 174 lines cut ...]
sphinxcontrib-jsmath      1.0.1                    pypi_0    pypi
sphinxcontrib-qthelp      1.0.3                    pypi_0    pypi
sphinxcontrib-serializinghtml 1.1.5                    pypi_0    pypi
sqlalchemy                1.4.15                   pypi_0    pypi
sqlite                    3.36.0               hce871da_0  
tables                    3.6.1                    pypi_0    pypi
tempora                   4.1.1                    pypi_0    pypi
threadpoolctl             2.2.0                    pypi_0    pypi
tifffile                  2021.8.8                 pypi_0    pypi
tk                        8.6.10               hb0a8c7a_0  
toposort                  1.6                      pypi_0    pypi
traits                    6.1.0                    pypi_0    pypi
traitsui                  7.1.0                    pypi_0    pypi
ujson                     4.1.0                    pypi_0    pypi
urllib3                   1.26.6                   pypi_0    pypi
werkzeug                  2.0.1                    pypi_0    pypi
wheel                     0.37.0             pyhd3eb1b0_0  
wxpython                  4.1.1                    pypi_0    pypi
xz                        5.2.5                h1de35cc_0  
zc-lockfile               2.0                      pypi_0    pypi
zeroconf                  0.17.7                   pypi_0    pypi
zipp                      3.5.0                    pypi_0    pypi
zlib                      1.2.11               h1de35cc_3 
</pre>
</details>

David-Baddeley, 2021-09-10:
Have you tried `View->Fit` from the menu and/or resizing the window.

py3.8 forces you onto a largely untested wx version, so that might be part of the issue. A few more questions:

- what version of macOS?
- what version of openGL?
- what type of screen, and what scaling compared with native? (one possibility is that wx's handling of high dpi screens such as retina displays has changed - by default mac reports downscaled coordinates to apps, and if we e.g. got the window size in OS coordinates, but scaled our viewport in device coordinates something like this could happen).

zacsimile, 2021-09-11:
View -> Fit resizes the object into the 1/4 size window perfectly.

* macOS Catalina (10.15.7)
* openGL 4.1
* MacBook Pro, Early 2013 retina display 1.3 inch 2560x1600. Resolution is set as default for display, no scaling.

I am running wxpython 4.1.1.

David-Baddeley, 2021-12-07:
We're also getting some other rendering glitches - e.g. icons for minimised overlays, display settings in PYMEImage

David-Baddeley, 2021-12-08:
- Still an issue on a fresh install 
-  "fixable" by manually setting `glCanvas.view_port_size` to twice the window size (rather than the window size)

David-Baddeley, 2021-12-08:
Above was with isosurfaces in dh5view. There are a bunch of other bugs related to layer creation and contexts that prevent PYMEVis working at all, fixes hopefully inbound ....

PR #1196 (2022-02-09): Fix display scaling on osx, py3.8
Fixes #1112
PR #1197 (2022-02-09): Fix display scaling, take 2
Fixes #1112 (properly this time)
closed 2022-02-09 by David-Baddeley