python-microscopy/python-microscopy #1114: [BUG] [21.07.07] stackSettings

barentine, 2021-09-14:
**Describe the bug**
If you install 21.07.07 on win10 from https://python-microscopy.org/downloads/automated_builds/, run PYMEAcquire (from anaconda prompt, because for some reason we dropped the PYMEAcquire (simulator) shortcut)

```
Traceback (most recent call last):
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\acquiremainframe.py", line 238, in doPostInit
    cm.run(self, self.scope)
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\ExecTools.py", line 205, in run
    self.codeObj(parent, scope)
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\Scripts\init.py", line 120, in fake_dmd
    from PYMEnf.Hardware import FakeDMD, DMDGui
ModuleNotFoundError: No module named 'PYMEnf'
DEBUG:PYME.Acquire.acquiremainframe:Loading GUI component for Laser controls
DEBUG:PYME.Acquire.acquiremainframe:Loading GUI component for Focus Keys
DEBUG:PYME.Acquire.acquiremainframe:Loading GUI component for Action manager
0
DEBUG:PYME.Acquire.acquiremainframe:Loading GUI component for Tiling
DEBUG:PYME.Acquire.acquiremainframe:Run all custom GUI init tasks
0
Traceback (most recent call last):
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\ui\mytimer.py", line 42, in Notify
    a()
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\acquiremainframe.py", line 170, in _check_init_done
    self.doPostInit()
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\acquiremainframe.py", line 279, in doPostInit
    self.seq_d = seqdialog.seqPanel(self, self.scope)
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\ui\seqdialog.py", line 214, in __init__
    self.stackSettings = self.scope.stackSettings
AttributeError: 'microscope' object has no attribute 'stackSettings'
```

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: Win10

<details> <summary> conda list </summary>

```
# Name                    Version                   Build  Channel
alabaster                 0.7.12                   py37_0
babel                     2.9.1              pyhd3eb1b0_0
blas                      1.0                         mkl
blosc                     1.19.0               h7bd577a_0
brotlipy                  0.7.0           py37h2bbff1b_1003
bzip2                     1.0.8                he774522_0
ca-certificates           2021.7.5             haa95532_1
cairo                     1.14.12              hf171d8a_3
certifi                   2021.5.30        py37haa95532_0
cffi                      1.14.5           py37hcd4344a_0
chardet                   4.0.0           py37haa95532_1003
cheroot                   8.5.2            py37haa95532_1
cherrypy                  18.6.0           py37haa95532_0
cloudpickle               1.6.0                      py_0
conda                     4.10.3           py37haa95532_0
conda-package-handling    1.7.3            py37h8cc25b3_1
cryptography              3.4.7            py37h71e12ea_0
cycler                    0.10.0                   py37_0
cython                    0.29.23          py37hd77b12b_0
cytoolz                   0.11.0           py37he774522_0
dask-core                 2021.6.2           pyhd3eb1b0_0
decorator                 5.0.9              pyhd3eb1b0_0
docutils                  0.17.1           py37haa95532_1
fftw                      3.3.9                h2bbff1b_1
freetype                  2.10.4               hd328e21_0
fsspec                    2021.6.0           pyhd3eb1b0_0
future                    0.18.2                   py37_1
hdf5                      1.10.4               h7ebc959_0
icc_rt                    2019.0.0             h0cc432a_1
icu                       58.2                 ha925a31_3
idna                      2.10               pyhd3eb1b0_0
ifaddr                    0.1.6                      py_0    david_baddeley
[... 88 lines cut ...]
toolz                     0.11.1             pyhd3eb1b0_0
toposort                  1.5                      py37_3    david_baddeley
tornado                   6.1              py37h2bbff1b_0
tqdm                      4.61.2             pyhd3eb1b0_1
traits                    6.2.0            py37h2bbff1b_0
traitsui                  7.1.0                      py_0
ujson                     4.0.2            py37hd77b12b_0
urllib3                   1.26.6             pyhd3eb1b0_1
vc                        14.2                 h21ff451_1
vs2015_runtime            14.27.29016          h5e58377_2
wheel                     0.36.2             pyhd3eb1b0_0
win_inet_pton             1.1.0            py37haa95532_0
wincertstore              0.2                      py37_0
wxpython                  4.0.4            py37ha925a31_0
xz                        5.2.5                h62dcd97_0
yaml                      0.2.5                he774522_0
zc.lockfile               2.0                        py_0
zeroconf                  0.24.0                   py37_3    david_baddeley
zlib                      1.2.11               h62dcd97_4
zstd                      1.4.9                h19a0ad4_0
```

</details>
**Additional context**
Note that 21.03.03 in https://python-microscopy.org/downloads/ is also broken for PYMEAcquire, see #1099

barentine, 2021-09-14:
Works fine on dev install which is more recent (though shares the same tag of course since we haven't marked anything a release since 07.07

barentine, 2021-09-14:
@David-Baddeley if you can mark a release asap I'm happy to test the executable install on windows today (PYMEAcquire -> localize -> visualize -> render -> map back to localizations).

barentine, 2021-09-14:
Some bug present in 06.23 build, by the way

David-Baddeley, 2021-09-14:
On it, will hopefully have stuff built within the next 30 mins or so

David-Baddeley, 2021-09-14:
dropping the simulator shortcut was in response to reviewer comments on the PYMEVis paper

barentine, 2021-09-14:
Awesome, thanks @David-Baddeley!

David-Baddeley, 2021-09-14:
OK, build failed due to problems with anaconda upload, will take a bit of time to debug, but hopefully mid morning ...

David-Baddeley, 2021-09-14:
Should have new packages in the automated builds folder now

barentine, 2021-09-14:
Thanks David - testing now

barentine, 2021-09-14:
Ahh, sorry. I'd missed something in the traceback:
```
Traceback (most recent call last):
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\ExecTools.py", line 101, in execFile
    then()
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\microscope.py", line 916, in _post_init
    self.stackSettings = stackSettings.StackSettings(self)
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\stackSettings.py", line 89, in __init__
    from PYME.Acquire import webui
ImportError: cannot import name 'webui' from 'PYME.Acquire' (C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\Acquire\__init__.py)
```

David-Baddeley, 2021-09-14:
ahh, that would make sense ... merged. Will see if I can trigger another release (might be tricky with how we do version numbers)

David-Baddeley, 2021-09-15:
Should be a new build (with the old name) up now, but not 100% it will have updated properly. Will retrigger tomorrow to be absolutely sure.

barentine, 2021-09-15:
Retriggering might be good - a `conda update python-microscopy` in the `Anaconda Prompt (python-microscopy)` thinks it is god to go and maybe updates some other packages, but not PYME.

barentine, 2021-09-15:
Appears that both of the 9/14's have the '0' build number

David-Baddeley, 2021-09-15:
that makes sense ... ultimately I should make `update_version` check for a release with the same version number / date and change it to, e.g. `<version>post0` etc ...

barentine, 2021-09-15:
Hmm, just tried the package that was uploaded last night at 20:09, still same webui import error. See you on zoom in ~5 min

barentine, 2021-09-15:
Fixed! Could close, but maybe we leave it open until we update the default 'Downloads' package?

closed 2021-09-23 by David-Baddeley