python-microscopy/python-microscopy #1117: [BUG] FitInto failing on latgaussfitfr

barentine, 2021-09-15:
**Describe the bug**
click a test frame localization:
```
DEBUG:PYME.DSView.fitInfo:running draw with i as none
DEBUG:PYME.DSView.fitInfo:running draw in NONE mode!!!
DEBUG:PYME.DSView.fitInfo:running draw with i as none
DEBUG:PYME.DSView.fitInfo:running draw in NONE mode!!!
DEBUG:PYME.DSView.fitInfo:running draw with i as none
DEBUG:PYME.DSView.fitInfo:running draw in NONE mode!!!
DEBUG:PYME.DSView.fitInfo:running draw with i as none
DEBUG:PYME.DSView.fitInfo:running draw in NONE mode!!!
C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\DSView\arrayViewPanel.py:1086: wxPyDeprecationWarning: Call to deprecated item ImageFromData. Use :class:`Image` instead.
  img = wx.ImageFromData(ima.shape[1], ima.shape[0], ima.ravel())
ERROR:PYME.DSView.displayOptions:Error in OnChange
Traceback (most recent call last):
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\DSView\displayOptions.py", line 332, in OnChange
    fcn()
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\DSView\dsviewer.py", line 317, in update
    uCallback(self)
  File "C:\Users\aesb\AppData\Local\python-microscopy\lib\site-packages\PYME\DSView\modules\LMAnalysis.py", line 823, in update
    if 'fitInf' in dir(self) and not self.dsviewer.playbackpanel.tPlay.IsRunning():
AttributeError: 'PlayZTPanel' object has no attribute 'tPlay'
```

**To Reproduce**
Steps to reproduce the behavior:
1. simulate data in PYMEAcquire, spooling to cluster of one h5, hit `Analyse` while running to pop open PYMEImage
2. Click on test frame
3. crack open the fitinfo panel
4. See error

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: win10
 - installer 21.09.15

**Additional context**
get the same error if I open the series again in PYMEImage

David-Baddeley, 2021-09-15:
This will be related to the xyztc data model changes. Should hopefully be a simple fix.

closed 2021-09-16 by David-Baddeley

barentine, 2021-09-16:
```
Traceback (most recent call last):
  File "/Users/Andrew/python-microscopy/lib/python3.7/site-packages/PYME/DSView/displayOptions.py", line 332, in OnChange
    fcn()
  File "/Users/Andrew/python-microscopy/lib/python3.7/site-packages/PYME/DSView/dsviewer.py", line 317, in update
    uCallback(self)
  File "/Users/Andrew/python-microscopy/lib/python3.7/site-packages/PYME/DSView/modules/LMAnalysis.py", line 823, in update
    if 'fitInf' in dir(self) and not self.dsviewer.playbackpanel.tPlay.playback_running:
AttributeError: 'PlayZTPanel' object has no attribute 'tPlay'
```

PR #1118 (2021-09-16): issue 1117
fixes #1117
commit 7c3d4698 (2021-09-16): fix #1117