python-microscopy/python-microscopy #1121: [BUG] Measure Objects appears to be broken

lukasfue, 2021-09-22:
I generated a jittered triangulation image, thresholded it, labeled it, and used "Get Segmented IDs from labeled image". Then I tried to "measure objects" and received the following traceback:

 Error whilst running Analysis>Measure objects
==============================================
python-microscopy=21.09.16.post0
python=3.7.9, platform=win32
numpy=1.19.1, wx=4.0.4 msw (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "c:\users\laf62\python-microscopy\PYME\ui\progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "c:\users\laf62\python-microscopy\PYME\LMVis\Extras\objectMeasurements.py", line 149, in OnMeasure
    dlg = OMDialog(pipeline.keys())
  File "c:\users\laf62\python-microscopy\PYME\LMVis\Extras\objectMeasurements.py", line 61, in __init__
    self.SetSizerAndFit(vsizer)
wx._core.wxAssertionError: C++ assertion "((HWND)GetHWND())" failed at ..\..\src\msw\window.cpp(1162) in wxWindow::GetLayoutDirection(): invalid window

This approach definitely worked before, so I'm not sure what the issue is.

zacsimile, 2021-09-22:
Hmm I tried analysis > measure objects on a mac and got a straight segfault...

zacsimile, 2021-09-22:
This appears to be the line causing the segfault

```
hsizer.Add(wx.StaticText(self, -1, 'Label column:'), 0, wx.ALIGN_CENTER_VERTICAL|wx.RIGHT, 2)
```

Very confused. I am also on wxpython=4.0.4.

David-Baddeley, 2021-09-23:
hmm, definitely confusing ... I suspect that this and Zach's errors are unrelated, but can't be sure.

@lukasfue - is it possible that that there is more to that traceback / an earlier traceback that can be seen in the console? I can't see any obvious reason why you should be getting that traceback.

zacsimile, 2021-09-23:
I ran it on Windows and this is my traceback

```
SystemError: <class 'wx._core.WindowDestroyEvent'> returned a result with an error set                                  Traceback (most recent call last):                                                                                        File "c:\users\zrc4\code\python-microscopy\PYME\ui\manualFoldPanel.py", line 759, in OnResize                             if (not self._in_fold1) and self.IsShownOnScreen():                                                                 RuntimeError: wrapped C/C++ object of type foldPanel has been deleted  
```

zacsimile, 2021-09-23:
I get the same error as @lukasfue on my windows 10 machine. No additional info.

lukasfue, 2021-09-23:
Here is the entire output from running visgui (note sure if this is helpful or not)

`(pyme) C:\Users\laf62>visgui
Trying to load 3rd party recipe module pyme_4pi.recipe_modules.conversion
Loaded 3rd party recipe module pyme_4pi.recipe_modules.conversion
Trying to load 3rd party recipe module ch_shrinkwrap.recipe_modules.surface_fitting
Loaded 3rd party recipe module ch_shrinkwrap.recipe_modules.surface_fitting
Trying to load 3rd party recipe module cc_drift_cor.plugins.recipes.io
Loaded 3rd party recipe module cc_drift_cor.plugins.recipes.io
Trying to load 3rd party recipe module cc_drift_cor.plugins.recipes.chung_cc
Loaded 3rd party recipe module cc_drift_cor.plugins.recipes.chung_cc
Trying to load 3rd party recipe module cc_drift_cor.plugins.recipes.processing
Loaded 3rd party recipe module cc_drift_cor.plugins.recipes.processing
Trying to load 3rd party recipe module nep_fitting.recipe_modules.nep_fits
Loaded 3rd party recipe module nep_fitting.recipe_modules.nep_fits
Trying to load 3rd party recipe module cc_drift_cor.plugins.recipes.localisations
Loaded 3rd party recipe module cc_drift_cor.plugins.recipes.localisations
INFO:OpenGL.GLUT.special:Using NT-specific GLUT calls with exit callbacks
quaternion module not found, disabling custom clip plane orientations
quaternion module not found, disabling custom clip plane orientations
New Canvas
DEBUG:PYME.LMVis.visCore:Creating VisGUI menu bar
DEBUG:PYME.LMVis.Extras:Initializing QPobjectSegment plugin
DEBUG:PYME.LMVis.Extras:Initializing animation plugin
DEBUG:PYME.LMVis.Extras:Initializing clusterAnalysis plugin
DEBUG:PYME.LMVis.Extras:Initializing dockedPanel plugin
DEBUG:PYME.LMVis.Extras:Initializing extra_layers plugin
DEBUG:PYME.LMVis.Extras:Initializing fiducial plugin
DEBUG:PYME.LMVis.Extras:Initializing flagZMotionArtifacts plugin
DEBUG:root:Adding menu items for event filters
DEBUG:PYME.LMVis.Extras:Initializing fusionUtils plugin
DEBUG:PYME.LMVis.Extras:Initializing multiColourAnalysis plugin
DEBUG:PYME.LMVis.Extras:Initializing multiviewMapping plugin
DEBUG:root:Adding menu items for multi-view manipulation
DEBUG:PYME.LMVis.Extras:Initializing objectMeasurements plugin
DEBUG:PYME.LMVis.Extras:Initializing particleTracking plugin
DEBUG:PYME.LMVis.Extras:Initializing perFrameVariable plugin
DEBUG:PYME.LMVis.Extras:Initializing photophysics plugin
DEBUG:PYME.LMVis.Extras:Initializing pointSetGeneration plugin
DEBUG:PYME.LMVis.Extras:Initializing pointwiseColoc plugin
DEBUG:PYME.LMVis.Extras:Initializing saved_views plugin
Saved Views are deprecated -  use the ones with animation instead and remove me!
DEBUG:PYME.LMVis.Extras:Initializing scaleBox plugin
DEBUG:PYME.LMVis.Extras:Initializing sequentialSpecies plugin
DEBUG:PYME.LMVis.Extras:Initializing setBGImage plugin
DEBUG:PYME.LMVis.Extras:Initializing shiftmapGenerator plugin
DEBUG:PYME.LMVis.Extras:Initializing showShortcuts plugin
DEBUG:PYME.LMVis.Extras:Initializing snapshot plugin
DEBUG:PYME.LMVis.Extras:Initializing spherical_harmonics plugin
DEBUG:root:Adding menu items for spherical harmonic (shell) fitting
DEBUG:PYME.LMVis.Extras:Initializing subsets plugin
DEBUG:PYME.LMVis.Extras:Initializing surface_fitting plugin
DEBUG:PYME.LMVis.Extras:Initializing temperaturePlot plugin
DEBUG:PYME.LMVis.Extras:Initializing vibration plugin
Creating fold panel
GenPanels
Creating datasource panel
RecipeView._layout
Input datasources:  []
Image list size: 1
Creating layers GUI
height:  0
NB best size: wx.Size(200, 0)
RecipeView._layout
Traceback (most recent call last):
  File "c:\users\laf62\python-microscopy\PYME\ui\manualFoldPanel.py", line 759, in OnResize
    if (not self._in_fold1) and self.IsShownOnScreen():
RuntimeError: wrapped C/C++ object of type foldPanel has been deleted
Input datasources:  []
recipe.execute()
[... 236 lines cut ...]
wx._core.wxAssertionError: C++ assertion "GetEventHandler() == this" failed at ..\..\src\common\wincmn.cpp(478) in wxWindowBase::~wxWindowBase(): any pushed event handlers must have been removed

The above exception was the direct cause of the following exception:

SystemError: <class 'wx._core.WindowDestroyEvent'> returned a result with an error set
wx._core.wxAssertionError: C++ assertion "GetEventHandler() == this" failed at ..\..\src\common\wincmn.cpp(478) in wxWindowBase::~wxWindowBase(): any pushed event handlers must have been removed

The above exception was the direct cause of the following exception:

SystemError: <class 'wx._core.WindowDestroyEvent'> returned a result with an error set
Traceback (most recent call last):
  File "c:\users\laf62\python-microscopy\PYME\ui\manualFoldPanel.py", line 759, in OnResize
    if (not self._in_fold1) and self.IsShownOnScreen():
RuntimeError: wrapped C/C++ object of type foldPanel has been deleted
Input datasources:  ['_Localizations', 'FitResults', 'Labelled Image 0']
lw update
hl update
Traceback (most recent call last):
  File "c:\users\laf62\python-microscopy\PYME\ui\progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "c:\users\laf62\python-microscopy\PYME\LMVis\Extras\objectMeasurements.py", line 149, in OnMeasure
    dlg = OMDialog(pipeline.keys())
  File "c:\users\laf62\python-microscopy\PYME\LMVis\Extras\objectMeasurements.py", line 61, in __init__
    self.SetSizerAndFit(vsizer)
wx._core.wxAssertionError: C++ assertion "((HWND)GetHWND())" failed at ..\..\src\msw\window.cpp(1162) in wxWindow::GetLayoutDirection(): invalid window`

David-Baddeley, 2021-09-23:
I've got a possible candidate ...

I think `super().__init__(*args, **kwargs)` on line 28 of objectMeasurements.py should be `super().__init__(self, *args, **kwargs)`. Will have a test locally and see if I can replicate/fix (although don't currently have easy access to a windows machine for testing).

David-Baddeley, 2021-09-23:
`super()` was not the issue, found a different issue which could be responsible, fixed in #1122. Will trigger another package build, but keep this open until verified to work on win10

David-Baddeley, 2021-11-11:
Going to call this fixed, feel free to reopen if not.

closed 2021-11-11 by David-Baddeley