python-microscopy/python-microscopy #1138: [BUG] Error saving image taken using F5 - possible bug from crop refactoring?

csoeller, 2021-11-04:
**Describe the bug**
Saving snapshot raises error related to cropping/slicing.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `PYMEAcquire`
2. Take Snapshot (`F5`).
3. Save as...
4. See error

**Error message**:

```
Error whilst running File>&Save As
==============================================
python-microscopy=21.09.23
python=3.7.6, platform=darwin
numpy=1.19.2, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/ui/progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/DSView/dsviewer.py", line 339, in OnSave
    self.image.Save()
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/IO/image.py", line 1326, in Save
    self.filename = dataExporter.ExportData(self.data_xyztc, self.mdh, self.events, filename=filename, progressCallback=progressCallback)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/IO/dataExporter.py", line 500, in ExportData
    exp.Export(ds, filename, slice(0, ds.shape[0], 1), slice(0, ds.shape[1], 1), slice(0, ds.shape[2], 1),mdh, events, origName, progressCallback=progressCallback)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py37/PYME/IO/dataExporter.py", line 304, in Export
    xmd.setEntry('cropping.tslice', tslice.indices(data.shape[3]))
AttributeError: 'NoneType' object has no attribute 'indices'
```

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
  - python-microscopy=21.09.23
  - python=3.7.6, platform=darwin
  - numpy=1.19.2, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Bug also confirmed on Win10 with actual camera (Zyla).

zacsimile, 2021-11-04:
Snapshot works fine for me on python-microscopy=21.09.23 python=3.7.11, platform=darwin, numpy=1.19.2, wx=4.0.4 if I use the simulator in pymeacquire. Haven't tested with a real scope.

csoeller, 2021-11-04:
This is with latest github. Anyway, I will try your fix #1138.

Unrelatedly, with current github I cannot acquire any data using an actual camera (Zyla) without a cluster running, even if saving to file. Is this a feature? I appreciate that generally it is a good idea to have a cluster running in all realistic scenarios.

zacsimile, 2021-11-04:
Yeah I realized I did in fact have the same error and deleted my comment and then submitted this fix.

No idea about the cluster stuff.

David-Baddeley, 2021-11-04:
Save to file should work without the cluster, but will struggle to keep up
at anything close to full data rate on a Zyla - the HDF5 compression is way
too slow. Did you get any particular error message?

csoeller, 2021-11-04:
Yes, it raises an error. I need to get into the lab to save it, will do tomorrow and raise a bug report, you guys can decide if this is genuinely an issue.

csoeller, 2021-11-05:
fixed via merged #1139

closed 2021-11-05 by csoeller