python-microscopy/python-microscopy #1147: [SUPPORT] niche bug: svgwrite an implicit dependency for recipes in jupyter?

barentine, 2021-11-17:
**Describe the bug**
Sorry for being sparse - headed to bed but don't want to forget this. Funny little bug. If you create a recipe in jupyter notebook, but don't assign it to a variable such that the __repr__ method gets called, it'll want to svg it up, using a package you might well not have installed because we don't list it in pyme-depends meta.

```
ModuleNotFoundError                       Traceback (most recent call last)
/opt/miniconda3/envs/pdev/lib/python3.7/site-packages/IPython/core/formatters.py in __call__(self, obj)
    343             method = get_real_method(obj, self.print_method)
    344             if method is not None:
--> 345                 return method()
    346             return None
    347         else:

~/code/python-microscopy/PYME/recipes/recipe.py in _repr_svg_(self)
    728     def _repr_svg_(self):
    729         """ Make us look pretty in Jupyter"""
--> 730         return self.to_svg()

~/code/python-microscopy/PYME/recipes/recipe.py in to_svg(self)
    696     def to_svg(self):
    697         from . import recipeLayout
--> 698         return recipeLayout.to_svg(self.dependancyGraph())
    699 
    700     def layout(self):

~/code/python-microscopy/PYME/recipes/recipeLayout.py in to_svg(dg)
    412 
    413 def to_svg(dg):
--> 414     import svgwrite
    415     node_positions, connecting_lines = layout(dg)
    416 

ModuleNotFoundError: No module named 'svgwrite'
```

**To Reproduce**
Steps to reproduce the behavior:
```
from PYME.recipes.recipe import Recipe
from PYME.recipes import modules

render_rec="""- localisations.DensityMapping:
    colours:
    - chan0
    - chan1
    inputLocalizations: color_mapped
    jitterScale: 30.0
    manualXYBounds:
    - -50.0
    - 1100.0
    - 30000.0
    - 30000.0
    outputImage: g2d
    pixelSize: 15.0
    renderingModule: Gaussian
    xyBoundsMode: manual
- output.ImageOutput:
    filePattern: '{output_dir}/{file_stub}_g2d.h5'
    inputName: g2d
- localisations.DensityMapping:
    colours:
    - chan0
    - chan1
    inputLocalizations: color_mapped
    jitterScale: 30.0
    jitterScaleZ: 30.0
    manualXYBounds:
    - -50.0
    - 1100.0
    - 30000.0
    - 30000.0
    outputImage: g3d
    pixelSize: 15.0
    xyBoundsMode: manual
    zBoundsMode: min-max
    zSliceThickness: 15.0
- output.ImageOutput:
    filePattern: '{output_dir}/{file_stub}_g3d.h5'
    inputName: g3d"""

Recipe.fromYAML(render_rec)
```
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10]
 - Python version:
 - numpy version:
 - wxpython version [if a UI bug]:

**Additional context**
Add any other context about the problem here.

David-Baddeley, 2021-11-17:
Thanks, we should probably add svgwrite to the dependencies... Will check package availability

David-Baddeley, 2021-11-25:
Argh ... `svgwrite` is not available by default via conda, and niche enough that I'm a bit loath to maintain our own packaging for it - best solution might be to find some way of patching `_repr_svg_` so that it fails gracefully and displays a message along the lines of "for a graphical recipe representation, install svgwrite".

PR #1155 (2021-11-26): issue1147
fix #1147 by failing gracefully / providing a text repr if svgwrite not available
closed 2021-11-26 by David-Baddeley