python-microscopy/python-microscopy #1148: [BUG] animation panel settings issues

barentine, 2021-11-17:
**Describe the bug**
#1, panel doesn't actually open when you click fold. commit messages make this seem like a known issues
#2 once you fix that, 
```
Traceback (most recent call last):
  File "c:\userfiles\andrew\code\python-microscopy\PYME\contrib\wxPlotPanel.py", line 69, in _onIdle
    self._SetSize()
  File "c:\userfiles\andrew\code\python-microscopy\PYME\contrib\wxPlotPanel.py", line 86, in _SetSize
    float( pixels[1] )/self.figure.get_dpi() )
  File "C:\Userfiles\andrew\miniconda3\envs\pyme\lib\site-packages\matplotlib\figure.py", line 902, in set_size_inches
    raise ValueError(f'figure size must be positive finite not {size}')
ValueError: figure size must be positive finite not [0.   5.97]
```

**To Reproduce**
Steps to reproduce the behavior:
1. open something in PYMEVis
2. Extras>animation
3. click settings
4. See nothing
5. fix code
6. see traceback

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: win10
 - Python version: 3.7
 - numpy version: 1.19.2
 - wxpython version [if a UI bug]: 4.0.4

**Additional context**
will probably submit a PR shortly

David-Baddeley, 2021-11-17:
Traceback seems to be unrelated to animation... Maybe a bit more context as to when this is being used. Only use of wxplotpanel in pymevis that I know of is to display fit status when embedded in dh5view doing real-time analysis.

David-Baddeley, 2021-11-17:
But I could easily have missed something

barentine, 2021-11-17:
Yeah, could have another look and see how repeatable that bit was. Regardless, settings panel is currently only updating after fold if you resize.

David-Baddeley, 2021-11-19:
I think I've got a fix for this ...

PR #1149 (2021-11-24): Annotation tidy up
Some progress on #252, adds deletion of annotations. Incidentally fixes #1148
closed 2021-11-24 by David-Baddeley