python-microscopy/python-microscopy #1151: [BUG] cropping refactor not working with a simple recipe

barentine, 2021-11-24:
**Describe the bug**

```
 Error whilst running Batch Analysis
==============================================
python-microscopy=21.09.23
python=3.7.6, platform=darwin
numpy=1.19.2, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

Traceback
=========
Traceback (most recent call last):
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/recipeGui.py", line 913, in OnBake
    self.last_run = batchProcess.bake(self.rm.activeRecipe, {'input':inputs[0]}, out_dir, num_procs=num_procs, metadata_defaults=self._default_md)
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/batchProcess.py", line 80, in bake
    last_rec = runRec(task)
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/batchProcess.py", line 27, in runRec
    r = runRecipe.runRecipe(*args)
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/runRecipe.py", line 105, in runRecipe
    recipe.save(context)
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/recipe.py", line 515, in save
    mod.save(self.namespace, context)
  File "/Users/Andrew/code/python-microscopy/PYME/recipes/output.py", line 219, in save
    v.Save(out_filename)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/image.py", line 1326, in Save
    self.filename = dataExporter.ExportData(self.data_xyztc, self.mdh, self.events, filename=filename, progressCallback=progressCallback)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/dataExporter.py", line 504, in ExportData
    exp.Export(ds, filename, slice(0, ds.shape[0], 1), slice(0, ds.shape[1], 1), slice(0, ds.shape[2], 1),mdh, events, origName, progressCallback=progressCallback)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/dataExporter.py", line 119, in Export
    data, tslice, xSize, ySize, nZ, nT, nChans, nframes = self._prepare(data, xslice, yslice, zslice, tslice)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/dataExporter.py", line 84, in _prepare
    cropped = CropDataSource.DataSource(data, xrange=xslice, yrange=yslice, zrange=zslice, trange=tslice)
  File "/Users/Andrew/code/python-microscopy/PYME/IO/DataSources/CropDataSource.py", line 88, in __init__
    assert(isinstance(dataSource, XYZTCDataSource))
AssertionError

```

**To Reproduce**
Steps to reproduce the behavior:
```
- localisations.DensityMapping:
    colours:
    - chan0
    - chan1
    inputLocalizations: mapped
    jitterScale: 30.0
    manualXYBounds:
    - 0.0
    - 0.0
    - 30000.0
    - 30000.0
    outputImage: g2d
    pixelSize: 15.0
    renderingModule: Gaussian
    xyBoundsMode: manual
- output.ImageOutput:
    filePattern: '{output_dir}/{file_stub}_g2d.h5'
    inputName: g2d
- localisations.DensityMapping:
    colours:
    - chan0
    - chan1
    inputLocalizations: mapped
    jitterScale: 30.0
    jitterScaleZ: 30.0
    manualXYBounds:
    - 17623.15
    - 0.0
    - 20021.52
    - 30000.0
    outputImage: g3d
    pixelSize: 15.0
    xyBoundsMode: manual
    zBoundsMode: min-max
    zSliceThickness: 15.0
- output.ImageOutput:
    filePattern: '{output_dir}/{file_stub}_g3d.h5'
    inputName: g3d
- processing.Projection:
    axis: 0
    inputName: g3d
    outputName: projected
- output.ImageOutput:
    filePattern: '{output_dir}/{file_stub}_g3d-side-projection.h5'
    inputName: projected

```

**Expected behavior**
crop and save the stack, don't freak out

**Platform (please complete the following information - or copy-paste from error dialog if shown):**
 - OS: [e.g. Win10] osx 10.15.7
 - Python version: 3.7
 - numpy version: 1.19.2
 - wxpython version [if a UI bug]: 4.0.4

**Additional context**
potentially see #1140

barentine, 2021-11-24:
Issue is the projection (filter) passes a list wrap, so it has 5 dimensions but is not an EYZTCArrayDataSource. Will change title accordingly

David-Baddeley, 2021-11-24:
I think I got ahead of myself with the xyztc stuff in recipes - in general one can't assume that a recipe input is xyztc. I don't think the exporter changes will help here. The safe workaround at this point is probably to promote to xyztc, if needed, within the cropping module.

David-Baddeley, 2021-11-24:
Sorry, traceback suggests it is just in export, will look more closely.

barentine, 2021-11-24:
Correct, just in export. Throwing the listWrapper into an XYZTC 'works' but I did it wrong and get a single-channel output (only the first color)

barentine, 2021-11-24:
#1152 now works for me, but it sounds like you're saying you want to handle it in the cropping module rather than exporter?

David-Baddeley, 2021-11-24:
Not sure - might have got it wrong on first read, will look over when back
in front of my computer.

David-Baddeley, 2021-11-24:
Have alternate fix queued in #1149 (assertion in `CropDataSource` was arguably too strict, and fixing it there is likely to prevent it hitting us elsewhere).

closed 2021-11-24 by David-Baddeley