python-microscopy/python-microscopy #1153: [ENHANCEMENT] GUI toggle crosshairs and scalebar in PYMEImage

barentine, 2021-11-24:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when I need to render 16 images and for each one I have to execute two lines of code in the shell. I can do it, but others might not know the secret passwords

**Describe the solution you'd like**
1. leave the crosshairs off unless someone clicks the button on the side panel.
2. scalebar toggle/options more like in PYMEVis

David-Baddeley, 2021-11-24:
looks like a duplicate of #876,

should now be fixed as of #1149 

[image]

can discuss whether default mode on startup should show crosshairs or not, but there is now an easy way to toggle them in the GUI. Probably not quite as slick as the VisGUI toolbar toggling, but more versatile (as it also allows toggling other overlays - e.g. annotations, point selections, etc ...

David-Baddeley, 2021-11-24:
[image]

David-Baddeley, 2021-11-25:
@barentine - will close as already fixed unless you feel strongly about crosshairs on startup. Another consideration would be to disable them on export (although this needs a fair bit of special-case logic).

closed 2021-11-29 by David-Baddeley