python-microscopy/python-microscopy #1157: [BUG] error rendering in VisGui - 'DSViewFrame' object has no attribute 'view'

csoeller, 2021-11-26:
**Describe the bug**
When rendering an H5R file an error is triggered, presumably because the dsviewer interface has changed with the latest merged PRs.

**To Reproduce**
Steps to reproduce the behavior:
1. Open `.h5` with `PYMEVisualize`
2. Generate rendering (I tried Gaussian based).
4. See error

**Versions**
- python-microscopy=21.09.23.post0.dev[git]e6dadf368933
- python=3.7.6, platform=darwin
- numpy=1.19.2, wx=4.0.4 osx-cocoa (phoenix) wxWidgets 3.0.5

**Traceback**
```
Traceback (most recent call last):
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/ui/progress.py", line 153, in func
    return fcn(*args, **kwargs)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/LMVis/renderers.py", line 313, in GenerateGUI
    glCanvas=self.visFr.glCanvas, parent=self.mainWind)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/DSView/dsviewer.py", line 524, in ViewIm3D
    parent=parent, glCanvas=glCanvas)
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/DSView/dsviewer.py", line 157, in __init__
    self.create_overlay_panel()
  File "/Users/csoe002/Documents/src/PYME-src/pyme-py3/PYME/DSView/dsviewer.py", line 262, in create_overlay_panel
    self.overlaypanel = OverlayPanel(self, self.view, self.image.mdh)
AttributeError: 'DSViewFrame' object has no attribute 'view'
```

David-Baddeley, 2021-11-26:
Thanks! That was an embarrassing slip, should be fixed with #1157.

David-Baddeley, 2021-11-26:
We need to find a way of automating gui tests ...

closed 2021-11-26 by David-Baddeley

csoeller, 2021-11-26:
Thanks for the prompt fix!

PR #1158 (2021-11-26): issue 1157
fixes #1157